merbanan/rtl_433 #582: Problem with solar radiation values with the WH3080

Chak10, 2017-07-13:
Hi, I think there is a problem with solar radiation value, in fact, the console has a totally different value. I noticed that the conversion index is probably another.

Maybe wrong [Line#L371](https://github.com/merbanan/rtl_433/blob/master/src/devices/fineoffset_wh1080.c#L371)
```c
const float wm = (get_rawlight(br)/6830);
```

I think this is correct
```c
const float wm = (get_rawlight(br)*0.00079);
```
[Look Reference](https://physics.stackexchange.com/questions/135618/rm-lux-and-w-m2-relationship)
[Inv_Square_Law.pdf](https://github.com/merbanan/rtl_433/files/1146139/Inv_Square_Law.pdf)

zuckschwerdt, 2017-07-13:
As the signal only contains decilux the derived values should be removed anyway. The reasoning is to output values with native units from the signal and let some presentation software define mappings and conversions.

Chak10, 2017-07-14:
Yes, that's true, but I only told this because I found different values from the console. Technically, I think the console also does this kind of conversion. However, having conversions already performed is generally more convenient.
P.S. Having even the raw values would be ideal.

zuckschwerdt, 2017-07-14:
Exactly, having a single true data point will carry more meaning. Each presentation layer can then define consistent conversion and rounding.

ovrheat, 2017-08-08:
Hi,
I really can't remember how I've found that conversion value, but I can confirm that by using that, rtl_433 "wm" output is *always* perfectly matching with my WH3080 LCD console's "wm" reading. 
Using your proposed value I see completely different output that absolutely does not match with the console, so my guess is that we have different versions of the same product and one of them carry revised conversion routines. 
Can you take a look to which WH3080 version do you own? Is it 433 or 868 Mhz or other? What version is it marked in the green round label? And could you run rtl_433 in debug/analyzing mode to see if your station messages are 87 or 88 bits longs (this seems to be one of the most important 'detector' between versions) ?

Chak10, 2017-08-08:
Hi, my station is a WH3080 863Mhz (868300000hz).

Debug(.txt) and Analize files (.data) with [my fork](https://github.com/Chak10/rtl_433)

[Test.zip](https://github.com/merbanan/rtl_433/files/1209485/Test.zip)

I do not know why, but the original conversion formula does not coincide with my console.
After all, the station sends us deci-lux.
If you convert:

100'000 lux ([_direct sunlight see on wikipedia_](https://en.wikipedia.org/wiki/Lux)) = 1'000'000 decilux 

with the original formula 

wm = 1'000'000 / 6830 ≃ 146.41 W/m2 ???

other formula

wm = 1'000'000 * 0.00079 = 790 W/m2

ovrheat, 2017-08-08:
Well, OK. At least now we're sure that we own different WH3080 versions, starting by your messages length (your is 87 bits longs, while my version works with 88 bits). Furthermore our frequencies are different (my station works on 433Mhz). 
It would be useful to know why our internal consoles conversions are different too; maybe because of a correction on the more recent model? Or differents light sensors with different sensibility are used? Or, again, is there some kind of limitation for the market for which each model is intended to? (for example: the sun in Australia is "a little bit" more powerful than in Iceland, and needs different kind of sensor...) :-) Who knows?
Now: a way to fix this could be to keep a single 'raw light' decilux field, like **zuckschwerdt** suggested some post ago, assuming the sensor is the same for both models and/or its sensitivity is the same too.
At the same time, however, I think that dropping the already converted fields (lux, wm, fc, the same fields that we can read onto the LCD console) would be a big problem for a not-so-scientist user, forcing him to crawl around the web looking for (hopefully) correct conversion formulas for their projects. The already calculated fields (when we'll have this bug fixed) are certainly more useful and reflecting the original LCD console's data usage.
On another side I don't own a professional solar radiation reader, so I don't know if my station is showing correct sun power values or not. All I can say is that rtl_433 calculated values are perfectly matching with these from my original LCD console...

So perhaps we should add the 'raw light' field and also keep the calculated fields, correcting them according to the version of WH3080 in use (looking at the 87-88 bits message length). Anybody could use what he choose as best for his project.

Zuckschwerdt, Benjamin, anyone... What do you think about it?

P.S: for Chak10: are your console's values perfectly matching while your use Wikipedia's conversion formula?

Chak10, 2017-08-09:
@ovrheat Yes, the values coincide perfectly.
Console => 306 W/m2
Wikipedia's conversion formula* => 306.68 W/m2

_*The source is not wikipedia but [this](https://physics.stackexchange.com/a/193212)._

ovrheat, 2017-08-26:
@Chak10 OK. Take a look at [this link](http://sandaysoft.com/forum/viewtopic.php?t=3979) also, there is some interesting post that might clarify the matter.

bostjanpo, 2018-07-10:
Hi, I experience the same as @Chak10. I have WH4000 weather station (it seems the outdoor modules are the sam as WH3080) which is recognized as WH3080. All parameters match console readings just radiation in W/m² does not. I get correct values if I use wm = dcLux * 0.00079 = xx W/m2 formula (and rtl_433  uses dcLux/6830, which does not mach my console readings)