merbanan/rtl_433 #588: Help decoding wanted

augustodinizl, 2017-07-15:
Hello everyone, greetings from Brazil !

I need some help from the rtl_experts :-D, i have some spare transmitters from a brazillian brand named "intelbras", and i'm attempting to decode the little bastard's, but due to my little (almost 0) experience, i'm having troubles with that, can anybody help me?

Ps.: Sorry, english isn't my first language
Ps2.: Below are the -A output from one of them (which i was unable to figure out, because it's FSK, OOK one's are simple to understand.)
Ps3.: I've made a pull request [#175](https://github.com/merbanan/rtl_433_tests/pull/175) following the rules (i guess)

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
Detected OOK package    @ 2017-07-15 09:55:29
Analyzing pulses...
Total count:    9,  width: 95608                (382.4 ms)
Pulse width distribution:
 [ 0] count:    1,  width:    43 [43;43]        ( 172 us)
 [ 1] count:    8,  width:  9953 [9952;9956]    (39812 us)
Gap width distribution:
 [ 0] count:    8,  width:  1991 [1979;2076]    (7964 us)
Pulse period distribution:
 [ 0] count:    1,  width:  2119 [2119;2119]    (8476 us)
 [ 1] count:    7,  width: 11933 [11932;11935]  (47732 us)
Level estimates [high, low]:  15963,     12
Frequency offsets [F1, F2]:   -1488,      0     (-5.7 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with fixed gap
Attempting demodulation... short_limit: 4998, long_limit: 2077, reset_limit: 2077, demod_arg: 0
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {9} 80 00 : 10000000 0

Detected OOK package    @ 2017-07-15 09:55:31
Analyzing pulses...
Total count:    9,  width: 95617                (382.5 ms)
Pulse width distribution:
 [ 0] count:    1,  width:    42 [42;42]        ( 168 us)
 [ 1] count:    8,  width:  9953 [9950;9957]    (39812 us)
Gap width distribution:
 [ 0] count:    8,  width:  1993 [1978;2078]    (7972 us)
Pulse period distribution:
 [ 0] count:    1,  width:  2120 [2120;2120]    (8480 us)
 [ 1] count:    7,  width: 11934 [11932;11936]  (47736 us)
Level estimates [high, low]:  15871,      5
Frequency offsets [F1, F2]:   -1507,      0     (-5.7 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with fixed gap
Attempting demodulation... short_limit: 4997, long_limit: 2079, reset_limit: 2079, demod_arg: 0
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {9} 80 00 : 10000000 0

zuckschwerdt, 2017-07-15:
Hi! At a first glance your signals and documentation are good, I'll have a look through them tomorrow.
If you want to inspect the signals visually try: https://github.com/zuckschwerdt/sdr-spectrogram -- It uses sox to turn the data into spectrogram pictures.
If you want to demodulate FSK data try Audacity with IQ-FM demod from https://github.com/zuckschwerdt/sdr-nyquist-plugins -- should give you an OOK-like waveform.

zuckschwerdt, 2017-07-15:
Also a single squashed commit for the PR would be much better. Perhaps see https://stackoverflow.com/questions/5189560/squash-my-last-x-commits-together-using-git and then use `push -f` to update the PR.

zuckschwerdt, 2017-07-15:
Looking at your https://github.com/augustodinizl/rtl_433_tests/blob/master/tests/intelbras/XAS_4010_Smart/01/gfile001.data shows plain 2-FSK with 621µs bit period and should come out as `1010101010101010101010100010110111010100001010110100001000010011` but the higher frequency (5 points per full oszillation ~ 50kHz) is rather big and might mess up demodulation.
Can you take another sample with 1024k sample rate (`-s 1024000 -a -t`)?

augustodinizl, 2017-07-15:
> blob/master/tests/intelbras/XAS_4010_Smart/01/gfile001.data shows plain

Greetings!!!

Samples taken and in the folder:

https://github.com/augustodinizl/rtl_433_tests/tree/master/tests/intelbras/XAS_4010_Smart/1024k

Thank you!

zuckschwerdt, 2017-07-15:
Good clean signal, you can even make out the bits in the spectrogram.

[image]

But somehow the demodulation isn't picking up on this. Perhaps you can take one more sample and this time offset the base frequency so that the lower frequency isn't so close to DC level. I.e. if you are using `-f 433920000` then try `-f 433930000` keep the `-s 1024000`. You can verify the sample by opening it in Audacity (import raw u8 2ch) and zooming in on both frequencies. They should look like proper sine waves.

augustodinizl, 2017-07-15:
Well, with your guidance, that's getting better
[image: Imagem inline 1]

but the demodulator still get same result

[image: Imagem inline 2]

zuckschwerdt, 2017-07-16:
Your inline images got lost when posting this to github by mail.

augustodinizl, 2017-07-16:
Oooops, my bad, sorry!!!

I think i've been able to "figure it out", with your help, just trying to
get the decoder reading the samples accurately.

Only change between the two states, is the four final "bits".

I know that control have a weak battery alarm, but i dont have any "dying"
battery at the moment.

Any guidance in these steps?

augustodinizl, 2017-07-16:
I think that's right now, since it's fsk modulated.
That transmitter show only two states (obviouslly), at open/close, and
every 15 minutes.
Attached, is my annalisis, that i guess it's right now...

Thanks a lot!

2017-07-16 11:09 GMT-03:00 Augusto Diniz Lisboa <augusto.diniz.l@gmail.com>:

zuckschwerdt, 2017-07-16:
Yes, it looks good. Those x1527 chips transmit 20 bit factory set ID and a 4 bit custom code.
Look at Akhan, Generic_remote, Chuango, Quhwa, Steffen, they all are for the same x1527 chip. Just change the timings and it should work right away (mostly).

zuckschwerdt, 2017-07-16:
Oops. Wrong issue. x1527 is OOK not FSK.

augustodinizl, 2017-07-16:
it's the HT 6P20 chip (well, it's specified at the manual. :-))

zuckschwerdt, 2017-07-16:
Very useful info! The HT6P20 is similar to the x1527. It's also an One-Time-Programmable (OTP) Encoder. The datasheet talks about 4-bit (8-bit?) control codes and 24-bit address codes. Those can be used in different modes: 24-Address 0-Data, 22-Address 2-Data, 20-Address 4-Data, (and maybe 16-Address 8-Data). 20+4 matches your findings. The encoding description looks strange and more like PWM than PCM/NRZ. I need to read that again...

zuckschwerdt, 2017-07-16:
Actually... are there markings on that "big" chip (right next to the FSK pad)? The HT6P20 is not advertised as having a 10-Pin SOP... should be 8 or 16 pin. And the signal does not match the description of the expected HT6P20 output.

augustodinizl, 2017-07-16:
HMMM, you're right, at manual, they said that usin HT6P20, but the "big chip" is another... here's the pic
[image]

zuckschwerdt, 2017-07-16:
Nice! That explains all :) That looks like a 10-MSOP and the markings `10C2 CC5Z 643` match the Si4010, which comes in 10-Pin MSOP! Actually the Si4010 is rather powerful with an MCU, not a simple state machine like the HT6P20 or the low-end x1527. And there is even AES encryption.
https://www.silabs.com/documents/public/data-sheets/Si4010.pdf
(It's something you expect in a high-end device! So far all my motion sensors and buttons feature something like the x1527 only…)

augustodinizl, 2017-07-16:
WOW!!! but it's "downused" by "intelbras", since its only sending the same message
 i've heard from some friends, that the intelbras is getting secretly huge steps in IoT directions...

Well, i will update the datasheet

Thanks a LOT!!!

merbanan, 2018-12-03:
This type of device should be supported with the flex decoder. @zuckschwerdt can you describe how one would figure out the FSK paramters for the intelbras/XAS_4010_Smart/01/gfile001.cu8 sample and how to add it to the config file ?

zuckschwerdt, 2018-12-04:
Just running `-A` should now suggest a working `-X` spec. Please reopen with a new sample if you need further help.

closed 2018-12-04 by zuckschwerdt

andrempo, 2019-04-27:
Hi @zuckschwerdt, I'm trying to decode the same transmitter but the -A option is not showing the correct spec:
` 
Guessing modulation: Pulse Width Modulation with fixed gap
Attempting demodulation... short_width: 160, long_width: 39748, reset_limit: 8328, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=160,l=39748,r=8328,g=0,t=15832,y=0'
`
I am expecting to be something like  -X 'n=alarm,m=FSK_PCM,s=620,l=620,r=7000' but i cant get it to work. Any ideias?

zuckschwerdt, 2020-05-09:
The `-A` analyzer should show "Detected FSK package", otherwise no FSK was demodulated.
Which sample file are you refering that shows this behaviour? Perhaps upload a zipped cu8 of your sample data so we can use it to improve the FSK detection. Not sure if that works out though.

azan3l4t, 2020-05-10:
Here are 2 test zipped. 

[g001_433.92M_1024k_OpenClose.zip](https://github.com/merbanan/rtl_433/files/4605190/g001_433.92M_1024k_OpenClose.zip)

It's the same as the one in: 
[https://github.com/merbanan/rtl_433_tests/tree/master/tests/intelbras/XAS_4010_Smart](url)

I don't know what i've done yet,  other than a reboot... but now i seem to be getting 8x the same "Detected FSK Package" (which corresponds to the 8x it's sent by the device in a row for each trigger,  but in the end rtl_433 has no clue in guessing modulation;

Detected FSK package    2020-05-10 02:21:00
Analyzing pulses...
Total count:   40,  width: 39.75 ms             (40703 S)
Pulse width distribution:
 [ 0] count:    1,  width:   12 us [12;12]      (  12 S)
 [ 1] count:    4,  width:   17 us [16;20]      (  17 S)
 [ 2] count:    1,  width:   76 us [76;76]      (  78 S)
 [ 3] count:    9,  width: 1349 us [1225;1561]  (1381 S)
 [ 4] count:    2,  width: 3546 us [3422;3671]  (3631 S)
 [ 5] count:    9,  width:  270 us [231;315]    ( 276 S)
 [ 6] count:    4,  width: 2375 us [2050;2542]  (2432 S)
 [ 7] count:    3,  width: 1005 us [931;1069]   (1029 S)
 [ 8] count:    2,  width:  754 us [703;805]    ( 772 S)
 [ 9] count:    2,  width:  501 us [485;518]    ( 513 S)
 [10] count:    1,  width: 1729 us [1729;1729]  (1771 S)
 [11] count:    2,  width:  347 us [346;348]    ( 355 S)
Gap width distribution:
 [ 0] count:   30,  width:   10 us [10;12]      (  10 S)
 [ 1] count:    7,  width:   16 us [14;19]      (  16 S)
 [ 2] count:    2,  width:   27 us [27;27]      (  28 S)
Pulse period distribution:
 [ 0] count:    1,  width:   23 us [23;23]      (  24 S)
 [ 1] count:    2,  width:   34 us [34;35]      (  35 S)
 [ 2] count:    2,  width:   45 us [44;47]      (  46 S)
 [ 3] count:    1,  width:   88 us [88;88]      (  90 S)
 [ 4] count:    9,  width: 1360 us [1240;1575]  (1393 S)
 [ 5] count:    2,  width: 3556 us [3432;3681]  (3641 S)
 [ 6] count:    9,  width:  280 us [241;333]    ( 287 S)
 [ 7] count:    4,  width: 2385 us [2062;2552]  (2442 S)
 [ 8] count:    3,  width: 1016 us [941;1080]   (1040 S)
 [ 9] count:    2,  width:  768 us [718;818]    ( 786 S)
 [10] count:    2,  width:  512 us [496;527]    ( 524 S)
 [11] count:    1,  width: 1740 us [1740;1740]  (1782 S)
 [12] count:    1,  width:  356 us [356;356]    ( 365 S)
Level estimates [high, low]:  15960,   2064
RSSI: -0.1 dB SNR: 8.9 dB Noise: -9.0 dB
Frequency offsets [F1, F2]:   20740, -32768     (+324.1 kHz, -512.0 kHz)
Guessing modulation: No clue...

Using URH i was able to decode and get very good signal demodulation:
[image]

Detailed 1-bit parameters;
[image]

zuckschwerdt, 2020-05-10:
The reboot likely shifted the frequency a bit. You are too close to one of the frequencies. Try to get between the fs or offset slightly (e.g. 434M should work).

Can you record a sample where both frequencies are well away from 0 Hz?

The FSK demod is tuned for much faster signals (usual FSK is e.g. 10x faster). This will add noise to the demodulated "bits". Perhaps this signal can be oversampled and then downsampled in the decoder.

azan3l4t, 2020-05-10:
Here are 2 new samples taken at 433.870MHz (between fs), and 434.000MHz (offset)

[g016_434M_1024k.zip](https://github.com/merbanan/rtl_433/files/4606726/g016_434M_1024k.zip)
[g014_433.887M_1024k.zip](https://github.com/merbanan/rtl_433/files/4606727/g014_433.887M_1024k.zip)

augustodinizl, 2020-08-16:
> Hello @zuckschwerdt , @merbanan
> I am too trying to decode the same device without success. The same exact spectrum shows up when recording it, decoding it with -A option gives me :
> 
> ```
> Detected OOK package    2020-05-07 21:49:34
> Analyzing pulses...
> Total count:    9,  width: 381.77 ms            (390932 S)
> Pulse width distribution:
>  [ 0] count:    1,  width:  131 us [131;131]    ( 134 S)
>  [ 1] count:    8,  width: 39732 us [39730;39735]       (40686 S)
> Gap width distribution:
>  [ 0] count:    8,  width: 7972 us [7921;8320]  (8163 S)
> Pulse period distribution:
>  [ 0] count:    1,  width: 8451 us [8451;8451]  (8654 S)
>  [ 1] count:    7,  width: 47654 us [47651;47660]       (48798 S)
> Level estimates [high, low]:  15944,     11
> RSSI: -0.1 dB SNR: 31.6 dB Noise: -31.7 dB
> Frequency offsets [F1, F2]:    1298,      0     (+20.3 kHz, +0.0 kHz)
> Guessing modulation: Pulse Width Modulation with fixed gap
> Attempting demodulation... short_width: 131, long_width: 39732, reset_limit: 8321, sync_width: 0
> Use a flex decoder with -X 'n=name,m=OOK_PWM,s=131,l=39732,r=8321,g=0,t=15840,y=0'
> pulse_demod_pwm(): Analyzer Device
> bitbuffer:: Number of rows: 1
> [00] { 9} 80 00     : 10000000 0
> ```
> 
> But i am expecting it to be FSK, Already tried forcing it with -Y with all possible options.
> If i use -X 'n=name,m=OOK_PWM,s=131,l=39732,r=8321,g=0,t=15840,y=0' as suggested, i get a signal detection, but no identification to be able to distinguish between the modules.
> 
> Please let me know what's the best course of action, not sure it's going to be of any help to upload another .cu8 sample that looks exact the same. Let me know...
> 
> @augustodinizl where you able to make it work in FSK Modulation after all ?
> 
> Thanks!

Nop, i've got nothing yet... just getting same troubles

augustodinizl, 2020-08-16:
i've got more samples, @zuckschwerdt @merbanan is there anything else i can do to help?

Ps. that's 3 files, first 2, from the same sensor from original samples, last one from another sensor still in OOK modulation

g004_433.92M_250k.cu8.zip
g005_433.92M_250k.cu8.zip
g006_433.92M_250k.cu8.zip

augustodinizl, 2020-08-16:
[g004_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/5079913/g004_433.92M_250k.cu8.zip)
[g005_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/5079914/g005_433.92M_250k.cu8.zip)
[g006_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/5079915/g006_433.92M_250k.cu8.zip)

augustodinizl, 2020-08-16:
Sorry about the previous (junky) samples, there is better samples for 2 sensors in 1024k for analysis
[sensor 1.zip](https://github.com/merbanan/rtl_433/files/5080002/sensor.1.zip)
[sensor 2.zip](https://github.com/merbanan/rtl_433/files/5080003/sensor.2.zip)

zuckschwerdt, 2020-08-16:
The new samples have been added in https://github.com/merbanan/rtl_433_tests/tree/master/tests/intelbras/XAS_4010_Smart/07 and https://github.com/merbanan/rtl_433_tests/tree/master/tests/intelbras/XAS_4010_Smart/08

augustodinizl, 2020-08-16:
> The new samples have been added in https://github.com/merbanan/rtl_433_tests/tree/master/tests/intelbras/XAS_4010_Smart/07 and https://github.com/merbanan/rtl_433_tests/tree/master/tests/intelbras/XAS_4010_Smart/08

Thanks man... is there something else i can do?

zuckschwerdt, 2020-08-16:
I'll try to look into this again. Maybe we can also get @evilpete interested :)

evilpete, 2020-08-16:
what needs to be done?

augustodinizl, 2020-08-16:
> what needs to be done?

Hello, i don´t know what to do next... maybe you can help?

augustodinizl, 2020-08-17:
@zuckschwerdt, @evilpete  on urh i was able to decode every sample i've take on rtl_433, but somehow the builtin decoder can't figure that my tx's using FSK, instead of OOK (-X suggest everytime to use OOK). that SS is somehow some good info for you?
[image]
[image]

boemekeld, 2025-06-03:
[sensor_1.zip](https://github.com/user-attachments/files/20578343/sensor_1.zip)
[sensor_2.zip](https://github.com/user-attachments/files/20578342/sensor_2.zip)

I'm trying to create a Flex decoder for this sensor. I tested a few hundred combinations of parameters, but I haven't had any success.

Below are 4 captures related to the opening and closing of 2 sensors (captured with -f 433920000 and -s 1024000).

Using URH, with Samples/Symbol = 600 and Error Tolerance > 2, I was able to decode:

SENSOR 1:
101010101010101010101010 001011011101010000100010101100000101 0011 - OPEN
101010101010101010101010 001011011101010000100010101100000101 0101 - CLOSE

SENSOR 2:
101010101010101010101010 001011011101010011100010101110110101 0011 - OPEN
101010101010101010101010 001011011101010011100010101110110101 0101 - CLOSE

@merbanan @zuckschwerdt can u help?

Thanks.

[image]

boemekeld, 2025-06-03:
@zuckschwerdt, thank you for your response and sorry for the low-quality capture. I’ll work on getting a better one. However, even with a high-quality capture like @augustodinizl’s (https://github.com/merbanan/rtl_433_tests/blob/master/tests/intelbras/XAS_4010_Smart/07/g001_433.92M_1024k_open.cu8), the parameters returned by -A for the flex decoder still don’t work:

```
rtl_433.exe -r g001_433.92M_1024k_open.cu8 -A
rtl_433 version 25.02 branch  at 202502191252 inputs file rtl_tcp RTL-SDR SoapySDR
[Input] Test mode active. Reading samples from file: g001_433.92M_1024k_open.cu8
Detected OOK package    @0.077500s
Analyzing pulses...
Total count:    9,  width: 382.50 ms            (391679 S)
Pulse width distribution:
 [ 0] count:    1,  width:  116 us [116;116]    ( 119 S)
 [ 1] count:    8,  width: 39811 us [39811;39812]       (40766 S)
Gap width distribution:
 [ 0] count:    8,  width: 7986 us [7936;8309]  (8178 S)
Pulse period distribution:
 [ 0] count:    1,  width: 8425 us [8425;8425]  (8627 S)
 [ 1] count:    7,  width: 47751 us [47747;47756]       (48897 S)
Pulse timing distribution:
 [ 0] count:    1,  width:  116 us [116;116]    ( 119 S)
 [ 1] count:    8,  width: 39811 us [39811;39812]       (40766 S)
 [ 2] count:    8,  width: 7986 us [7936;8309]  (8178 S)
 [ 3] count:    1,  width: 100001 us [100001;100001]    (102401 S)
Level estimates [high, low]:   8994,     84
RSSI: -2.6 dB SNR: 20.3 dB Noise: -22.9 dB
Frequency offsets [F1, F2]:     -97,      0     (-1.5 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with fixed gap
view at https://triq.org/pdv/#AAB10400749B821F32FFFF82929292929292929355
Attempting demodulation... short_width: 116, long_width: 39811, reset_limit: 8310, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=116,l=39811,r=8310,g=0,t=15878,y=0'
[pulse_slicer_pwm] Analyzer Device
codes     : {9}800
```

```
rtl_433.exe -r g001_433.92M_1024k_open.cu8 -X "n=name,m=OOK_PWM,s=116,l=39811,r=8310,g=0,t=15878,y=0"
rtl_433 version 25.02 branch  at 202502191252 inputs file rtl_tcp RTL-SDR SoapySDR
[Input] Test mode active. Reading samples from file: g001_433.92M_1024k_open.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.077500s
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 9            data      : 800
codes     : {9}800
```

zuckschwerdt, 2025-06-03:
That sample is recorded with one frequency at nearly 0 Hz. The demod is probably thrown off by that.

boemekeld, 2025-06-03:
[g002_433.92M_1024k_FSK_CLOSE_G10.zip](https://github.com/user-attachments/files/20581601/g002_433.92M_1024k_FSK_CLOSE_G10.zip)

With this recording (made with Gain 10), I received the parameters “m=FSK_PCM,s=621,l=621,r=636000” – FSK for the first time! It’s not perfect yet; sometimes it works, other times it doesn’t, and it often repeats. However, I believe we’re close to a solution!

[image]

evilpete, 2025-06-04:
Try this:

[Intelbras.conf](https://github.com/user-attachments/files/20585269/Intelbras.conf.txt)

`rtl_433 -s 1024k -r g002_433.92M_1024k_FSK_CLOSE_G10.cu8 -c Intelbras.conf`

Note that most of the field names are complete f'ing guesses.

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.280527s
model     : Intelbras    count     : 1             num_rows  : 1             len       : 40
data      : 2dd4e2bb55   id        : e2bb
cmd       : CLOSE
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.328228s
model     : Intelbras    count     : 1             num_rows  : 1             len       : 40
data      : 2dd4e2bb55   id        : e2bb
cmd       : CLOSE
```

boemekeld, 2025-06-04:
Same thing, it only works sometimes..

[image]

evilpete, 2025-06-04:
> @evilpete <https://github.com/evilpete> Thanks for that. It works
> occasionally:
>
> image.png (view on web)
>
> However, it only succeeds about once in every ten attempts. I believe the
> issue lies elsewhere now.
>
> I’ve tested different distances and antennas (I’m using an RTL-SDR v4 with
> a dipole antenna), and I also tried a 19 cm 433 MHz antenna, experimenting
> with various gain settings.
>
> The demodulation appears to be highly sensitive to the quality of the
> reception. I’m not sure what else I can do to improve it. My plan is to
> have multiple sensors spread around the house at different distances and
> behind various obstacles.

it's probably not getting the full preamble, send me a few samples of the
"missed" transmission and I'll just to config to be more forgiving.

boemekeld, 2025-06-04:
Sure, here are the samples from two distinct sensors at different distances. I really appreciate your help.

[samples.zip](https://github.com/user-attachments/files/20598727/samples.zip)

boemekeld, 2025-06-05:
I wrote a small js script that can correctly decode all the samples:

```
{
  'g001_433.92M_1024k.cu8': [
    '1010101010101010101010100010110111010100111000101011101101010101'
  ],
  'g002_433.92M_1024k.cu8': [
    '1010101010101010101010100010110111010100111000101011101101010011'
  ],
  'g043_433.9M_1024k.cu8': [
    '1010101010101010101010100010110111010100111000101011101101010101'
  ],
  ...
}
```
https://github.com/boemekeld/fsk-decoder/blob/main/index.js

I’m wondering if I should use it, and if so, how. I was thinking of running it with the -S parameter and monitoring each newly created file. Do you have a better approach to suggest? @zuckschwerdt @evilpete

boemekeld, 2025-06-09:
Thanks, I will test more.