merbanan/rtl_433 #589: Chuango sensor being detected as Acurite rain gauge?

hdemel, 2017-07-15:
Hi, I recently bought a Chuango RF PIR sensor from Amazon (The kind with the extendable metal antenna) and it worked fine and was detected correctly. Subsequently, I bought 5 more and they all are recognized as Acurite Rain Gauges!

Haha, if anyone knows how I might go about troubleshooting this I would appreciate it. 

Detected OOK package    @ 2017-07-15 17:24:08
Analyzing pulses...
Total count:    4,  width:  3057                (12.2 ms)
Pulse width distribution:
 [ 0] count:    1,  width:   440 [440;440]      (1760 us)
 [ 1] count:    1,  width:   959 [959;959]      (3836 us)
 [ 2] count:    1,  width:    39 [39;39]        ( 156 us)
 [ 3] count:    1,  width:    17 [17;17]        (  68 us)
Gap width distribution:
 [ 0] count:    1,  width:  1505 [1505;1505]    (6020 us)
 [ 1] count:    1,  width:    17 [17;17]        (  68 us)
 [ 2] count:    1,  width:    80 [80;80]        ( 320 us)
Pulse period distribution:
 [ 0] count:    1,  width:  1945 [1945;1945]    (7780 us)
 [ 1] count:    1,  width:   976 [976;976]      (3904 us)
 [ 2] count:    1,  width:   119 [119;119]      ( 476 us)
Level estimates [high, low]:   1000,     43
Frequency offsets [F1, F2]:   -7709,      0     (-29.4 kHz, +0.0 kHz)
Guessing modulation: No clue...

rct, 2017-07-15:
First, the acurite rain sensor decoder has a problem with false positives.  It's not enabled by default. So either don't use `-G` to enable all protocols, or only enable the protocol you need with -R. 

I would compare the signal timings from the good device that is being decoded vs. the new devices.  It is possible the tolerances on the timing (widths) of the pulses/gaps is too tight and needs to be relaxed a bit.   

Do you get successful decodes if you move them closer to or farther away from the rtl antenna?

zuckschwerdt, 2017-07-15:
The Chuango looks like an application of the HS1527 chip (the 25 pulses and 20+4 bit scheme), same as the Akhan and Generic remote. But they all have slightly different timing and the modulation is set to exactly match that (`OOK_PULSE_PWM_PRECISE`) so the output can imply the specifc sensor type/brand. That way the exact timing is important. (The datasheet talks about something like crystal oscillator div 256 as base timing or such.)

Also the Chuango only looks at the first row, which should only contain the first sync pulse.
We'd need to fix that someday. You can try this be changing the `0` to `1` or greater in both:
https://github.com/merbanan/rtl_433/blob/master/src/devices/chuango.c#L27
and line 32.

And your signal analysis should look more like this:
```
Total count:   25,  width: 13663		(54.7 ms)
Pulse width distribution:
 [ 0] count:   14,  width:   424 [422;428]	(1696 us)
 [ 1] count:   11,  width:   142 [142;144]	( 568 us)
Gap width distribution:
 [ 0] count:   14,  width:   139 [138;140]	( 556 us)
 [ 1] count:   10,  width:   420 [420;421]	(1680 us)
```

zuckschwerdt, 2017-07-15:
Ah well, forget about changing the row index. Each row is decoded in a fresh bitbuffer due to the PWM_PRECISE demodulation.

hdemel, 2017-07-16:
Is there a resource I can reference to start working on incorporating this
into rtl_433 as a plugin?

zuckschwerdt, 2017-07-16:
Just copy one of the existing devices as template and look at one of the last commits adding a device for all the files that need to be modified.
But what exactly are your findings here? Maybe this can be worked into the existing code? Do you have the timings or a sample of the transmission?

merbanan, 2018-12-03:
This type of signal is best handled by the flex decoder. Closing issue.

closed 2018-12-03 by merbanan