merbanan/rtl_433 #591: Need help implementing my sensor

siriuzwhite, 2017-07-29:
Hi guys,

I made a pull request to the rtl_433_tests repository with my sample set: https://github.com/merbanan/rtl_433_tests/pull/177

```
# rtl_433 -r gfile001.data -a
...
Test mode active. Reading samples from file: gfile001.data
Input format: uint8
*** signal_start = 52804, signal_end = 327683
signal_len = 274879,  pulses = 279
Iteration 1. t: 80    min: 2 (3)    max: 158 (276)    delta 11236
Iteration 2. t: 80    min: 2 (3)    max: 158 (276)    delta 0
Pulse coding: Short pulse length 2 - Long pulse length 158

Short distance: 420, long distance: 1060, packet distance: 4003

p_limit: 80
bitbuffer:: Number of rows: 6 
[00] {47} ff ff ff fd ff fe : 11111111 11111111 11111111 11111101 11111111 1111111
[01] {48} ff bf ff ff ff ef : 11111111 10111111 11111111 11111111 11111111 11101111 
[02] {46} ff ff ff ff ff fc : 11111111 11111111 11111111 11111111 11111111 111111
[03] {46} ff ff ff ff ff fc : 11111111 11111111 11111111 11111111 11111111 111111
[04] {46} ff ff ff ff ff fc : 11111111 11111111 11111111 11111111 11111111 111111
[05] {46} ff ff ff ff ff fc : 11111111 11111111 11111111 11111111 11111111 111111
Test mode file issued 5 packets
```
I'm not sure if this binary representation is decoded correctly, so I maybe need some assist to fine tune the signal.

Any help would be appreciated :-)

zuckschwerdt, 2017-07-29:
The `-A` analyzer gives a clean picture:
```
Pulse width distribution:
 [ 0] count:    4,  width:   257 [255;260]	(1028 us)
 [ 1] count:   42,  width:   147 [140;152]	( 588 us)
Gap width distribution:
 [ 0] count:    4,  width:   246 [246;248]	( 984 us)
 [ 1] count:    1,  width:  1968 [1968;1968]	(7872 us)
 [ 2] count:   21,  width:   462 [438;485]	(1848 us)
 [ 3] count:   19,  width:  1013 [993;1033]	(4052 us)
Pulse period distribution:
 [ 0] count:   25,  width:   592 [503;633]	(2368 us)
 [ 1] count:    1,  width:  2118 [2118;2118]	(8472 us)
 [ 2] count:   19,  width:  1161 [1138;1182]	(4644 us)
```
Also looking at the signal in Audacity gives:
Four sync pulses, 1028 us wide with 984 us gap followed by a 7872 us gap.
Then 40 data pulses, 588 us wide with either 1848 us gap or 4052 us gap.
The packet distance is 16052 us.
This is Pulse Position Modulation (PPM).

The signal has a somewhat high noise floor that might be a problem, otherwise this will give you a decoder:
```
.modulation     = OOK_PULSE_PPM_RAW,
.short_limit    = 3000, // Threshold between short and long gap [us]
.long_limit     = 9000, //  Maximum gap size before new row of bits [us]
.reset_limit    = 17000, // Maximum gap size before End Of Message [us].
```

siriuzwhite, 2017-07-30:
Very nice, the decoded bytes exactly contain my recorded temp and humidity :)
How you got that? I have to create a new .c file in src/devices with your variables, right? And in it use the callback function to extract the information out of the bytes. Anything else I have to note? Need to refresh my c knowledge first ;-)

|  ?        |  ?        |  ?    |  ?     |           T                |   H  |   H  |     ? |
|  :---:    |  :----:  |  :----:    |  :----:   | :----:                      | :----:    |   :----:  |    :----: |
| 0000 | 1111 | 0011 | 0000 | 0101 1100 1110 | 0111 | 0110 | 0001 |
|     |    |      |     | 1486                   | 7       |   7  |      |
|     |    |      |     | 148.6° F - 90 = 58.6° F = 14,7778° C                   | 77 %       |     |      |

siriuzwhite, 2017-07-31:
Any ideas how to get the callback function only triggered once?
Here is what I have so far:

```
#include "rtl_433.h"
#include "util.h"
#include "pulse_demod.h"
#include "data.h"

static int infactory_callback(bitbuffer_t *bitbuffer) {

    bitrow_t *bb = bitbuffer->bb;
    uint8_t *b = bb[0];
    data_t *data;

    if (bitbuffer->bits_per_row[0] == 40) {

      //bitbuffer_print(bitbuffer);

      uint hum1 = b[3] & 0xF;
      uint hum2 = b[4] >> 4;
      uint humidity = hum1 * 10 + hum2;

      uint8_t temp1 = b[2];
      uint temp2 = b[3] >> 4;
      uint16_t temp = (temp1 << 4) | temp2;

      double temp_f = (double)temp / 10 - 90;
      double temp_c = (temp_f - 32)*5/9;

      data = data_make("model",         "",	   DATA_STRING, "inFactory sensor",
          "temperature_c", "Temperature",DATA_FORMAT, "%.02f °C", DATA_DOUBLE, temp_c,
          "humidity",      "Humidity",   DATA_FORMAT, "%u %%", DATA_INT, humidity,
          NULL);
      data_acquired_handler(data);

      //printf("Test: %.1f\n", temp_c);
    }

    return 1;
}

static char *output_fields[] = {
    "model",
    "temperature_c",
    "humidity",
    NULL
};

r_device infactory = {
    .name          = "inFactory",
    .modulation    = OOK_PULSE_PPM_RAW,
    .short_limit   = 3000, // Threshold between short and long gap [us]
    .long_limit    = 5000, //  Maximum gap size before new row of bits [us]
    .reset_limit   = 6000, // Maximum gap size before End Of Message [us].
    .json_callback = &infactory_callback,
    .disabled      = 0,
    .fields        = output_fields
};
```

zuckschwerdt, 2017-08-01:
Perhaps post a handful of raw data, I'll see if I can figure out the checksum.

siriuzwhite, 2017-08-01:
@zuckschwerdt 
```
[00] {40} 8b 80 6b b6 61 : 10001011 10000000 01101011 10110110 01100001 // "id" : 139, "temperature_F" : 82.300, "humidity" : 66
[00] {40} 8b 60 6c 05 71 : 10001011 01100000 01101100 00000101 01110001 // "id" : 139, "temperature_F" : 82.800, "humidity" : 57
[00] {40} 8b 50 6b c5 61 : 10001011 01010000 01101011 11000101 01100001 // "id" : 139, "temperature_F" : 82.400, "humidity" : 56
[00] {40} 8b e0 6b b5 51 : 10001011 11100000 01101011 10110101 01010001 // "id" : 139, "temperature_F" : 82.300, "humidity" : 55
[00] {40} 8b e0 6b b5 51 : 10001011 11100000 01101011 10110101 01010001 // "id" : 139, "temperature_F" : 82.300, "humidity" : 55
[00] {40} 62 60 6b 75 61 : 01100010 01100000 01101011 01110101 01100001 // "id" :  98, "temperature_F" : 81.900, "humidity" : 56
[00] {40} 62 e0 6b b5 71 : 01100010 11100000 01101011 10110101 01110001 // "id" :  98, "temperature_F" : 82.300, "humidity" : 57
[00] {40} 62 f0 6b b5 61 : 01100010 11110000 01101011 10110101 01100001 // "id" :  98, "temperature_F" : 82.300, "humidity" : 56
[00] {40} 62 c0 6b b5 51 : 01100010 11000000 01101011 10110101 01010001 // "id" :  98, "temperature_F" : 82.300, "humidity" : 55
[00] {40} 62 c0 6b b5 51 : 01100010 11000000 01101011 10110101 01010001 // "id" :  98, "temperature_F" : 82.300, "humidity" : 55
[00] {40} bf 10 6b 55 41 : 10111111 00010000 01101011 01010101 01000001 // "id" : 191, "temperature_F" : 81.700, "humidity" : 54
[00] {40} bf 60 6b a5 51 : 10111111 01100000 01101011 10100101 01010001 // "id" : 191, "temperature_F" : 82.200, "humidity" : 55
[00] {40} bf 70 6b a5 41 : 10111111 01110000 01101011 10100101 01000001 // "id" : 191, "temperature_F" : 82.200, "humidity" : 54
[00] {40} bf 50 6b b5 31 : 10111111 01010000 01101011 10110101 00110001 // "id" : 191, "temperature_F" : 82.300, "humidity" : 53
[00] {40} bf 90 6b 65 31 : 10111111 10010000 01101011 01100101 00110001 // "id" : 191, "temperature_F" : 81.800, "humidity" : 53

```