merbanan/rtl_433 #593: Command just quitting on Hop?

coogle, 2017-08-01:
Hello,

I have a series of devices that I need to listen for on different frequencies. Specifically a series of Honeywell sensors and then an Acurite 592TXR..

I thought I could use the -H command to potentially listen to both frequencies but this is what happens:

```
$ rtl_433 -f 344975000 -f 433920000 -R 40 -R 70 -H 5 -F json
Registering protocol [1] "Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning"
Registering protocol [2] "Honeywell Door/Window Sensor"
Registered 2 out of 90 device decoding protocols
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 344975000 Hz.
Signal caught, exiting!

User cancel, exiting...
```

Here's the trouble -- I didn't ^C at all... after 5 seconds (the length of the -H time, if I change it to 1 it happens immediately) the command just dies instead of switching frequencies.

Am I doing something wrong?

zuckschwerdt, 2017-08-01:
This might be related to #85c4e76faa4ba226ccabb97a1a91cd3bde28e7a2
I'll need to have a look.

coogle, 2017-08-01:
I can confirm that by rolling back #530 this bug goes away:

A binary without the changes in #530:

```
$ ./rtl_433 -f 344975000 -f 433920000 -R 40 -R 70 -H 5 -F json
Registering protocol [1] "Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning"
Registering protocol [2] "Honeywell Door/Window Sensor"
Registered 2 out of 90 device decoding protocols
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 344975000 Hz.
Tuned to 433920000 Hz.
Tuned to 344975000 Hz.
Tuned to 433920000 Hz.
Tuned to 344975000 Hz.
Tuned to 433920000 Hz.
{"time" : "2017-08-01 11:46:15", "model" : "Acurite tower sensor", "id" : 9134, "channel" : "A", "temperature_C" : 19.500, "humidity" : 43, "battery" : 0, "status" : 68}
{"time" : "2017-08-01 11:46:15", "model" : "Acurite tower sensor", "id" : 9134, "channel" : "A", "temperature_C" : 19.500, "humidity" : 43, "battery" : 0, "status" : 68}
{"time" : "2017-08-01 11:46:15", "model" : "Acurite tower sensor", "id" : 9134, "channel" : "A", "temperature_C" : 19.500, "humidity" : 43, "battery" : 0, "status" : 68}
Tuned to 344975000 Hz.
Tuned to 433920000 Hz.
Tuned to 344975000 Hz.
```

zuckschwerdt, 2017-08-01:
Maybe moving https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L606 down one or two statements (the `if`s) will help.

coogle, 2017-08-01:
I don't have the bandwidth to mess around with it right now, but feel free to poke me down the road if you haven't found time to fix this and maybe I can step up! I'm just not familiar enough with these interrupts to know immediately the right answer.

zuckschwerdt, 2017-08-01:
Yeah, I made the mess in the first place. I'll fix it. Take my rambling as note-to-self ;)

zuckschwerdt, 2017-08-01:
I can't reproduce this, perhaps my device is switching faster or my rtl-sdr (20170611) is different.
The problem seems to be that `rtlsdr_cancel_async()` is too slow.
Can you test if https://github.com/zuckschwerdt/rtl_433/tree/fix-timer resolves the issue for you?

coogle, 2017-08-01:
How do I check the version of RTLSDR?

I tried that fork and I got the same error (quits instead of hopping)

zuckschwerdt, 2017-08-01:
I didn't see any version markings on my rtl-sdr and looked up my package managers version info.

I added another commit which now cancels a running timer on `rtlsdr_cancel_async()`. Does this  help? Sorry to bother you again, timers can be complicated.

coogle, 2017-08-01:
Just tried master and same problem.

zuckschwerdt, 2017-08-01:
the change is on the fix-timer branch ;)

coogle, 2017-08-01:
Opps!

That said, same thing didn't fix it:

```
~/rtl_433-fixtimer-patch/build/src$ ./rtl_433 -f 344975000 -f 433920000 
-R 40 -R 70 -H 1 -F json -q
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Tuned to 344975000 Hz.
Signal caught, exiting!

User cancel, exiting...
```

zuckschwerdt, 2017-08-01:
Ok, one last idea. Maybe the first rtlsdr_read_async callback after hopping is late? Can you rerun with the latest commit (f6820df)?

coogle, 2017-08-01:
No luck :( So far the only solution has been to remove #530's patches 
completely which made it work just fine.

zuckschwerdt, 2017-08-02:
Last straw: perhaps it's a platform quirk and the timer fires too quickly. I've added a commit that sets the timer to 3 seconds instead of 1. What platform are you running this on, which kernel and distro?

coogle, 2017-08-02:
This is running on a older (i.e. $100 refurb) Dell running Debian 8 and Linux 3.16.0-4-amd64 x86_64

Latest pull did nothing to help. Exact same behavior - program starts, listens on first change for HOP seconds, then signal caught and it bails.

zuckschwerdt, 2017-08-02:
But the abort is 3 seconds after the hop, right? I.e. with `-H 1` you get 4 seconds run time? There are two timer sets (initial and for each callback), both with 3 seconds timeout now. If the abort is immediate then something else is going on.

coogle, 2017-08-02:
That is correct - The total runtime for -H 1 seems to be roughly 4 seconds before it bails.

zuckschwerdt, 2017-08-02:
This suggests that you really don't recieve data for 3 seconds (it should be twice a second normally). Another commit, now the initial timer is 10 seconds and each follow up 20 seconds. Proper rtl-sdr should be able to invoke the callback in a 10 second window. If the abort is at 11 seconds the callback is never called after the hop. That is what the this code actually should prevent: broken rtl-sdr stops delivering callbacks but does not end the async loop. This would happen if you unplug the receiver or it malfunctions.

coogle, 2017-08-02:
Okay I'm an idiot and I am very sorry.

The env I was testing this in doesn't have my usual bash-magic to show what branch I am on in a repo and I've been half-paying-attention testing this the past two days without ever actually being on the fix-timer branch apparently. I have questions about how this has gone on so long (i.e. why did the code base seem to be responding just now to your 3 second delay...), but those really aren't that important when put alongside my own stupid mistake!

I've gone back through your commits on the fix-timer branch and I found that c81d177fcc46160d30f542cde9309e16df64ad64 seems to have done the trick:

```
~/rtl_433-fixtimer-patch$ git checkout c81d177fcc46160d30f542cde9309e16df64ad64
HEAD is now at c81d177... cancel timer on rtlsdr_cancel_async
~/rtl_433-fixtimer-patch$ cd build
~/rtl_433-fixtimer-patch/build$ cmake ../
-- Build type not specified: defaulting to release.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/john/rtl_433-fixtimer-patch/build
~/rtl_433-fixtimer-patch/build$ make clean
~/rtl_433-fixtimer-patch/build$ make
...
[100%] Built target data-test
~/rtl_433-fixtimer-patch/build$ cd src
~/rtl_433-fixtimer-patch/build/src$ ./rtl_433 -f 344975000 -f 433920000 -R 40 -R 70 -H 1 -F json -q
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Tuned to 344975000 Hz.
Tuned to 433920000 Hz.
Tuned to 344975000 Hz.
Tuned to 433920000 Hz.
Tuned to 344975000 Hz.
^CSignal caught, exiting!
Signal caught, exiting!

User cancel, exiting...
```

Sorry about wasting so much time here!

zuckschwerdt, 2017-08-02:
Hehe, no worries. Actually you *did* say that you wouldn't have time to really debug/test this ;)
And the result is solid, `rtlsdr_cancel_async` is too slow exiting the loop. I'll re-check everything, squash the commits and PR.

felmoltor, 2017-08-07:
I have the same behavior with a RTLSDR dongle Yooso (https://www.amazon.com/Yosoo-100KHz-1-7GHz-RTL-SDR-Tuner-Receiver/dp/B00PIBZ76I). 

The binary rtl_433 just quit itself after a few seconds; no ^C pressed at all by me:

```
$ ./rtl_433 -A -t 
[...]
Registered 72 out of 91 device decoding protocols
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
Signal caught, exiting!

User cancel, exiting...
```

How can I quick fix this issue with #530?

Thanks!

zuckschwerdt, 2017-08-07:
The text output from the exit handler is misleading. The abort is intended and is most likely because the receiver had problems.
To verify remove both `alarm(1);` calls, at https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L615 and https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L1304
If that fixes the problem restore the calls but increase the timeout. 1 second should always be enought because the data comes twice per second, but maybe the timer is early or alike. So use 2 or more seconds.