merbanan/rtl_433 #598: Can you use multiple dongles?

coogle, 2017-08-14:
Hello,

I am not sure if this is a bug per-se, but I couldn't find an answer anywhere I looked. Is it possible for RTL_433 to listen to two different frequencies using two different USB radios? I would like to monitor two frequencies without hopping (as I don't want to miss a transmission).

johnjore, 2017-09-12:
Did you try using the -d parameter and run two instances of rtl_433?!?

[-d <RTL-SDR USB device index>] (default: 0)

coogle, 2017-09-12:
I did, and it didn't seem to work. Both would run but I didn't seem to be getting data from both.

rct, 2017-09-12:
On the same system, I've run rtl_fm on one dongle and rtl_433 on another without any apparent issues.

You need to have enough USB bus bandwidth to handle both.  Given that the default sample rate for rtl_433 is fairly low (250Ksps), I'd be surprised if you couldn't run at least two dongles.  

Also if you are using something like an RPi, you need to have enough electrical power to run two RTL-SDRs, which use a 300-400 ma each,   You should also use USB extension cables to get them away from each other (and the RPi) to deal with noise issues.

coogle, 2017-09-12:
I'll give it another shot then - this is running on a desktop server so power and bus should be fine. I will post what I observe here (assuming I didn't just mess up).

coogle, 2017-09-18:
Okay so I've done a little testing and I'm still experiencing the problem. I suppose I might be doing something wrong here, but I don't think so.

What I have are two RTL-SDR dongles plugged in to a desktop machine running Debian 8. 

These are the two instances of rtl433 I am trying to execute, each listens for a different frequency on a different device:

```
/usr/local/bin/rtl_433 -d 0 -f 344975000 -R 40 -R 70 -F json -q
```
```
/usr/local/bin/rtl_433 -d 1 -f 433920000 -R 70 -R 40 -F json -q
```

If I run them one at a time, all is well and it seems to be working. However, if I try to run both at the same time it doesn't work:

One at a time:

```
john@sarah:~/ha$ /usr/local/bin/rtl_433 -d 0 -f 344975000 -R 40 -R 70 -F json -q
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Tuned to 344975000 Hz.
^CSignal caught, exiting!
Signal caught, exiting!
cat
User cancel, exiting...
john@sarah:~/ha$ /usr/local/bin/rtl_433 -d 1 -f 433920000 -R 70 -R 40 -F json -q
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Tuned to 433920000 Hz.
^CSignal caught, exiting!
Signal caught, exiting!

User cancel, exiting...
```

Running one in the background, then trying to run the other:

```
john@sarah:~/ha$ /usr/local/bin/rtl_433 -d 0 -f 344975000 -R 40 -R 70 -F json -q &
[1] 18417
john@sarah:~/ha$ Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Tuned to 344975000 Hz.

john@sarah:~/ha$ /usr/local/bin/rtl_433 -d 1 -f 433920000 -R 70 -R 40 -F json -q &
[2] 18423
john@sarah:~/ha$ Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Tuned to 433920000 Hz.
Failed to submit transfer 31!
^C
john@sarah:~/ha$ Signal caught, exiting!
WARNING: async read failed.

User cancel, exiting...
```

Any thoughts on how I might get this working?

rct, 2017-09-18:
If I'm following along correctly, you are trying to run them under the same terminal session in the background without redirection or piping to some other process?

What happens if you run them in two separate windows, ssh sessions, terminal sessions, etc. without putting them in the background?

Also where did you get librtlsdr from?

coogle, 2017-09-18:
I just tried running the two commands at the same time from two different shells (ssh'd in and this time I didn't run anything in the background) and I got the same error trying to run the second instance.

Honestly I can't remember where I got librtlsdr from at this point (it's been awhile since it was initially set up with the first dongle) but IIRC I just followed the instructions on the rtl_433 site?

rct, 2017-09-19:
So if I understand the problem you are seeing is the 2nd instance gets `Failed to submit transfer 31!` and then it just hangs? 

Try using https://github.com/librtlsdr/librtlsdr  I think it is the most up to date in terms of fixes.

It might be worth test 2 instances with the rtl_test, rtl_fm, or rtl_sdr binaries you get from compiling a newer librtlsdr.

I'm not sure if librtlsdr/librtlsdr has all of keenerds fixes in it.

coogle, 2017-09-20:
It doesn't just hang, it gets a Signal caught and bails as the output describes (the first process keeps running, the second errors out and exits). 

I will try a new librtlsdr when I have a chance and report what I find.

zuckschwerdt, 2017-09-20:
Is this maybe related to #597 ? The 31 USB buffers/transfers rang a bell…

coogle, 2017-09-20:
That was it! A new pull that included that PR and a re-compile fixed my issue and now it works just fine!

skx, 2018-03-10:
I guess that means this issue can be closed now, @coogle ?

zuckschwerdt, 2018-03-10:
thanks @skx

closed 2018-03-10 by zuckschwerdt