merbanan/rtl_433 #600: Help to implement rain gauge from Biltema

Timopen, 2017-08-18:
I have a rain gauge from Biltema.se Art.No: 84-0861.
It sometimes triggers the springfield.c device but not always.
Just like springfield it has six sets of 37 bits and a last set of 36 bits.

Nibble[0] and nibble[1] is the id, changes with every reset.
Nibble[2] first bit is battery (0=OK).
Nibble[3] bit 1 is tx button pressed.
Nibble[3] bit 2 = below zero, subtract temperature with 1024.
Nibble[3](bit 3 and 4) + nibble[4] + nibble[5] is the temperature in Celsius with one decimal.
Nibble[2](bit 2-4) + nibble[6] + nibble[7] is the rain rate, increases  25!? with every tilt of the teeter (1.3 mm rain) after 82 tilts  it starts over but carries the rest to the next round e.g tilt 82 = 2 divide by 19.23 to get mm.
Nibble[8] is checksum, have not figured it out yet. Last bit is sync? or included in checksum?.

Detected OOK package    @ 2017-08-18 00:54:12
Analyzing pulses...
Total count:  266,  width: 228238               (913.0 ms)
Pulse width distribution:
 [ 0] count:  266,  width:   123 [120;134]      ( 492 us)
Gap width distribution:
 [ 0] count:    7,  width:  2195 [2194;2197]    (8780 us)
 [ 1] count:  147,  width:   487 [484;490]      (1948 us)
 [ 2] count:  111,  width:   977 [972;1008]     (3908 us)
Pulse period distribution:
 [ 0] count:    7,  width:  2322 [2318;2330]    (9288 us)
 [ 1] count:  147,  width:   610 [608;616]      (2440 us)
 [ 2] count:  111,  width:  1100 [1096;1129]    (4400 us)
Level estimates [high, low]:  15941,   1222
Frequency offsets [F1, F2]:  -11535,      0     (-44.0 kHz, +0.0 kHz)
Guessing modulation: Pulse Position Modulation with fixed pulse width
Attempting demodulation... short_limit: 732, long_limit: 1009, reset_limit: 2198, demod_arg: 0
pulse_demod_ppm(): Analyzer Device
bitbuffer:: Number of rows: 8
[00] {0} :
[01] {37} 4f 61 05 0e c8 : 01001111 01100001 00000101 00001110 11001 -temp 26.1 -rate 80.60 -bat OK -tx no
[02] {37} 4f 61 05 0e c8 : 01001111 01100001 00000101 00001110 11001
[03] {37} 4f 61 05 0e c8 : 01001111 01100001 00000101 00001110 11001
[04] {37} 4f 61 05 0e c8 : 01001111 01100001 00000101 00001110 11001
[05] {37} 4f 61 05 0e c8 : 01001111 01100001 00000101 00001110 11001
[06] {37} 4f 61 05 0e c8 : 01001111 01100001 00000101 00001110 11001
[07] {36} 4f 61 05 0e c0 : 01001111 01100001 00000101 00001110 1100

Examples:
4f 50 ff 5f 88 : 01001111 01010000 11111111 01011111 10001 -temp  25.5 -rate 71.50		-bat OK  -tx no
14 00 fe 00 f0 : 00010100 00000000 11111110 00000000 11110 -temp  25.4 -rate  0.00 	-bat OK  -tx no
14 00 fb fa 70 : 00010100 00000000 11111011 11111010 01110 -temp  25.1 -rate 13.00		-bat OK  -tx no
14 10 fa 13 d0 : 00010100 00010000 11111010 00010011 11010 -temp  25.0 -rate 14.30		-bat OK  -tx no
14 20 fa 0d 78 : 00010100 00100000 11111010 00001101 01111 -temp  25.0 -rate 27.30		-bat OK  -tx no
14 30 fb 07 70 : 00010100 00110000 11111011 00000111 01110 -temp  25.1 -rate 40.30		-bat OK  -tx no
14 40 fd 01 08 : 00010100 01000000 11111101 00000001 00001 -temp  25.3 -rate 53.30		-bat OK  -tx no
14 41 00 fb b8 : 00010100 01000001 00000000 11111011 10111 -temp  25,6 -rate 66,30		-bat OK  -tx no
14 50 fd 14 18 : 00010100 01010000 11111101 00010100 00011 -temp  25.3 -rate 67.60		-bat OK  -tx no
14 50 fd f5 30 : 00010100 01010000 11111101 11110101 00110 -temp  25.3 -rate 79.30		-bat OK  -tx no
14 61 00 0e 58 : 00010100 01100001 00000000 00001110 01011 -temp  25.6 -rate 80.60		-bat OK  -tx no
14 60 ff ef 78 : 00010100 01100000 11111111 11101111 01111 -temp  26.1 -rate 92.30 		-bat OK  -tx no
14 70 fd 08 a8 : 00010100 01110000 11111101 00001000 10101 -temp  25.5 -rate 93.60 		-bat OK  -tx no
14 01 00 02 10 : 00010100 00000001 00000000 00000010 00010 -temp  25.3 -rate  0.10		-bat OK  -tx no What to do when there is a roundtrip? Mod25 somehow?
14 08 fe 02 a0 : 00010100 00001000 11111110 00000010 10100 -temp  25.4 -rate  0.10 		-bat OK  -tx yes
14 00 fe 02 70 : 00010100 00000000 11111110 00000010 01110 -temp  25.4 -rate  0.10 		-bat OK  -tx no
14 09 02 02 18 : 00010100 00001001 00000010 00000010 00011 -temp  25.8 -rate  0.10		-bat OK  -tx yes
14 01 02 02 c8 : 00010100 00000001 00000010 00000010 11001 -temp  25.8 -rate  0.10 		-bat OK  -tx no
7d 88 ff 00 10 : 01111101 10001000 11111111 00000000 00010 -temp  25.5 -rate  0.00 		-bat Low -tx yes
7d 07 f6 64 d8 : 01111101 00000111 11110110 01100100 11011 -temp  -1.0 -rate  5.20 		-bat OK -tx no
7d 07 de 64 70 : 01111101 00000111 11011110 01100100 01110 -temp  -3.4 -rate 5.20 		-bat OK -tx no
7d 07 8d 64 a8 : 01111101 00000111 10001101 01100100 10101 -temp -11.5 -rate 5.20 		-bat OK -tx no
[gfile086.zip](https://github.com/merbanan/rtl_433/files/1234209/gfile086.zip)

zuckschwerdt, 2017-08-18:
What you see as last bit is most likely a sync pulse (actually the sync for the preceeding packet) that is just picked up as data bit. It should be safe to discard.

Timopen, 2017-08-18:
(Rate + (Rate Mod 25) / 2 * 2048) / 19.23 should fix the roundtrip.

Timopen, 2017-08-18:
I changed springfield.c in my installation, so now it's working. Only need to figure out the checksum part because it's triggered by my prologue temp sensor now as well.

#include "rtl_433.h" 
#include "data.h" 
#include "util.h"

// Actually 37 bits for all but last transmission which is 36 bits
#define	NUM_BITS	36
static int springfield_callback(bitbuffer_t *bitbuffer) {
	int ret = 0;
	char time_str[LOCAL_TIME_BUFLEN];
	int row;
	int cs;
	int i;
	int nibble[NUM_BITS/4+1];
	int sid, battery, rain, transmit, channel, temp;
	float tempC, rainrate;
	//int moisture, uk1;
	int checksum;
	data_t *data;
	unsigned tmpData;
	unsigned savData = 0;
	local_time_str(0, time_str);
	for(row = 0; row < bitbuffer->num_rows; row++) {
		if(bitbuffer->bits_per_row[row] == NUM_BITS || bitbuffer->bits_per_row[row] == NUM_BITS + 1) {
			cs = 0;
			tmpData = (bitbuffer->bb[row][0] << 24) + (bitbuffer->bb[row][1] << 16) + (bitbuffer->bb[row][2] << 8) + bitbuffer->bb[row][3];
			if (tmpData == 0xffffffff) {
				continue; // prevent false positive checksum
			}
			for(i = 0; i < (NUM_BITS/4); i++) {
				if((i & 0x01) == 0x01)
					nibble[i] = bitbuffer->bb[row][i >> 1] & 0x0f;
				else
					nibble[i] = bitbuffer->bb[row][i >> 1] >> 0x04;
				if(i < 8) cs ^= nibble[i];
			}
			cs = (cs & 0xF);
			checksum = nibble[8];
			//if(checksum == cs && tmpData != savData) {
			if(tmpData != savData) {
				savData = tmpData;
				sid = (nibble[0] << 4) + nibble[1];
				battery = (nibble[2] >> 3) & 0x01;
				transmit = (nibble[3] >> 3) & 0x01;
				//channel = (nibble[2] & 0x03) + 1;
				temp = (((nibble[3]&0x03) << 8) + (nibble[4] << 4) + nibble[5]);
				if(nibble[3]&0x04) temp = temp - 0x1000;
				tempC = temp / 10.0;
				//moisture = nibble[6];
				//uk1 = nibble[8]; /* unknown. */
				rain = (((nibble[2]&0x07)<<8) + (nibble[6] << 4) + nibble[7]);
				int rest = rain % 25;
				if(rest % 2){
					rain + ((rest / 2) * 2048);
				}
				else {
					(rain + ((rest+1) / 2) * 2048) + 24576;
				}
				rainrate = rain/19.23F;
				data = data_make(
					"time", "", DATA_STRING, time_str,
					"model", "", DATA_STRING, "Biltema rain gauge",
					"sid", "SID", DATA_INT, sid,
					"channel", "Channel", DATA_INT, channel,
					"battery", "Battery", DATA_STRING, battery ? "LOW" : "OK",
					"transmit", "Transmit", DATA_STRING, transmit ? "MANUAL" : "AUTO",
					"temperature_C","Temperature", DATA_FORMAT, "%.01f C", DATA_DOUBLE,tempC,
					"rainrate", "Rainrate/hour", DATA_FORMAT, "%.02f mm/h", DATA_DOUBLE,rainrate,
					//"moisture", "Moisture", DATA_INT, moisture, // // "uk1", "uk1", DATA_INT, uk1,
					NULL);
				data_acquired_handler(data);
				ret++;
			}
		}
	}
	return ret;
}
static char *output_fields[] = {
	"time",
	"model",
	"sid",
	"channel",
	"battery",
	"transmit",
	"temperature_C",
	"rainrate",
	NULL
};
r_device springfield = {
	.name = "Biltema rain gauge",
	.modulation = OOK_PULSE_PPM_RAW,
	.short_limit = 2000,
	.long_limit = 4000,
	.reset_limit = 9200,
	.json_callback = &springfield_callback,
	.disabled = 0,
	.demod_arg = 0,
	.fields = output_fields
};