merbanan/rtl_433 #605: Include support for devices on other frequencies

brendanheyu, 2017-10-02:
In Australia and some other countries, we also use devices that utilise the 315mhz frequency. Could this library be extended to support them also?

panarom, 2017-10-02:
just pass the -f 315000000 argument to the invocation: `rtl_433 -f 315000000`.  I'm currently decoding signals on 315MHz, 345MHz, & 433MHz (not simultaneously).
Out of curiosity, are you trying to detect a wireless doorbell?

brendanheyu, 2017-10-02:
I've got a doorbell and a floating pool thermometer. Am looking to see if I can pick up a simple weather station and pull these values back into Home Assistant.

Breakerz, 2017-11-01:
It’s will be nice to have the « perfect guide to decode rf » i got also devices i would like to decode.

gtoal, 2017-12-11:
I have a mixture of 315MHz and 433MHz devices (Bunker Hill motion detectors from Harbor Freight are on the default 433, whereas the PIR sensor in the "SAM C5 Expansion Pack" is a Chuango on 314880000) - any possibility of either the software accessing both simultaneously, *or* switching fast enough to sample both without losses?  They're both repeated often enough it ought to be possible to listen for 2 transmit periods to be sure to detect at least one of the many transmissions and then switch to the other frequency...?

zuckschwerdt, 2017-12-12:
https://github.com/merbanan/rtl_433/blob/master/README.md#running
try `[-H <seconds>] Hop interval for polling of multiple frequencies (default: 600 seconds)`
with multiple `[-f <frequency>] [-f...] Receive frequency(s) (default: 433920000 Hz)`.

gtoal, 2017-12-12:
I saw that option but had not considered it because the parameter was in seconds, and to be able to detect on either frequency wouldn't it have to switch frequencies at millisecond or at least centisecond intervals?  Some of these devices just send 3 copies of short transmissions, well under 1 second total.

Or does the parameter allow decimal parts of a second to be specified?  That had not occurred to me until now.

I'll certainly give it a try though in case I've misunderstood the model.

zuckschwerdt, 2017-12-12:
The rtl receivers are known to tune frequencies very fast. However it is more likely you'll end up truncating signals on both frequencies and end up with nothing at all.
I if 100% reception on too bands is important, I'd just go for two receivers.
There is new option `[-E] Stop after outputting successful event(s)` which might be close to what you want. Recieve on one band, then after each signal cancel and restart on the other band.
If you try this, please report back. I might consider implementing a "switch after each signal" option.

gtoal, 2017-12-12:
I'll have to get back to you on this after building a linux version later tonight.  I was running on a Windows portable and it turns out the pre-built version I have here doesn't have the -H option - must be older code.  (It's rtl433_win64_11112016 - I tried the other prebuilt version I could find, and it didn't receive anything for me at all)

I do have a couple of spare RTL sticks, and a two-pi solution would work OK, but I would like to see  if I can make this work on a single device, preferably with one stick though I could definitely live with using two. (They just need a short extension cable as two sticks won't fit next to each other in my USB ports)

I realise that if the frequencies are too far apart you can't take advantage of the SDR's ability to receive multiple frequencies at once and I don't know if 315 and 433 are close enough to allow that, but there are devices which are quite close in frequency where it definitely would work (eg 433715000 and 433920000, or 313800000 and 315000000).  Is it a possibility that the code will ever receive on two frequencies at once?

gtoal, 2017-12-12:
> There is new option [-E] Stop after outputting successful event(s) which might be close to what you want. Recieve on one band, then after each signal cancel and restart on the other band.
If you try this, please report back. I might consider implementing a "switch after each signal" option.

That scheme won't work unless you know the two devices alternate.  But something like a motion sensor is unpredictable, so you would end up always waiting for one motion sensor that never triggers...

zuckschwerdt, 2017-12-13:
> SDR's ability to receive multiple frequencies at once and I don't know if 315 and 433 are close enough to allow that

You are already receiving multiple frequencies: 250k (250000) -- the sample rate. At a **center** frequency of 433920000 you will receive 433.67M - 434.17M.
(433M-315M =~ 120M -- not impossible but the rtl is limited to 3.2M at best.)

merbanan, 2017-12-13:
@gtoal in the case where you listen for temperature readings it is perfectly fine to cycle through the frequency's at 10 minute intervals. But if you are waiting for unpredictable events then this solution is not applicable and you would need one more device.

closed 2017-12-30 by zuckschwerdt