merbanan/rtl_433 #610: rtl_433 only works once. Need to pull and re-insert dongle to run again

james-fry, 2017-10-11:
Hi,

I using the latest rtl_433 that built today.
I am seeing a strange problem where it works the first time it executes after the RTL dongle is inserted, however any subsequent attempt to execute I get this:

```
Registered 91 out of 91 device decoding protocols
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
Signal caught, exiting!

User cancel, exiting...
```

i.e. immediate exit of rtl_433.
Pulling out and re-inserting the DVB dongle clears this issue - maybe because the usb device number increments when I do this?

from syslog:
Oct 11 22:20:44 hassio kernel: [86329.258326] usb 1-1: USB disconnect, device number **5**
Oct 11 22:20:50 hassio kernel: [86335.326443] usb 1-1: new high-speed USB device number **6** using ehci-pci
Oct 11 22:20:50 hassio kernel: [86335.755231] usb 1-1: New USB device found, idVendor=0bda, idProduct=2838
Oct 11 22:20:50 hassio kernel: [86335.755234] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Oct 11 22:20:50 hassio kernel: [86335.755236] usb 1-1: Product: RTL2838UHIDIR
Oct 11 22:20:50 hassio kernel: [86335.755237] usb 1-1: Manufacturer: Realtek
Oct 11 22:20:50 hassio kernel: [86335.755238] usb 1-1: SerialNumber: 00000001

I think its a software issue in either rtl_433 or in rtl_sdr because an older version that I found in this docker image works fine: https://hub.docker.com/r/sysrun/rpi-rtl_433/
(It was built a year ago according to docker hub - you can see support for 67 devices vs the current 91 from above)

```
Registered 67 out of 67 device decoding protcols
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
2017-10-11 21:34:05 :   CurrentCost TX
        Device Id:       3063
        Power 0:         512 W
        Power 1:         0 W
        Power 2:         0 W
2017-10-11 21:34:11 :   CurrentCost TX
        Device Id:       3063
        Power 0:         508 W
        Power 1:         0 W
        Power 2:         0 W
```

Any ideas why this is happening, and how to resolve?

james-fry, 2017-10-28:
I bought another rtl dongle with a different tuner (R820T rather than FC0012) and tried again.
I get same result - the 1st time I run rtl_433 it works great... but the second/subsequent exections I get the following:

> ...
> Registered 72 out of 91 device decoding protocols
> Found 1 device(s):
>   0:  Realtek, RTL2832UHIDIR, SN: 00000001
> 
> Using device 0: Generic RTL2832U OEM
> Found Rafael Micro R820T tuner
> Exact sample rate is: 250000.000414 Hz
> [R82XX] PLL not locked!
> Sample rate set to 250000.
> Bit detection level set to 0 (Auto).
> Tuner gain set to Auto.
> Reading samples in async mode...
> Tuned to 433920000 Hz.
> Signal caught, exiting!
> 
> User cancel, exiting...

The only way to resolve the failures is to pull out the tuner dongle and re-insert it.

zuckschwerdt, 2017-10-28:
Maybe #593 (i.e. the change in #530) is related to this. Perhaps you can try to remove the `alarm(1);` calls. The underlying cause would be some trouble with the receiver or the involved libs though.

james-fry, 2017-10-28:
Thanks - I'll have another look at those (I saw them from searching prior to raising this issue).

james-fry, 2017-10-28:
That fixed it! 
I commented out the alarm(1) calls below and recompiled.
https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L615 and https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L1304

I also noticed that firing up the old version of rtl433 in docker also "cleared" the issue and I could re-run with latest without getting the Signal caught, exiting! User cancel, exiting...
It would be interesting to understand what is the root cause of this. My money is on rtlsdr rather than tuner.

Thanks for the support!

zuckschwerdt, 2017-10-29:
It would interesting to see how many seconds the device needs to recover (i guess that's also what running the old version does for you). The second call (L.1304) is the initial timer, could be that using say `3` would resolve the issue too? Maybe you can experiment with that.
Without those timers rt_433 will just keep running, even when you unplug the receiver...

iceman73, 2018-04-20:
It looks like setting Alarm(3) from Alarm(1) has solved this problem, I am running with this change for over 5 hrs now. I will update if this is the final fix.

iceman73, 2018-04-20:
Looks like this might be it. I just hit almost 14 hrs and still stable, without this change I got at most 2 hrs before the failure, will report back after 24 hrs.

Chris

iceman73, 2018-04-22:
48 hrs and still stable this is the fix I will issue a pull request

ay-kay, 2018-05-03:
Same issue here, works only once, even with the latest version. On each following try, the following output is received:

```
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433740000 Hz.
Allocating 15 zero-copy buffers
Async read stalled, exiting!
WARNING: async read failed (-5).
```

I already tried to increase the alarm timer values, however, without success so far. Any ideas?

zuckschwerdt, 2018-05-04:
There must be a root cause why the receiver dongle is so unreliable. Usually we expect an async read 4 times per second (at least, lowest sample rate). Not receiving data for seconds means the connection is stalled. Without the `alarm` there won't be any noticable issue but there won't be any data either -- the program would be "dead".
I'm considering to raise the timeout to 2 or 3 seconds, but after that I don't think the connection will just recover, better to stop the program.

james-fry, 2018-05-04:
I tried two different dongles (diff manufacturer, different tuner) and got same result in both cases.
Admittedly these are cheapo ex-DVB devices (less than 10$) rather than more expensive devices specifically designed for SDR.
I dont have ANY issues with the devices when in operation. Rock solid for months uptime.
(I am using as a rtl433 bridge for RF sensors into home automation)
The only problem is when rtl433 is stopped and try to restart. Restart fails and the USB dongle needs to be pulled and re-inserted. then it will run one time again.
Removing the alarms stopped this problem and the solution is fully working - def there is data.

zuckschwerdt, 2018-05-04:
Is that on linux? Can you see what dmesg (or alike) show? I'd guess the device is in invalid state, a reset is needed (similar to replugging) and that reset triggers the timeout, maybe.

merbanan, 2018-12-04:
Unfortunately a combination of unreliable usb ports, flaky rtl-sdr dongles and kernel bugs might cause issues for rtl_433. The solution to that is to make sure every component is as updated as possible. But still things may fail so for the best possible experience design the setup with that in mind.

We can't do much from the rtl_433 end.

closed 2018-12-04 by merbanan