merbanan/rtl_433 #611: RTL_433 Chuango DWC-102

alcadic, 2017-10-12:
Hello all,

I try to config my DWC-102 for what RLT listen it.
I changed in chuango.c line 88 (.pulse_tolerance	= 40,) to 35 (in reference  to: https://github.com/merbanan/rtl_433_tests/commit/d3662cfbe6d43440bb1695a42b67b8f5e3aee72f) but I get nothing.
Also, I dont know if problem is the Zone Configuration about my chuango device, I don't know if it's configuration is in "Normal Zone". 

Somebody can help me??

Thanks in advance.

alcadic, 2017-10-21:
hi all!!

I just execute rtl_433 -A -R12 (Oregon Scientific Weather Sensor) and I had see that library detect this device as Oregon device and not as Chuango device....

I executed rtl_433 -A -R12 five times:

1-------------------

Detected OOK package    @ 2017-10-21 22:06:59
Analyzing pulses...
Total count:  104,  width: 69996                (280.0 ms)
Pulse width distribution:
 [ 0] count:   47,  width:   497 [486;509]      (1988 us)
 [ 1] count:   56,  width:   223 [189;256]      ( 892 us)
 [ 2] count:    1,  width:    16 [16;16]        (  64 us)
Gap width distribution:
 [ 0] count:   39,  width:    39 [33;48]        ( 156 us)
 [ 1] count:    4,  width:  3978 [3965;4004]    (15912 us)
 [ 2] count:   52,  width:   313 [257;348]      (1252 us)
 [ 3] count:    5,  width:    29 [25;32]        ( 116 us)
 [ 4] count:    2,  width:    51 [50;52]        ( 204 us)
 [ 5] count:    1,  width:    14 [14;14]        (  56 us)
Pulse period distribution:
 [ 0] count:   97,  width:   537 [532;541]      (2148 us)
 [ 1] count:    4,  width:  4214 [4214;4215]    (16856 us)
 [ 2] count:    2,  width:   268 [263;273]      (1072 us)
Level estimates [high, low]:  14288,   2192
Frequency offsets [F1, F2]:   14301,      0     (+54.6 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with startbit/delimiter
Attempting demodulation... short_limit: 119, long_limit: 360, reset_limit: 4005, demod_arg: 0
pulse_demod_pwm_ternary(): Analyzer Device
bitbuffer:: Number of rows: 2
[00] {75} ec 83 15 76 41 8a bb 20 c5 40
[01] {28} ec 83 15 70 : 11101100 10000011 00010101 0111

2--------------------------

Detected OOK package    @ 2017-10-21 22:09:50
Analyzing pulses...
Total count:    6,  width:  3121                (12.5 ms)
Pulse width distribution:
 [ 0] count:    2,  width:   222 [217;228]      ( 888 us)
 [ 1] count:    4,  width:   456 [438;466]      (1824 us)
Gap width distribution:
 [ 0] count:    2,  width:   315 [308;322]      (1260 us)
 [ 1] count:    3,  width:    73 [71;79]        ( 292 us)
Pulse period distribution:
 [ 0] count:    5,  width:   536 [535;539]      (2144 us)
Level estimates [high, low]:  13909,   1773
Frequency offsets [F1, F2]:   15518,      0     (+59.2 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with fixed period
Attempting demodulation... short_limit: 339, long_limit: 323, reset_limit: 323, demod_arg: 0
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {6} a0 : 101000

3-------------------------------

Detected OOK package    @ 2017-10-21 22:11:54
Analyzing pulses...
Total count:  299,  width: 200994               (804.0 ms)
Pulse width distribution:
 [ 0] count:  132,  width:   470 [451;485]      (1880 us)
 [ 1] count:  167,  width:   214 [198;232]      ( 856 us)
Gap width distribution:
 [ 0] count:  131,  width:    66 [54;85]        ( 264 us)
 [ 1] count:  156,  width:   322 [304;341]      (1288 us)
 [ 2] count:   11,  width:  4003 [3996;4016]    (16012 us)
Pulse period distribution:
 [ 0] count:  287,  width:   537 [532;541]      (2148 us)
[... 83 lines cut ...]
[07] {2} 40 : 01
[08] {4} 10 : 0001
[09] {3} 20 : 001
[10] {1} 80 : 1
[11] {2} 40 : 01
[12] {1} 80 : 1
[13] {1} 80 : 1
[14] {1} 80 : 1
[15] {1} 80 : 1
[16] {3} 20 : 001
[17] {1} 80 : 1
[18] {1} 80 : 1
[19] {2} 40 : 01
[20] {2} 40 : 01
[21] {2} 40 : 01
[22] {4} 10 : 0001
[23] {3} 20 : 001
[24] {3} f0 : 111
... Maximum number of rows reached. Message is likely truncated.

---------------------------------------------------------

any idea ??

thanks in advance!!

merbanan, 2018-12-07:
No samples signals referenced in the issue, closing.

closed 2018-12-07 by merbanan