merbanan/rtl_433 #616: FineOffset (wh1081?) funny readings

marianowadel, 2017-10-26:
Hi! First of all, I'd like to thank the creator and every collaborator for this project. It's simply amazing, and the possibilities are endless!
I have been using rtl_433 at my parents' to report my wh1081 readings for almost a year now, without issues.
Now I installed another RPi at my grandmothers' (about 80km away from home) running the latest rtl_433 build to sniff on a neighbour's PWS ([another wh1081 I believe](https://photos.app.goo.gl/pGHFMwMRtGvyYDFs2)). The station is detected automatically as 'Fine Offset Electronics WH1080/WH3080 Weather Station'. The problem is I'm having funny readings. 
Here is the decoded JSON from a package:

`{u'msg_type': 0, u'battery': u'LOW', u'gust': 11.016, u'direction_deg': u'180', u'rain': 365.7, u'humidity': 72, u'temperature_C': 311.3, u'direction_str': u'S', u'time': u'2017-10-26 11:14:37', u'model': u'Fine Offset Electronics WH1080/WH3080 Weather Station', u'speed': 4.896, u'id': 107}`

Of course it's not 311.3°C out there, otherwise my grandmother would be martian. This doesn't change during the day, nor does the relative humidity (it's stuck at 72%). Wind direction, gust and speed work fine.
My first guess was that this is related to the low battery condition in the transmitter, but I want to be absolutely sure about this. To compare, my wh1081 has reported low battery for about six months and works fine. No funny readings.
I tried running an older rtl_433 build (the one that I use to report my wh1081 readings, before wh3081 support was implemented) in the new setup and, interestingly enough, the station is detected as Esperanza EWS (which is impossible because I think it doesn't have wind sensors), not FineOffset wh1081. 
Although I'm almost positive that this error is related to the low battery condition, I would like to know if there's anything else I can do to confirm the problem is transmitter's fault. Could this errors be related to the [recently added support for wh3081](https://github.com/merbanan/rtl_433/commits/master/src/devices/fineoffset_wh1080.c)? 

Here's `rtl_433 -a -t` output: [gfile001.data.zip](https://github.com/merbanan/rtl_433/files/1418201/gfile001.data.zip)

Many thanks in advance.

Cheers!

marianowadel, 2018-04-29:
So nobody else with this issue? I'm having this problem with a different RPi, different WH1081, and different SDR. I have no idea how to debug this.
Thanks again

zuckschwerdt, 2018-04-29:
You could compare the sample data you got with samples in the test repo, e.g. https://github.com/merbanan/rtl_433_tests/tree/master/tests/fineoffset_wh1080
If those basically match work though the algorithm, see e.g. https://github.com/merbanan/rtl_433/blob/master/src/devices/fineoffset_wh1080.c#L396

marianowadel, 2018-04-29:
@zuckschwerdt thank you for your response. Sorry, I don't know how to open or compare .cu8 files. Here's an example of what I'm recieving now:
[g001_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/1959149/g001_433.92M_250k.cu8.zip)
What's most misleading is that this was working properly, and that the console is working ok. I'll try replacing the batteries (I can do it here).
Thanks again for your help.

zuckschwerdt, 2018-04-29:
Use http://triq.net/iqs to compare visually. With Audacity you can then dig into the signal waveform. Use import raw with unsigned 8-bit 2 channel.

ovrheat, 2018-05-23:
Hi, sorry all for this late answer...
As far as I can tell there are many models of these cheap weather stations, each one targeted for some different world zone market. Moreover there are several versions for each model and there is no way, from a wireless point of view, to recognize each model.... 
An example is WH1080 vs WH3080 weather data: they are identical, there is no different ID to distinguish one from another. There are also -at least- 2 different protocols versions for each model, with different packet length (87 and 88 bits) and -moreover- I have also seen in some forum that some particular version looks different also in the datetime part, so datetime decoding for that model is completely messed up.
With this in mind, apart from the good @zuckschwerdt advises, you could upload your samples in the test repository, in a separate directory, putting also a .txt file with your readings on the LCD display for each file, it would be very useful to try to cope with your model. 
Thank you.

merbanan, 2018-12-08:
I have been looking at the signals with regards to the raw view before demodulation and there are no reliable parameters (signal length, preamble, pulse lengths, etc) that can be used to classify the signal. They seem to all use the same transmitter with difference in the actual payload placement. It seems that the preamble and checksum is the same (start and end).

So to support this type of device we need a way to separate the different payloads based on contents. And to do that we need a lot of sample signals and their corresponding readout values from the official receiver. So issue closed and feel free to follow the new device guide with regards to test signal data.

https://github.com/merbanan/rtl_433#supporting-additional-devices-and-test-data

closed 2018-12-08 by merbanan