merbanan/rtl_433 #617: W-Mbus support

bilbolodz, 2017-10-26:
Any chance for Wireless MBus protocol? It's commonly used in Europe in Electricity/Water/Gas meters.
Some documents/projects:
https://github.com/jmichelp/gr-wmbus
https://en.wikipedia.org/wiki/Meter-Bus

merbanan, 2017-10-26:
No test signals == no chance that the protocol gets supported.

reaper7, 2018-11-23:
Hi,
I'm trying to use protocol 104 w-mbus T/C,
but I still get a decoding error:
```
pi@rpi:~ $ rtl_433 -D -R 104 -f 868900000 -s 1600000
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433.conf"...
Trying conf file at "/etc/rtl_433.conf"...
Registering protocol [104] "Wireless M-Bus, Mode C&T, 100kbps (-f 868950000 -s 1200000)"
Registered 1 out of 112 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
[R82XX] PLL not locked!
Sample rate set to 1600000 S/s.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 868.900MHz.
M-Bus: Mode T
Experimental - Not tested
Decode 0, 80
Decode D
Decode 31
Decode 1, 80
Decode 32
Decode D
Decode 2, 80
Decode 31
Decode 1A
Decode 3, 80
Decode 15
Decode 2D
M-Bus: Decoding error
M-Bus: Mode T
Experimental - Not tested
Decode 0, 80
Decode 9
Decode 31
M-Bus: Decoding error
M-Bus: Mode T
Experimental - Not tested
Decode 0, 80
Decode 9
Decode 31
M-Bus: Decoding error
M-Bus: Mode T
Experimental - Not tested
Decode 0, 80
Decode 9
Decode 31
M-Bus: Decoding error
M-Bus: Mode T
Experimental - Not tested
Decode 0, 80
Decode 9
Decode 31
M-Bus: Decoding error
M-Bus: Mode T
Experimental - Not tested
Decode 0, 80
Decode 9
Decode 31
M-Bus: Decoding error
```
I checked before if the data is broadcast with the same frequency/sample rate and rtl_wmbus correctly read data (manufacteur and meters number hidden!):
```
rtl_sdr -f 868900000 -s 1600000 - 2>/dev/null | rtl_wmbus 2>/dev/null
T1;1;1;2018-11-23 06:52:38.000;17;13;nnnnnnnn;0x4e44ppppnnnnnnnn3077a3e004005b32fcfffd3dfd10282dbc6ac116c0e4dc521b4635b7c846740de51d13a8dcc1c5e54aae65324dc7c6e73d6070e6d75d2aeb913177a7c14700afb74e8bae5feba
T1;1;1;2018-11-23 06:52:43.000;109;65;nnnnnnnn;0x4e44ppppnnnnnnnn3077a3a004005b19dc6f5c0ed68bd8202e6eeb52188505d7e5c1e3bb9e287cdb57a767306f689e287bd9aac19108fb2f779dc6172ad9476ba3650f4048adc890459a3faf76542
T1;1;1;2018-11-23 06:52:44.000;138;145;nnnnnnnn;0x4e44ppppnnnnnnnn3077a240040056e226ea47a860f9e7ce9e792a2786d1f1fac87bc7d38a50fe9558c841f9c4697bfc70cba717cefdb24ec709f2d382f93b0eb156fb8623084ef115163d6973106
T1;1;1;2018-11-23 06:52:45.000;33;19;nnnnnnnn;0x4e44ppppnnnnnnnn3077a1e0040053f14ee750acba21e7cba54ef4905b3af145e77f9599133d15aaf721976b93f8c7f4f50d1ee51d67030b533662e4b12c0f5dadd1fb90c0f7171efa2055a632da3
```

looks like something is wrong with received data when it's checked in m_bus_decode_3of6_buffer
but I don't know what...
Which part of the data this function checking ?
Data looks good, len, meter type, manufacteur, meter number and payload.

How can I help You with improve decode ?
I can send You raw data for test's

merbanan, 2018-11-23:
Record a few samples signals and submit them to the rtl_433_tests repository.

reaper7, 2018-11-23:
done
https://github.com/merbanan/rtl_433_tests/pull/246

merbanan, 2018-11-23:
@vestom any chance for a look at the samples?

merbanan, 2018-11-23:
@reaper7 thanks

reaper7, 2018-12-28:
@vestom - any chance for a fix ?
How can I help?

kidpixo, 2019-02-17:
I'm interested too for some Heat Allocator. 
They also show ouput with rtl_wmbus.
Is this working?

Thanks for rtl_433!

antonmeyer, 2019-02-28:
Hi,
just to have mentioned it, there is a decoder working for me:
https://github.com/xaelsouth/rtl-wmbus

But yours looks much more promissing

reaper7, 2019-02-28:
@antonmeyer - can You share code?

antonmeyer, 2019-02-28:
https://github.com/antonmeyer/RFM69_MBus/blob/master/misc/t1_c1_packet_decoder.h

you need to replace the file from the origin

reaper7, 2019-02-28:
@antonmeyer - but this file is for rtl-wmbus project? rtl-wmbus works for me as-is.
I am still waiting for a comprehensive solution with rtl_433, which I use for other home devices.
Only reading the water meters via w-mbus is missing

merbanan, 2019-10-13:
https://github.com/merbanan/rtl_433/tree/feat-fsk_demod_new

Try this branch and see if you get any messages.

reaper7, 2019-10-14:
looks promising! tnx!

I haven't analyzed the data yet, but the meter numbers / manufacturer are correct

[image]

merbanan, 2019-10-14:
Can you send me a signal sample ? I guess you don't want to add it to the samples repo but can I have a few files to play with locally?

reaper7, 2019-10-14:
Of course I can send for private use.
Please give me the parameters to set for rtl_433
and email address

merbanan, 2019-10-14:
Add -s 1000k -S all and send a few files to banan@ludd.ltu.se.

reaper7, 2019-10-14:
done

merbanan, 2019-10-14:
Do you have the decryption key for this?

reaper7, 2019-10-14:
PM

reaper7, 2019-11-01:
current git version support my mbus meters,
tnx @merbanan for your hard work.

merbanan, 2019-11-01:
Yeah, you need a high sample rate (-s 1600k) for it to work well.

merbanan, 2019-11-01:
Better wmbus support is planned. Ie actual decoding.

reaper7, 2019-11-01:
tested also with two freq:
rtl_433 -R 55 -R 104 -f 433920000 -f 868900000 -s 1600000 -g 35 -H 60 -T 120
works well!

merbanan, 2019-11-04:
Wireless mbus is now supported by default for mode t. There are several issues left but those issues have to be raised for those specifics sensors.

closed 2019-11-04 by merbanan