merbanan/rtl_433 #621: Schrader TPMS protocol (additions)

hkazu, 2017-11-05:
I've got 4 of those sensors on e-bay, but with slight protocol difference (FCC ID: MRXEG53MA4), and made some changes in the schraeder.c file after some captured signal observations and reading the fcc technical report.

What is remaining is some interpretation of the flags (battery, state fags). Those sensors report a state flag to differentiate distinct modes between spinning wheel, stopped but with pressure difference climbing/declining, etc...

The reported temperature seems to be in degrees F units. Initial values agrees as compared to an external thermometer measuring ambient temperature.

The pressure seems to be in 25 mbar units as the original sensor (no changes). The sensors are not installed in a car yet, so I've put it inside a normal bicycle rubber tube cut in half (the sensor stem must be removed and replaced by an ~23cm wire, because it is its antenna, in order to fit inside the tube), inflated it and measured with a digital pressure calibration tool I got to compare with reported values. The sensor values are reported during pressurization/de-pressurization or you also can spin the wheel or use a tpms learning tool.

The CRC (checksum) is basically the sum of all data bytes modulus 256. This occupies the last field.

Those sensors can be forced to enter in learning mode by using a tpms leaning tool. The tool emits a signal which consists of an 125khz continuous sine wave magnetic pulse. I've assembled a breadboard circuit using a 4mhz crystal, hex inverter, binary divider, audio amp ic and a coil. Which works at close range. The learning mode reports comes in 16 packets (frame) instead of 6 frames in normal use.

The report signal frequency is 315mhz, decoded with OOK_PULSE_MANCHESTER_ZEROBIT, short_limit 120, as with previous sensor.

The battery flags (if any) could be detected by inducing low battery condition, but the sensor must be destroyed.

If anyone can review the simple code additions and willing to make some further testing/refinements in real field condition, I'm glad to submit a pull request as modest contribution. Thx.

merbanan, 2017-11-05:
If you send some test signals I will review the changes. I have a sensor also that I think is based on the schraeder one.

zuckschwerdt, 2017-11-05:
I'm also very interested. Your testbed sounds awesome!

hkazu, 2017-11-05:
Ok, I've put some test signal files and corresponding device source on google drive because its easier to me:

[tpms test signal files](https://drive.google.com/drive/folders/0ByPsmMWYzDTeYjZneTZSRVJRM3c?usp=sharing)

1) The *.pdf are device files from FCC.
 
2) The SDRSharp_20171027_132615Z_315250000Hz_IQ.wav is a test signal recorded with sdr-sharp. Can be opened in audacity.

3) The *.complex are signal files captured with [urh](https://github.com/jopohl/urh) in ambient pressure which means zero pressure values for each of 4 different sensors. But in urh I was only able to roughly observe the general data format.

4) The schraeder_EG53MA4.c file is the rtl_433 device file I modified to be used for decoding the signals. You can review it and add to the repo, if you judge okay. Modify as necessary.

5) The *.gfile are bust capture in one de-pressurization session captured with rtl_433 already with modified schrader decoder above. For example, can be decoded with `rtl_433 -f 314868000 -s 1000000 -R 60 -r gfile018.data`. Update the device number as needed.

I'm not very sure of the meaning of the rest of r_device parameter I hope you can deduce from the signal files.
 
Thx

merbanan, 2018-12-09:
Issue solved, closing.

closed 2018-12-09 by merbanan

chuongnh98, 2023-09-23:
Your driver link has expired. Can you share me the updated link?

chuongnh98, 2023-09-23:
[image]

hkazu, 2023-09-23:
I contributed a tpms driver for EG53MA4 long ago, It should be in the main source code.

chuongnh98, 2023-09-23:
I need for fcc id MRXAFFPA4. Can you help me. Thank you very much