merbanan/rtl_433 #635: Cannot recognize WH2 sensor on Linux, while Windows binaries work fine

hellothere3, 2017-11-20:
This Windows binary decodes the WH2 weather sensor and recognizes it as such just fine - https://cognito.me.uk/uploads/rtl433/rtl433_win64_11112016.zip

When I compile (https://github.com/merbanan/rtl_433/) on a Raspberry Pi 3, I get the same data, but the device is not recognized - please see the attached image - even though the pulses are the same.

I get some warnings when I make rtl_433, but it seems to work fine other than not recognizing my device: https://pastebin.com/81Y4UWiH

Any input on what could be wrong would be appreciated 

[image]

zuckschwerdt, 2017-11-21:
Look at the build date (2016-11-11?) and try a version near that date (git checkout <commit>). Maybe you can pinpoint where things start going wrong.

DL4APT, 2017-12-17:
I have a newer windows binary here that worked quite fine for me: https://github.com/DL4APT/rtl_433/releases 
At the moment I'm looking into making an automated builder for windows binarys

shtirlic, 2018-04-14:
Got something similar on my setup and latest versions, fixed it via playing with `gain` otpion for rtl_433

merbanan, 2018-12-09:
@hellothere3 without any sample signals there is nothing we can help with. If we get samples that don't decode with a newer version but did with an old version we can fix the decoder. Without any samples we can't do anything. Closing issue.

closed 2018-12-09 by merbanan