merbanan/rtl_433 #641: Stack-fault using MSVC

gvanem, 2017-12-07:
After some troubles building with *clang-cl*, I get a stack-fault on a simple command like `rtl_433.exe -h`.
Since MSVC's *cl* is not supported it seems. But I do use MSVC' *link*. 

The culprit is the huge ` r_device devices[]` array on stack in `main()`.  
Linking with `-stack:20000000` fixes it (<a href="https://www.codecogs.com/eqnedit.php?latex=\fn_jvn&space;\tiny&space;20*10^6" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\fn_jvn&space;\tiny&space;20*10^6" title="\tiny 20*10^6" /></a> bytes!!).  The default stack-size is 2MByte AFAIK.

So, can this be done differently?

zuckschwerdt, 2017-12-07:
`r_device` should be 304 bytes on 64-bits (tested with clang) and there are 98 devices right now. 30k should do (I see 29792 bytes). There is something else wrong here. Can you debug where the memory is going? Maybe pepper it with fprintf...sizeof?

gvanem, 2017-12-07:
You're right. The size of `devices[]` is only 28224 bytes. I'm on 32-bit (although 64-bit Win-10).
But linking with `-stack:0x200000` fixes the crash (which happens before `main()`). So it must be something else that triggers it. I fail to see why. No constructors in the code AFAICS.

Will try MinGW next.

*PS*. You can change the `__MINGW32__` to `_WIN32` around `setenv()`.

zuckschwerdt, 2017-12-07:
Does `_WIN32` catch MinGW as well as MSVC?
Other than `sentenv()` https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L1035 there is also `#include "getopt/getopt.h"` in https://github.com/merbanan/rtl_433/blob/master/include/rtl_433.h#L22 -- should we update that too?

zuckschwerdt, 2017-12-07:
If you are using cmake, can you take a look at https://github.com/merbanan/rtl_433/blob/master/CMakeLists.txt#L32 and see if we can enable some compiler flags?

gvanem, 2017-12-07:
> Does _WIN32 catch MinGW as well as MSVC?

Catch? Yes, `_WIN32` is a built-in on both. If that is what you mean.

And the `rtl_433.h` WIN32-part, should IMHO be written a bit clearer. Like this:
```c
#ifdef _WIN32
  #include <windows.h>
  #include <io.h>
  #include <fcntl.h>
  #ifdef _MSC_VER
    #include "getopt/getopt.h"
    #define F_OK  0
  #endif
#endif

#ifndef _MSC_VER
  #include <unistd.h>
  #include <getopt.h>
#endif
```

MinGW's `<unistd.h>` includes `<getopt.h>`.

zuckschwerdt, 2017-12-07:
Very good, thanks! #643 should cover this. I'll try to attribute it to you on merge.

gvanem, 2017-12-07:
I've now built using MinGW ([TDM-gcc](http://tdm-gcc.tdragon.net/)), 32-bit.  And I get from a `pedump rtl_433.exe`:
```
 stack reserve size            200000
 stack commit size             1000
```
which is the double reserve size as MSVC link uses. So MinGW works fine as-is. But the question remains; who the heck needs so much stack?

I even added `-fms-compatibility`to my clang-cl `CFLAGS`. So I think there's an issue with clang-cl v.50. The best thing would be to get MSVC's cl working in this project. The  major hurdle is all the non-constant arrays like: `char buffer[element_size];`. Why not `alloca()` instead?

zuckschwerdt, 2017-12-07:
Wouldn't `alloca()` use stack the same way as a local declaration?

The arrays in the devices should all be rather small (`time_str`, `code_str`, …). Maybe `static` would be a small improvement but killing reentrance will bite us someday. Using `malloc()` seems overkill.

There shouldn't be any big buffers around. The biggest patch of memory `dm_state` is already on the heap. A pointer to something that looks bad is always welcome;)

gvanem, 2017-12-07:
> Wouldn't alloca() use stack the same way as a local declaration?

Sure, that's the point. The `a` means automatic. 
I see no more effective asm-code in e.g. `print_json_data()`.
Disassemble this gcc-snippet:
```
char buffer[element_size];
```
and compare to a:
```
char *buffer = alloca (element_size);
```
It's nearly the same asm-code.

gvanem, 2017-12-07:
> The best thing would be to get MSVC's cl working in this project. 

And also since MSVC generates a lot of useful warnings. Like here:
```c
void data_csv_free(void *aux)
{
	data_csv_aux_t *csv = aux;
	free(csv->fields);  !! this is a const
  ...
```

zuckschwerdt, 2017-12-07:
I don't really follow C's logic on `const` placement in pointer declarations. But I guess the declaration (`const char **fields;`) here is correct: (right to left) a (mutable) pointer to const char (immutable) pointer (mutable)?

gvanem, 2017-12-07:
Back to the issue. Disassembling clang-cl's `main()`:
```
 _main:
         push            ebp
         mov             ebp,esp
         push            ebx
         push            edi
         push            esi
         and             esp,0xfffffff8
         mov             eax,0x00147818      <! the size clang-cl think is needed in main()
         call            j^__chkstk
```
Using `cl -analyze:stacksize1000` on rtl_433.c gives this:
```
Function uses '1339828' bytes of stack:  exceeds /analyze:stacksize '1000'.
 Consider moving some data to heap.
```
`0x00147818` is 1 341 464 bytes. Pretty close to what `analyze` reports. But if it's not the size of `devices`,  then what?

gvanem, 2017-12-07:
I think I found it. `rtl_433.c`, line 1215:
```
    if (in_filename) {
        int i = 0;
        unsigned char test_mode_buf[DEFAULT_BUF_LENGTH];
        float test_mode_float_buf[DEFAULT_BUF_LENGTH];
```
These are the stack-eaters. Why do they need to be on the stack?

zuckschwerdt, 2017-12-07:
I guess it's a buffer instead of mmap so that we can read from stdin. The buffer is too big with 16*16384 instead of 131072 and the float buffer comes in at 1M. Putting it on the stack? I don't see a reason other than laziness.
I'll fix it right away.

gvanem, 2017-12-07:
Good job. I'll close this now as the problem is fixed. Thanks!

closed 2017-12-07 by gvanem