merbanan/rtl_433 #652: Honeywell ActivLink Doorbell (868 Mhz)

Jc2k, 2017-12-17:
I'm trying to add support for [Honeywell Actilink doorbells](https://www.amazon.co.uk/Honeywell-DC915N-200-LED-Doorbell/dp/B016MR3FEW/).

I'm running rtl_433 with:

```
/usr/local/bin/rtl_433 -f 868428000 -A
```

I'm expecting up to 2 packets from the doorbell - i believe the receiver potentially replies (the marketing material talks about a 'confidence LED' to indicate the button is working - the marketing doesn't make it clear whether it confirms the receiver got it or not. It might also be "forwarding" the event (again marketing material talks about range extending - 'Link two doorbells together to boost your range up to 400m.'). 

The analysis device picks out the following parameters:

```
Guessing modulation: Pulse Width Modulation with startbit/delimiter
Attempting demodulation... short_limit: 64, long_limit: 104, reset_limit: 117, demod_arg: 2
pulse_demod_pwm_ternary(): Analyzer Device 
```

So I took those parameters and added a new device. My callback just pretty prints the contents of the bitbuffer. Weirdly though instead of getting the same data as the auto-analysis pretty printed my callback seems to get called multiple times with bitbuffers that have 2 empty rows. If i increase the the reset_limit it can batch the packets together - and i'll get 2 lots of 25 rows that are empty.

I'm at a bit of a loss as to why the parameters detected automatically don't seem to work when plugged in manually.

Below is the full report it from the auto-analysis. 

```
trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 868428000 Hz.
Detected OOK package	@ 2017-12-17 10:36:40
Analyzing pulses...
Total count: 1200,  width: 146405		(585.6 ms)
Pulse width distribution:
 [ 0] count:  298,  width:    84 [83;87]	( 336 us)
 [ 1] count:  878,  width:    44 [43;47]	( 176 us)
 [ 2] count:   24,  width:   124 [124;126]	( 496 us)
Gap width distribution:
 [ 0] count:  298,  width:    35 [34;38]	( 140 us)
 [ 1] count:  877,  width:    75 [74;77]	( 300 us)
 [ 2] count:   24,  width:   114 [113;115]	( 456 us)
Pulse period distribution:
 [ 0] count: 1175,  width:   119 [117;123]	( 476 us)
 [ 1] count:   24,  width:   238 [238;240]	( 952 us)
Level estimates [high, low]:  15874,     26
Frequency offsets [F1, F2]:  -19098,      0	(-72.9 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with startbit/delimiter
Attempting demodulation... short_limit: 64, long_limit: 104, reset_limit: 116, demod_arg: 2
pulse_demod_pwm_ternary(): Analyzer Device 
bitbuffer:: Number of rows: 25 
[00] {48} 8d f5 00 20 00 01 : 10001101 11110101 00000000 00100000 00000000 00000001 
[01] {48} 8d f5 00 20 00 01 : 10001101 11110101 00000000 00100000 00000000 00000001 
[02] {48} 8d f5 00 20 00 01 : 10001101 11110101 00000000 00100000 00000000 00000001 
[03] {48} 8d f5 00 20 00 01 : 10001101 11110101 00000000 00100000 00000000 00000001 
[04] {48} 8d f5 00 20 00 01 : 10001101 11110101 00000000 00100000 00000000 00000001 
[05] {48} 8d f5 00 20 00 01 : 10001101 11110101 00000000 00100000 00000000 00000001 
[06] {48} 8d f5 00 20 00 01 : 10001101 11110101 00000000 00100000 00000000 00000001 
[07] {48} 8d f5 00 20 00 01 : 10001101 11110101 00000000 00100000 00000000 00000001 
[08] {48} 8d f5 00 20 00 01 : 10001101 11110101 00000000 00100000 00000000 00000001 
[09] {48} 8d f5 00 20 00 01 : 10001101 11110101 00000000 00100000 00000000 00000001 
[10] {48} 8d f5 00 20 00 01 : 10001101 11110101 00000000 00100000 00000000 00000001 
[11] {48} 8d f5 00 20 00 01 : 10001101 11110101 00000000 00100000 00000000 00000001 
[12] {48} 8d f5 00 20 00 01 : 10001101 11110101 00000000 00100000 00000000 00000001 
[13] {48} 8d f5 00 20 00 01 : 10001101 11110101 00000000 00100000 00000000 00000001 
[14] {48} 8d f5 00 20 00 01 : 10001101 11110101 00000000 00100000 00000000 00000001 
[... 151 lines cut ...]

Detected OOK package	@ 2017-12-17 10:36:43
Analyzing pulses...
Total count:   53,  width:  6524		(26.1 ms)
Pulse width distribution:
 [ 0] count:   39,  width:    44 [42;47]	( 176 us)
 [ 1] count:    2,  width:   124 [123;126]	( 496 us)
 [ 2] count:   12,  width:    84 [83;89]	( 336 us)
Gap width distribution:
 [ 0] count:   38,  width:    75 [72;78]	( 300 us)
 [ 1] count:    2,  width:   115 [114;116]	( 460 us)
 [ 2] count:   12,  width:    35 [31;37]	( 140 us)
Pulse period distribution:
 [ 0] count:   50,  width:   120 [117;123]	( 480 us)
 [ 1] count:    2,  width:   239 [239;240]	( 956 us)
Level estimates [high, low]:  15982,     12
Frequency offsets [F1, F2]:  -17380,      0	(-66.3 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with startbit/delimiter
Attempting demodulation... short_limit: 64, long_limit: 104, reset_limit: 117, demod_arg: 2
pulse_demod_pwm_ternary(): Analyzer Device 
bitbuffer:: Number of rows: 3 
[00] {2} 00 : 00
[01] {48} 8d f5 00 20 00 08 : 10001101 11110101 00000000 00100000 00000000 00001000 
[02] {1} 00 : 0
```

Jc2k, 2017-12-17:
Test driver: https://github.com/Jc2k/rtl_433/blob/master/src/devices/honeywell_activlink.c

Have tried with short_limit, long_limit and reset_limit exactly as in output and increasing and reducing them as well.

Large reset_limit avoids lots of small empty events, but instead yields large (number of rows) empty events.

Jc2k, 2017-12-17:
Samples added here: https://github.com/merbanan/rtl_433_tests/pull/201

zuckschwerdt, 2017-12-17:
PWM (ternary) looks right.
```
{48} 72 0a ff df ff f7 : 01110010 00001010 11111111 11011111 11111111 11110111
```
Also looks like the sample data.

The symbols are 480 us (120 samples @250k) long,
with 160 us pulse + 320 us gap ( = one bit)
and  320 us pulse + 160 us gap ( = zero bit)
and then a 480 us pulse + 480 us gap to terminate the packet

Perhaps try the `feat-gdecode` branch with
```
rtl_433 -q -X activelink:OOK_PWM_RAW:240:400:720:0:48 -r ../rtl_433_tests/tests/honeywell_activlink/01/one-press_868.428M_250k.cu8
```

that's `OOK_PULSE_PWM_RAW` with
```
short_limit = (160+320)/2; // Threshold between short and long pulse [us]
long_limit = (320+480)/2; //  Maximum gap size before new row of bits [us]
reset_limit = 480*1.5; // Maximum gap size before End Of Message [us].
demod_arg = 0; // Do not remove any startbits
```

which works well.

If you try
```
rtl_433 -q -X activelink:OOK_PWM_TERNARY:240:400:720:2:48 -r ../rtl_433_tests/tests/honeywell_activlink/01/one-press_868.428M_250k.cu8
```

which is OOK_PULSE_PWM_TERNARY with
```
short_limit = (160+320)/2; // Threshold between short and middle pulse [us]
long_limit = (320+480)/2; // Threshold between middle and long pulse [us]
reset_limit = 480*1.5; // Maximum gap size before End Of Message [us].
demod_arg = 2: // Short pulse is 0, middle is 1, long is sync
```

it doesn't work so well.

zuckschwerdt, 2017-12-17:
Though 
```
rtl_433 -q -X activelink:OOK_PWM_TERNARY:240:400:400:2:48 -r ../rtl_433_tests/tests/honeywell_activlink/01/one-press_868.428M_250k.cu8
```
i.e.`reset_limit = 400;` works but runs your callback for each packet.

Breakerz, 2018-02-26:
I would like to buy this device as my doorbell for my home automation... do you think it’s will be add to the master branch ?

zuckschwerdt, 2018-02-26:
Just use the `-X` codes given above and filter for your device id (`match=`). It's the prefered way for doorbell-style devices.

Breakerz, 2018-03-29:
I got a regular Honeywell doorbell. I don't think the signal can ben demodulate. It's is possiblie trap  the signal ?  I just want to trigger my home assistant.  I am confuse about short, long, reset limit... can you help me.

https://fccid.io/HS951380SL

The doorbell have 3 type of rings (set with a tiny selector switch)... this is ring #2.

```
sudo rtl_433 -f 345011000 -a -A
Tuned to 345011000 Hz.
*** signal_start = 677037, signal_end = 1028308
signal_len = 351271,  pulses = 1199
Iteration 1. t: 77    min: 52 (308)    max: 102 (692)    delta 362
Iteration 2. t: 77    min: 52 (308)    max: 102 (692)    delta 0
Distance coding: Pulse length 77

Short distance: 46, long distance: 96, packet distance: 1735

p_limit: 77
bitbuffer:: Number of rows: 25 
[00] {4} 40 : 0100
[01] {12} 30 40 : 00110000 0100
[02] {12} 30 40 : 00110000 0100
[03] {12} 30 40 : 00110000 0100
[04] {12} 30 40 : 00110000 0100
[05] {12} 30 40 : 00110000 0100
[06] {12} 30 40 : 00110000 0100
[07] {12} 30 40 : 00110000 0100
[08] {12} 30 40 : 00110000 0100
[09] {12} 30 40 : 00110000 0100
[10] {12} 30 40 : 00110000 0100
[11] {12} 30 40 : 00110000 0100
[12] {12} 30 40 : 00110000 0100
[13] {12} 30 40 : 00110000 0100
[14] {12} 30 40 : 00110000 0100
[15] {12} 30 40 : 00110000 0100
[16] {12} 30 40 : 00110000 0100
[17] {12} 30 40 : 00110000 0100
[18] {12} 30 40 : 00110000 0100
[19] {12} 30 40 : 00110000 0100
[20] {12} 30 40 : 00110000 0100
[21] {12} 30 40 : 00110000 0100
[22] {12} 30 40 : 00110000 0100
[23] {12} 30 40 : 00110000 0100
[24] {6} 30 : 001100
... Maximum number of rows reached. Message is likely truncated.
```

klohner, 2018-09-24:
The doorbell detailed by @Jc2k seems to be very similar to the North American 916.8 MHz model variants I've been playing with.  The only difference seems to be the frequency used.

I have a GitHub project started for the [Honeywell Wireless Doorbell](https://github.com/klohner/honeywell-wireless-doorbell).  Please let me know if you'd like to help with this effort!

Jc2k, 2018-09-24:
@klohner Looking promising! I don't have much (any!) spare time right now but you want me to check anything out with my devices let me know and i'll see what i can do.

klohner, 2018-09-24:
@Jc2k @zuckschwerdt Thanks so much for the replies.  Yes, these devices aren't really that "dumb" and I think a proper decoder for them would be useful.  

I've put together a quick form to submit signals at https://goo.gl/forms/SuxA3qgVRivXmNMf1 if anyone would like to help with my efforts.

merbanan, 2018-12-06:
#898 adds support for this. @zuckschwerdt can you review the code? All samples decode fine.

merbanan, 2018-12-09:
Support for this merged. Closing issue.

closed 2018-12-09 by merbanan