merbanan/rtl_433 #653: Segmentation fault on multiple devices

derjoern, 2017-12-17:
Hi there!

I have two devices on an Raspberry Pi 2 with Raspian Stretch and latest rtl_433:

root@smartberry:~/software/rtl_433# rtl_test -t
Found 2 device(s):
  0:  Generic, RTL2832U, SN: 77771111153705700
  1:  Realtek, RTL2838UHIDIR, SN: 00000001

If I start rtl_433 normal everything went fine:

root@smartberry:~/software/rtl_433# rtl_433 -I 2 -q
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Tuned to 433920000 Hz.

If I try to use -d 0 or -d1 it will end in an Segmentation fault. 

root@smartberry:~/software/rtl_433# rtl_433 -d 0 -I 2 -q
Segmentation fault

root@smartberry:~/software/rtl_433# rtl_433 -d 1 -I 2 -q
Segmentation fault

Any advice for me?

zuckschwerdt, 2017-12-17:
I don't see anything obvious with the code. Does this happen only when both devices are plugged in? Does this only happen on the Raspi?

zuckschwerdt, 2017-12-17:
Ah well, I see it. Stupid me. Try to put the `-d 0` as last arg and it should work. I'll fix it right away.

derjoern, 2017-12-17:
Thanks for the fast answer. Putting -d 0/-d 1 at the end has no effect, it keeps segfaulting. It happens only when I use this arguments. Both devices functions on its own (i.e. swap them so the order changes and use rtl_433 without -d).

zuckschwerdt, 2017-12-17:
It's already fixed with https://github.com/merbanan/rtl_433/commit/15205bb -- can you test most recent `master`?

derjoern, 2017-12-17:
Wow, thats was a seriuos fast solution :) It works. Many thanks!

closed 2017-12-17 by derjoern