merbanan/rtl_433 #657: Prtologue Sensors not working with latest git-source

storole, 2017-12-20:
Hello! I built rtl_433 from the latest source (2017-12-19) and I noticed that there where a number of new sensors listed on the page. I found that none of my Prologue based sensors worked anymore.  All my Oregon S. sensors worked fine. Went back to a clone I did 2017-10-16, and all Prologue sensors showed up again.

merbanan, 2017-12-20:
Can you do a git bisect or send a signal sample ?

storole, 2017-12-20:
OK! I will send signal sample generated with latest git source.

zuckschwerdt, 2017-12-20:
I just check, `rtl_433_tests/tests/prologue/02/` has json files and works, `rtl_433_tests/tests/prologue/01/` doesn't and does not work. Will be interesting to compare samples.

zuckschwerdt, 2017-12-20:
I found the bug and I'll fix it tomorrow. Thanks for reporting and the new sample data.

zuckschwerdt, 2017-12-20:
It's fixed now. The false positive reduction vs alecto/auriol was too harsh.

storole, 2017-12-21:
Thanks! Just tested it and it works perfectly!

closed 2017-12-21 by storole