merbanan/rtl_433 #660: Decoding signal Globaltronics gt-ts-05

alcadic, 2017-12-29:
Hi All!

I try to detect the temperature from my weather station but I have a problem.....I receive always this trace with the command rtl_433 -a -f 433810500

*** signal_start = 6017639, signal_end = 6278731
signal_len = 261092,  pulses = 268
Iteration 1. t: 86    min: 19 (1)    max: 153 (267)    delta 4
Iteration 2. t: 86    min: 19 (1)    max: 153 (267)    delta 0
Pulse coding: Short pulse length 19 - Long pulse length 153

Short distance: 509, long distance: 1021, packet distance: 2249

p_limit: 86
bitbuffer:: Number of rows: 9
[00] {1} 00 : 0
[01] {38} ff ff ff ff fc : 11111111 11111111 11111111 11111111 111111
[02] {38} ff ff ff ff fc : 11111111 11111111 11111111 11111111 111111
[03] {38} ff ff ff ff fc : 11111111 11111111 11111111 11111111 111111
[04] {38} ff ff ff ff fc : 11111111 11111111 11111111 11111111 111111
[05] {38} ff ff ff ff fc : 11111111 11111111 11111111 11111111 111111
[06] {38} ff ff ff ff fc : 11111111 11111111 11111111 11111111 111111
[07] {38} ff ff ff ff fc : 11111111 11111111 11111111 11111111 111111
[08] {1} 80 : 1

This trace is with 18.6ºC but with other temperatures is the same, and I don't know why........

Somebody can oriented me please?.

Thanks in advance!!

zuckschwerdt, 2017-12-29:
Take some sample recordings with `rtl_433 -q -a -t`. Check the spectrograms in http://triq.net/iqs
Try the Pulse Analyzer `rtl_433 -q -A -r FILENAME.cu8`.

alcadic, 2017-12-29:
I can't see diferences between temperatures, nor with X nor the web.
I attach the firts and last file (files.zip), and I can't see nothing different.
[files.zip](https://github.com/merbanan/rtl_433/files/1593727/files.zip)

zuckschwerdt, 2017-12-29:
The signal isn't very good but decodes fine as Prologue sensor with valid data:
```
{"time" : "@0.524288s", "model" : "Prologue sensor", "id" : 9, "rid" : 171, "channel" : 1, "battery" : "LOW", "button" : 0, "temperature_C" : 18.100, "humidity" : 10}

{"time" : "@0.524288s", "model" : "Prologue sensor", "id" : 9, "rid" : 171, "channel" : 1, "battery" : "LOW", "button" : 0, "temperature_C" : 17.600, "humidity" : 10}
```

What exactly isn't working for you?

alcadic, 2017-12-29:
wow, how did you do it? I mean, I want extract the temperature with rtl_433 reading bits colums from all result rows in C but I see that in both, all rows are the same, specific, rows between 01 and 07 are the same:

[01] {38} ff ff ff ff fc: 11111111 11111111 11111111 11111111 111111

How I can do it in C ?? How can you extract the information in JSON? it's amazing

zuckschwerdt, 2017-12-29:
`rtl_433 -q -F json -C si` should just work. It's what `rtl_433` does.
And `rtl_433 -q -F json -C si -r g058_433.92M_250k.cu8` for your sample file.

alcadic, 2017-12-29:
I understand but ..... have a problem ..... rtl_433 don't show me nothing without "-a" argument, and with these argument I cant extract the information in identical bits :(

I have also tried the option to read from the .cu8 file (althoug I won't use normally) and the result is:
Test mode active. Reading samples from file: g001_433.92M_250k.cu8

No more .....

zuckschwerdt, 2017-12-29:
Strange. I just tested. It works fine. Maybe your version or compile of rtl_433 is broken.
```
rtl_433 -q -F json -C si -r g001_433.92M_250k.cu8
Test mode active. Reading samples from file: g001_433.92M_250k.cu8
{"time" : "@0.524288s", "model" : "Prologue sensor", "id" : 9, "rid" : 171, "channel" : 1, "battery" : "LOW", "button" : 0, "temperature_C" : 18.100, "humidity" : 10}
```

zuckschwerdt, 2017-12-30:
I remember now ;) Prologue had problems until 10 days ago. See #657. Just get the latest revision and compile rtl_433 again.

closed 2017-12-30 by zuckschwerdt