merbanan/rtl_433 #662: decodage lacrosse ws1650 station meteo 5n1

yves-jose, 2018-01-01:
Bonjour 
Je  travaille sur le décodage d'une station WS1650 qui émet en 868 mhz  modulation FSK_PULSE_PCM.
Je suis arrivé a ce résultat :
[00] {285} 00 aa aa aa aa aa 2d d4 e4 8e 7f e1 8f df ef 98 ff 6e db f8 ff 1b 71 80 1e 70 20 10 67 00 91 24 07 00 00 00
2018-01-01 16:34:34     :       LaCrosse :      WS1650  :       142
        Battery:         OK
        NewBattery:      0
        Temperature:     6.7 C
        Humidity:        91 %
        Wind speed:      2.00 m/s
        Wind speed:      7.20 km/h
        wind_dir_deg:    157.5
        wind_dir:        SSE
        rainfall Cumul:  72.40 mm
Qui  est correct par rapport a ce que donne l'afficheur, mais n'utilisant que  8 octets de fin de trame : 70 20 10 67 00 91 24 07, je me pose des questions sur le reste de la trame ( le debut ) , je recherche les rafales de vent, le sensor ID ( rolling au changement de la batterie)

zuckschwerdt, 2018-01-01:
Sorry, english please.

yves-jose, 2018-01-01:
Hello
I am working on the decoding of a WS1650 station that emits in 868 mhz modulation FSK_PULSE_PCM.
I arrived at this result:
[00] {285} 00 aa aa aa aa aa 2d d4 e4 8e 7f e1 8f df ef 98 ff 6e db f8 ff 1b 71 80 1e 70 20 10 67 00 91 24 07 00 00 00
2018-01-01 16:34:34 : LaCrosse : WS1650 : 142
Battery: OK
NewBattery: 0
Temperature: 6.7 C
Humidity: 91 %
Wind speed: 2.00 m/s
Wind speed: 7.20 km/h
wind_dir_deg: 157.5
wind_dir: SSE
rainfall Cumul: 72.40 mm
Correct results compared to the display, but using only 8 bytes of end of frame : "70 20 10 67 00 91 24 07" , I ask myself about the rest of the frame (the beginning), I search the gusts of wind, the sensor ID (rolling at the change of the battery)
=> Used Google translate

yves-jose, 2018-01-04:
Travail terminé, juste quelques valeurs à affiner ( température négative et rafale de vent )
Work done, just a few values ​​to refine (negative temperature and gust of wind)
I am not a computer scientist and do not speak English 
Bye

closed 2018-01-04 by yves-jose