merbanan/rtl_433 #663: Cheap motion sensor on EV1527

nalesnikizdzemem, 2018-01-06:
Hello all,
first of all sorry is I ask stupid questions but I new in rtl_433. Thanks for understanding.

Recently I purchased some Chines cheap PIR motion detector in order to use it on Domoticz with rtl_433. 
Device is based on EV1527 chip.
Unfortunately rtl_433 is detecting the activation signal as series of generic_remote and is unusable in Domoticz:

_@2.621440s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@2.621440s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@2.621440s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@2.621440s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F_ 

 and so on.

I have no clue how to make it work with rtl_433 and so with Domoticz.
Could somebody help me with that please?
The rtl_433 -a -A shows as follow:

_Test mode active. Reading samples from file: gfile005.data
Input format: uint8
*** signal_start = 50318, signal_end = 393217
signal_len = 342899,  pulses = 121
Iteration 1. t: 2    min: 2 (81)    max: 3 (40)    delta 4
Iteration 2. t: 145    min: 0 (0)    max: 290 (121)    delta 82373
Iteration 3. t: 145    min: 290 (121)    max: 0 (0)    delta 168200
Iteration 4. t: 145    min: 290 (121)    max: 0 (0)    delta 0
Distance coding: Pulse length 145

Short distance: 1238, long distance: 4960, packet distance: 11475

p_limit: 145
bitbuffer:: Number of rows: 6
[00] {8} b3 : 10110011
[01] {8} c2 : 11000010
[02] {12} 60 c0 : 01100000 1100
[03] {26} 19 09 02 00 : 00011001 00001001 00000010 00
[04] {61} 29 02 32 a8 04 02 20 c0
[05] {0} :
Test mode file issued 6 packets_

Here is the link to device photos and gfile:

https://1drv.ms/f/s!AkQvnJv4Atd9aTBxnmZjzvroQgQ

Thanks in advance.

zuckschwerdt, 2018-01-06:
Yes, it looks like the usual x1527, the gfile is very good.
25 pulses per packet, 26 packets per transmission.

1600 us for a full pulses + gap (9597 samples for 24 pulses).

So the long pulse is 1200 us + 400us gap,
short pulse is 400 us + 1200us gap.

The gap between packets is 12800 us.

Try the feat-gdecode branch
```
git clone https://github.com/zuckschwerdt/rtl_433.git
cd rtl_433
git checkout feat-gdecode
mkdir build ; cd build
cmake .. && make
```

with
```
rtl_433 -q -C si -F json -R 1 -X "pir:OOK_PWM_RAW:800:2000:13000,match={24}0x5a1ea6,minrepeats=3" -r ~/Desktop/gfile005.data
```

This will give you one JSON line per PIR detection.

nalesnikizdzemem, 2018-01-06:
Thanks a lot! I will try it ASAP.

nalesnikizdzemem, 2018-01-06:
I got result as below. What should I do now?

 _rtl_433 -q -C si -F json -R 1 -X "pir:OOK_PWM_RAW:800:2000:13000,match={24}0x5a1ea6,minrepeats=3" -r ~/gfile005.data
Test mode active. Reading samples from file: /home/pi/gfile005.data
{"time" : "@1.048576s", "model" : "pir", "num_rows" : 25, "rows" : [{"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 25, "data" : "5a1ea680"}, {"len" : 16, "data" : "5a1f"}]}_

nalesnikizdzemem, 2018-01-06:
Domoticz is still stating:

_2018-01-06 17:00:53.702 Rtl433: Unhandled sensor type, please report: (2018-01-06 17:00:53,Generic Remote,42465,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
)
2018-01-06 17:00:55.007 Rtl433: Unhandled sensor type, please report: (2018-01-06 17:00:53,Generic Remote,42465,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
)
2018-01-06 17:00:55.509 Rtl433: Unhandled sensor type, please report: (2018-01-06 17:00:53,Smoke detector GS 558,21565,,,,,,,,,,,,,,,,,,,5,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,0,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,9a87a5,,,,,,,,,,,,,,,,,,,,,,,,
)_

zuckschwerdt, 2018-01-06:
The output `{"time" :…}` is JSON. It's all you get. No database.
I don't know about Domoticz. The `-R 1` (or `-R 0` in newest version) should disable Generic Remote, Smoke detector, …
I guess this an issue with Domoticz not rtl_433.

closed 2018-01-06 by zuckschwerdt

nalesnikizdzemem, 2018-01-06:
I am sorry for stupid question but how to add it to rtl_433?

zuckschwerdt, 2018-01-06:
No problem. But if this an issue with Domoticz, you need to ask there.
```
rtl_433 -q -F json -R 1 -X "pir:OOK_PWM_RAW:800:2000:13000,match={24}0x5a1ea6,minrepeats=3"
```
is all you can do with `rtl_433` (the `-R 1` is there, see?)
Or what do you mean, add _it_ to rtl_433?

nalesnikizdzemem, 2018-01-07:
Hi, thanks for your help.
To add it I mean to detect it in rtl_433 like of/off switch. I can imagine the output of rtl_433 could be 
- device id
- activation signal (I believe the device sends only activation signal)

like it is happening in case of Temp sensors:
_**Test mode active. Reading samples from file: gfile008.cu8
Input format: uint8
@1.048576s :    AlectoV1 Temperature Sensor
        House Code:      83
        Channel:         2
        Battery:         OK
        Temperature:     8.50 C
        Humidity:        20 %
        :       CHECKSUM
Test mode file issued 3 packets**_

Right  now I receive quite long group of signals:

 _**Test mode active. Reading samples from file: gfile016.cu8
Input format: uint8
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Generic Remote
        House Code:      42465
        Command:         89
        Tri-State:       !!FF1!0FFF!F
@1.048576s :    Smoke detector GS 558   :       21565   :       5       :       0
        Raw Code:        9a87a5
Test mode file issued 3 packets**_