merbanan/rtl_433 #664: [FEATURE] Option to prevent a device from "firing" multiple times

BenWoodford, 2018-01-07:
Just to give an example, my car key will send 3 or 4 MQTT messages. It would be nice if rtl_433 had a flag that meant it would "sleep" detection of a device for n milliseconds if it is detected. This would mean that a device that spams signals would only be detected once.

zuckschwerdt, 2018-01-07:
The guiding principle is to output everything that comes over the air fast and reliable. Coalescing events would introduce a lag. Pausing a decoder would need state, which decoders should not have.
It's not a bad idea as such, but needs to be generalized and designed well. Maybe something like token bucket rate limiting in the outputs.

skx, 2018-03-10:
I documented using this project here, to react to simple radio-buttons:

* https://steve.fi/Hardware/433-radio-button/

The only code change I made was to run an external program when a transmission was recognized, but I also made sure to do that only once per second.  A very similar approach in the receiver could be used to do this - but I guess that @zuckschwerdt is right in saying that @merbanan wants to output __everything__.

I did consider sending a pull-request to allow (optional) limiting to once per second for all decoders, but I suspect that it would be rejected.

zuckschwerdt, 2018-03-10:
We can't have real coalescing (i.e. count the output) as that would introduce unwanted lag. But an option(!) to suppress repeats is ok in my mind -- I'm also thinking of *not* spamming MQTT brokers here. Maybe something simple like store timestamp and last output (global not per device) with an option to then silence for some milliseconds. (But not extending the timeout thus giving a periodical output if needed. And every different output would reset the logic / or maybe compare to the last 3 or so outputs.) A token bucket limiting option could be more flexible though.

zuckschwerdt, 2018-03-10:
Nice use-case with your buttons and good write up. You can use the flexible decoder (`-X`) to tailor a decoder for your button(s) without the need to patch. Along with the upcoming MQTT support this should enable you to send exactly the messages you need.

merbanan, 2018-12-10:
PR #908 rtl_433 -q -Y 1 -r rtl_433_tests/tests/generic_remote/01/gfile003.cu8 will filter out duplicates from these type of devices.

gdt, 2023-09-24:
I have opened #2635 about the big picture and linked the PR.   I'd like to close this as we have an unmanageable about of open issues and it would be good to reduce/coalesce.

gdt, 2023-09-25:
This is now a dup of #2640

closed 2023-09-25 by gdt