merbanan/rtl_433 #675: remote control shutters

mttstt, 2018-02-05:
Hi,
I would like reproduce (with _433D program) the segnal of my remote control shutters, but  I am not able to interpret the output of rtl_433 -A. Can you help me?

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
Detected OOK package    @ 2018-02-05 20:29:10
Analyzing pulses...
Total count:   78,  width: 20737                (82.9 ms)
Pulse width distribution:
 [ 0] count:   27,  width:    80 [77;84]        ( 320 us)
 [ 1] count:   51,  width:   169 [167;171]      ( 676 us)
Gap width distribution:
 [ 0] count:   61,  width:   100 [96;103]       ( 400 us)
 [ 1] count:    1,  width:   982 [982;982]      (3928 us)
 [ 2] count:   15,  width:   189 [189;190]      ( 756 us)
Pulse period distribution:
 [ 0] count:   11,  width:   177 [175;180]      ( 708 us)
 [ 1] count:    1,  width:  1062 [1062;1062]    (4248 us)
 [ 2] count:   65,  width:   270 [267;272]      (1080 us)
Level estimates [high, low]:  15991,     49
Frequency offsets [F1, F2]:    7488,      0     (+28.6 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
Attempting demodulation... short_limit: 124, long_limit: 191, reset_limit: 983, sync_width: 0
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2
[00] {12} ff f0 : 11111111 1111
[01] {66} cc 09 00 00 b9 01 a2 00 00

Detected OOK package    @ 2018-02-05 20:29:10
Analyzing pulses...
Total count:   78,  width: 20737                (82.9 ms)
Pulse width distribution:
 [ 0] count:   27,  width:    80 [77;84]        ( 320 us)
 [ 1] count:   51,  width:   169 [167;171]      ( 676 us)
Gap width distribution:
 [ 0] count:   61,  width:    99 [96;102]       ( 396 us)
 [ 1] count:    1,  width:   981 [981;981]      (3924 us)
 [ 2] count:   15,  width:   188 [188;191]      ( 752 us)
Pulse period distribution:
 [ 0] count:   11,  width:   177 [173;182]      ( 708 us)
 [ 1] count:    1,  width:  1059 [1059;1059]    (4236 us)
 [ 2] count:   65,  width:   270 [267;273]      (1080 us)
Level estimates [high, low]:  15975,     46
Frequency offsets [F1, F2]:    6616,      0     (+25.2 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
Attempting demodulation... short_limit: 124, long_limit: 192, reset_limit: 982, sync_width: 0
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2
[00] {12} ff f0 : 11111111 1111
[01] {66} cc 09 00 00 b9 01 a2 00 00

Detected OOK package    @ 2018-02-05 20:29:10
Analyzing pulses...
Total count:   78,  width: 20739                (83.0 ms)
Pulse width distribution:
 [ 0] count:   27,  width:    81 [79;85]        ( 324 us)
 [ 1] count:   51,  width:   169 [166;171]      ( 676 us)
Gap width distribution:
 [ 0] count:   61,  width:    99 [95;104]       ( 396 us)
 [ 1] count:    1,  width:   982 [982;982]      (3928 us)
 [ 2] count:   15,  width:   189 [188;191]      ( 756 us)
Pulse period distribution:
 [ 0] count:   11,  width:   177 [175;180]      ( 708 us)
 [ 1] count:    1,  width:  1062 [1062;1062]    (4248 us)
 [ 2] count:   65,  width:   270 [267;272]      (1080 us)
Level estimates [high, low]:  15893,     63
Frequency offsets [F1, F2]:    6648,      0     (+25.4 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
Attempting demodulation... short_limit: 125, long_limit: 192, reset_limit: 983, sync_width: 0
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2
[00] {12} ff f0 : 11111111 1111
[01] {66} cc 09 00 00 b9 01 a2 00 00

Detected OOK package    @ 2018-02-05 20:29:10
Analyzing pulses...
Total count:   78,  width: 20738                (83.0 ms)
Pulse width distribution:
 [ 0] count:   27,  width:    81 [79;84]        ( 324 us)
 [ 1] count:   51,  width:   169 [167;171]      ( 676 us)
Gap width distribution:
 [ 0] count:   61,  width:    99 [96;102]       ( 396 us)
 [ 1] count:    1,  width:   981 [981;981]      (3924 us)
 [ 2] count:   15,  width:   188 [188;191]      ( 752 us)
Pulse period distribution:
 [ 0] count:   11,  width:   177 [176;180]      ( 708 us)
 [ 1] count:    1,  width:  1060 [1060;1060]    (4240 us)
 [ 2] count:   65,  width:   270 [267;274]      (1080 us)
Level estimates [high, low]:  15945,     46
Frequency offsets [F1, F2]:    5523,      0     (+21.1 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
Attempting demodulation... short_limit: 125, long_limit: 192, reset_limit: 982, sync_width: 0
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2
[00] {12} ff f0 : 11111111 1111
[01] {66} cc 09 00 00 b9 01 a2 00 00

Detected OOK package    @ 2018-02-05 20:29:10
Analyzing pulses...
Total count:   78,  width: 20738                (83.0 ms)
Pulse width distribution:
 [ 0] count:   27,  width:    80 [78;85]        ( 320 us)
 [ 1] count:   51,  width:   169 [168;170]      ( 676 us)
Gap width distribution:
 [ 0] count:   61,  width:    99 [96;102]       ( 396 us)
 [ 1] count:    1,  width:   982 [982;982]      (3928 us)
 [ 2] count:   15,  width:   189 [188;190]      ( 756 us)
Pulse period distribution:
 [ 0] count:   11,  width:   177 [175;181]      ( 708 us)
 [ 1] count:    1,  width:  1061 [1061;1061]    (4244 us)
 [ 2] count:   65,  width:   270 [268;272]      (1080 us)
Level estimates [high, low]:  15963,     45
Frequency offsets [F1, F2]:    7232,      0     (+27.6 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
Attempting demodulation... short_limit: 124, long_limit: 191, reset_limit: 983, sync_width: 0
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2
[00] {12} ff f0 : 11111111 1111
[01] {66} cc 09 00 00 b9 01 a2 00 00

Detected OOK package    @ 2018-02-05 20:29:10
Analyzing pulses...
Total count:   78,  width: 20738                (83.0 ms)
Pulse width distribution:
 [ 0] count:   27,  width:    80 [79;84]        ( 320 us)
 [ 1] count:   51,  width:   169 [168;171]      ( 676 us)
Gap width distribution:
 [ 0] count:   61,  width:   100 [96;102]       ( 400 us)
 [ 1] count:    1,  width:   980 [980;980]      (3920 us)
 [ 2] count:   15,  width:   189 [188;191]      ( 756 us)
Pulse period distribution:
 [ 0] count:   11,  width:   177 [176;180]      ( 708 us)
 [ 1] count:    1,  width:  1060 [1060;1060]    (4240 us)
 [ 2] count:   65,  width:   270 [268;272]      (1080 us)
Level estimates [high, low]:  15895,     63
Frequency offsets [F1, F2]:    6767,      0     (+25.8 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
Attempting demodulation... short_limit: 124, long_limit: 192, reset_limit: 981, sync_width: 0
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2
[00] {12} ff f0 : 11111111 1111
[01] {66} cc 09 00 00 b9 01 a2 00 00

zuckschwerdt, 2018-02-06:
A summary (like `-A`) won't work if you want to reproduce the signal. Look at the actual waveform.

closed 2018-02-06 by zuckschwerdt

mttstt, 2018-02-06:
Can someone help me to analyse the recorded signal?

Is it FSK or OOK modulation?

https://github.com/mttstt/rtl_433_tests/tree/c81f77065b73426cc1e3d0e5f2b5d0b67f200ad2/tests/aerf/01

zuckschwerdt, 2018-02-07:
Use http://triq.net/iqs to get a spectrogram (SoX is also okay but negative frequencies are not possible). One frequency (a line: `---`) means OOK, two frequencies (two lines: `--_--__`) means FSK.
Use Audacity to count the signal timing:
[image]
```
rtl_433 -q -F json -R 0 -X aerf:OOK_PWM:340:700:1000:100:700 -r g001_433.92M_250k.cu8

{"time" : "@0.000000s", "model" : "aerf", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 12, "data" : "fff"}]}
{"time" : "@0.000000s", "model" : "aerf", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 66, "data" : "55090000007640000"}]}
```

merbanan, 2018-02-07:
Based on this image and the rtl_433 output the signal has a preamble then the actual payload. The signal is a OOK signal with short pulses of length 330 us and long pulses of 660 us. The modulation is as the analyzer guesses:
Pulse Width Modulation with multiple packets

To generate the preamble send the short pulse 12 times and then wait for some duration then send the actual payload of 66 pulses.

mttstt, 2018-02-11:
In your experience which is the most effective solution to generate the signal (433Utils, pigs, ...)?
I have a raspberry and a cheap 433 Transmitter.
Any example?
Thanks

mttstt, 2018-02-16:
I found it !
https://github.com/mttstt/remote-control-shutters

zuckschwerdt, 2018-02-16:
good example. is this using one of those very cheap OOK transmitters (with just one synchronous data pin)?

mttstt, 2018-03-07:
YES !