merbanan/rtl_433 #677: mingw-w64 no longer compiles

pclov3r, 2018-02-11:
So, It appears that mingw-w64 can no longer compile rtl_433

Upon attempting to compile it I get an error saying that netdb.h is missing. Looking this isn't a part of mingw-w64 and to use winsock2.h as a replacement.

Doing that edit in data.c allows it to proceed further with it saying netinet/in.h is missing which also isn't a part of mingw-w64. Removing that allows it to proceed with various errors of things that are undeclared or undefined. 

`/usr/local/src/rtl_433_windows/src/data.c:633:5: error: unknown type name ‘socklen_t’
     socklen_t addr_len;
     ^~~~~~~~~
/usr/local/src/rtl_433_windows/src/data.c: In function ‘datagram_client_open’:
/usr/local/src/rtl_433_windows/src/data.c:642:21: error: storage size of ‘hints’ isn’t known
     struct addrinfo hints, *res, *res0;
                     ^~~~~
/usr/local/src/rtl_433_windows/src/data.c:650:22: error: ‘AI_ADDRCONFIG’ undeclared (first use in this function)
     hints.ai_flags = AI_ADDRCONFIG;
                      ^~~~~~~~~~~~~
/usr/local/src/rtl_433_windows/src/data.c:650:22: note: each undeclared identifier is reported only once for each function it appears in
/usr/local/src/rtl_433_windows/src/data.c:651:13: warning: implicit declaration of function ‘getaddrinfo’ [-Wimplicit-function-declaration]
     error = getaddrinfo(host, port, &hints, &res0);
             ^~~~~~~~~~~
/usr/local/src/rtl_433_windows/src/data.c:653:33: warning: implicit declaration of function ‘gai_strerror’ [-Wimplicit-function-declaration]
         fprintf(stderr, "%s\n", gai_strerror(error));
                                 ^~~~~~~~~~~~
/usr/local/src/rtl_433_windows/src/data.c:657:36: error: dereferencing pointer to incomplete type ‘struct addrinfo’
     for (res = res0; res; res = res->ai_next) {
                                    ^~
/usr/local/src/rtl_433_windows/src/data.c:667:5: warning: implicit declaration of function ‘freeaddrinfo’ [-Wimplicit-function-declaration]
     freeaddrinfo(res0);
     ^~~~~~~~~~~~
/usr/local/src/rtl_433_windows/src/data.c: At top level:
/usr/local/src/rtl_433_windows/src/data.c:763:19: error: ‘_POSIX_HOST_NAME_MAX’ undeclared here (not in a function)
     char hostname[_POSIX_HOST_NAME_MAX + 1];`

Looking none of this exists for MinGW. 

Is Windows no longer a supported platform?

zuckschwerdt, 2018-02-11:
I'm not aware that Windows (mingw) was ever a supported platform. However it should work, even if we can't support and test it. I'll see if I can just disable the whole network stuff for Windows.

pclov3r, 2018-02-11:
That would be the best way to go about it most likely.

Caused by https://github.com/merbanan/rtl_433/commit/c30a2a66bffdff6a58c722661434ee18e8cbff99 i suppose.

zuckschwerdt, 2018-02-11:
Yes. If https://github.com/zuckschwerdt/rtl_433/commit/ad7799a shows no regression it will be available shortly.

zuckschwerdt, 2018-02-11:
latest master now disables Syslog output for win32.

pclov3r, 2018-02-11:
Thanks!

Sadly, I ran into another issue when I thought it was going to build correctly. This time around undefined reference to `localtime_r'

A post here about it.

https://mail.gnome.org/archives/vala-list/2011-June/msg00040.html

Any ideas?

pclov3r, 2018-02-11:
Caused by I figure again https://github.com/merbanan/rtl_433/commit/aada5fd3dc351d19b9b7deaa394cdfce3f1f4ea2

Is it possible to use localtime instead of localtime_r for windows?

zuckschwerdt, 2018-02-11:
Ah, well Win32 and Posix… I reverted that change.

pclov3r, 2018-02-11:
Seems to all be good now and building and working. 

Thanks for the quick reply and changes!

closed 2018-02-11 by pclov3r

rct, 2018-02-11:
@zuckschwerdt - I wasn't aware syslog output had been added since commits that aren't attached to issues don't trigger github notifications on watched repos. 

I see you are doing a lot of work and there are lots of commits.  It would be nice for anyone following this repo if the commits were tied to an issue.  This would be especially helpful when there are multiple commits related to a particular feature or bug.

I see there were also a number of changes related to moving from _TERNARY to _PRECISE.  I'm curious about that one, because originally I had too much trouble with _PRECISE and due to the timings (shape of the pulse) changing with varying signal levels.   At this point I don't remember whether vestom's auto signal level work had matured at that point or not.

zuckschwerdt, 2018-02-11:
Most of these are preparations for features like MQTT and SoapySDR. Syslog is kind of a test ballon on some simple network stuff. There shouldn't be any features or functional changes. I'm testing with automated Travis builds, but there really should be more cases. Maybe I can setup something in the future.
I try to bundle and announce features with PRs but sometimes it's just plumbing and already too many branches to keep track of. Especially the smaller stuff is broken into descriptive commits, you should be fine just skimming the commit history. At some point we should really add a Changelog though.

"TERNARY" was a misnomer and really most protocols have some kind of sync pulse. (Incidentally, the few devices using TERNARY were just to get flipped bits…). Also the distinction between PRECISE and RAW wasn't user-friendly at all. The new PRECISE will cover all cases and it replaces both. You can just switch between "raw" and "precise" mode by setting the `tolerance`. Also the `sync_width` is always available if you need sync symbols (syncs are counted on each row without producing empty rows). The logic for row separation and "End of Message" is improved and should cover all use cases. I.e. you should get consistent results switching from "raw" to "precise" with no other changes need, if you like.

Unseen the biggest change was getting rid of three layers of abstractions on data output (of which I introduced one years back…) – one should now again be able to just read that code and understand it.
I'm hoping to shape up the older parts of the code and clear a way for more features. If something breaks just let me know -- it's not intended ;)

rct, 2018-02-11:
I know you are doing a lot of good work. Especially work that no one else is stepping up to do.

My comments were made after looking at the commit history. There are a lot of commits.  Particularly when there are multiple commits it isn't clear what benefit (feature or bug fix) it was tied to and what else might need to be backed out if there is a problem.

I'm sure you don't intend to break anything intentionally, but there isn't a really good test bed.  The situation is compounded by the lack of releases and version numbers. 

If simple things break that causes a failure to compile that's really no sweat. What I'm more concerned with is the stuff that could take a while to track down, like the reliability of device decoding has decreased depending on signal level.  

You changed the decoder for a bunch of devices on the master branch with no general warning. At least if it was mentioned in an github issue, the built in github notification for the 100 people who are following this repo would have a heads up.  Alternatively maybe there should be a development or clean up branch that could get tested by those who are willing?  I see there are two feature branches.

zuckschwerdt, 2018-02-12:
Good points. And I feel that a broader discussion is needed in many places. (look at https://github.com/merbanan/rtl_433/commit/d0a309b -- it was completly foobar, now way forward without breaking stuff. Someone should track down the owner of such devices). Using PRs was always the idea and I try to stick to that as best as possible. Also I'll compile a Changelog and carefully add some version tags ;)

rct, 2018-02-12:
Great - Thanks!