merbanan/rtl_433 #681: Build warnings on Debian unstable

RichiH, 2018-02-12:
I am seeing build warnings when building rtl_433 on current Debian sid:

````
% git log | head -n1
commit b58c3273a4db2b71283e00c735308bb196d2082f
% autoreconf --install
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
configure.ac:12: installing './compile'
configure.ac:15: installing './config.guess'
configure.ac:15: installing './config.sub'
configure.ac:5: installing './install-sh'
configure.ac:5: installing './missing'
Makefile.am:4: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
src/Makefile.am:1: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
src/Makefile.am: installing './depcomp'
% ./configure --prefix=/usr/local/rtl_433
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking whether make sets $(MAKE)... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking for gcc option to accept ISO C99... none needed
checking for gcc option to accept ISO Standard C... (cached) none needed
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
[... 148 lines cut ...]
  CC       devices/tpms_renault.o
  CC       devices/infactory.o
  CC       devices/fordremote.o
  CC       devices/philips.o
  CC       devices/nexa.o
  CC       devices/thermopro_tp12.o
  CC       devices/x10_sec.o
devices/x10_sec.c: In function ‘x10_sec_callback’:
devices/x10_sec.c:88:30: warning: ‘%02x’ directive writing between 2 and 8 bytes into a region of size 5 [-Wformat-overflow=]
         sprintf(x10_id_str, "%02x%02x", b[0], b[4]);
                              ^~~~
devices/x10_sec.c:88:29: note: directive argument in the range [0, 2147483647]
         sprintf(x10_id_str, "%02x%02x", b[0], b[4]);
                             ^~~~~~~~~~
devices/x10_sec.c:88:29: note: directive argument in the range [0, 255]
devices/x10_sec.c:88:9: note: ‘sprintf’ output between 5 and 11 bytes into a destination of size 5
         sprintf(x10_id_str, "%02x%02x", b[0], b[4]);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CCLD     rtl_433
make[2]: Leaving directory '/home/richih/work/git/rtl_433/src'
make[2]: Entering directory '/home/richih/work/git/rtl_433'
make[2]: Leaving directory '/home/richih/work/git/rtl_433'
make[1]: Leaving directory '/home/richih/work/git/rtl_433'
%
````

merbanan, 2018-02-12:
I am guessing that the compiler doesn't understand %02. A change to a safe sprintf would be good.

zuckschwerdt, 2018-02-16:
we "know" this won't overflow, but there are no gurantees. Do you like to PR changes to `snprintf(x10_id_str, sizeof(x10_id_str),`…, wherever applicable?

alexmyczko, 2018-09-17:
Meanwhile, it builds fine: http://sid.ethz.ch/debian/rtl-433/

zuckschwerdt, 2018-09-18:
Thanks. Please reopen if some compiler still chokes.

closed 2018-09-18 by zuckschwerdt