merbanan/rtl_433 #682: Signal caught, exiting! on General purpose decoder

jonathanmbradshaw, 2018-02-13:
The General Purpose Decoder is a great addition to already great decoder, thanks for all of the work.

I have the following exception when running rtl_433, the build is up to date on origin/master as of today. 

`{"time" : "2018-02-13 13:19:30", "model" : "dSignal caught, exiting!
orbell#1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 3, "data" : "0"}]}
{"time" : "2018-02-13 13:19:30", "model" : "doorbell#1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 1, "data" : "0"}]}`

also seeing
`User cancel, exiting...`

Command line; 

`rtl_433 -R 0 -X "doorbell#1:OOK_PWM:186:251:752,match={20}c1a7d8,countonly" -X "doorbell#3:OOK_PWM:186:251:752,match={20}fb39d8,countonly" -F json`

zuckschwerdt, 2018-02-13:
that's the safe-guard timeout when the rtlsdr driver locks up (i.e. no more data is recieved). I mostly see this with hardware problems (dongle overheating).
Is this consistent on every reception? Does it work again if you just restart rtl_433? Maybe try to increase the timeout in https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L663 and report back.

jonathanmbradshaw, 2018-02-13:
I'll do some testing and come back to you.  FYI when running with no signal input the code seems to be happy running forever.  All the Best Jonathan

zuckschwerdt, 2018-02-13:
Strange. Is this on embedded hardware, maybe something slow like a Raspi 1? Could be that there is too much latency or lag under full cpu or bus load.

jonathanmbradshaw, 2018-02-13:
Looks like I have work to do on the filter (-X), I was not expecting to see the doorbell#3 in the console output following.

Yep this is on a rPi 

`sudo rtl_433 -R 0 -X "doorbell#1:OOK_PWM:186:251:752,match={20}c1a7d8,countonly" -X "doorbell#3:OOK_PWM:186:251:752,match={20}fb39d8,countonly" -F json
[sudo] password for jmb:
Disabling all device decoders.
Registering protocol [1] "General purpose decoder 'doorbell#1'"
Registering protocol [2] "General purpose decoder 'doorbell#3'"
Registered 2 out of 99 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Elonics E4000 tuner
Using device 0: Terratec T Stick PLUS
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.

{"time" : "2018-02-13 15:31:44", "model" : "doorbell#1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 1, "data" : "0"}]}

-- 90 lines removed --
{"time" : "2018-02-13 15:31:45", "model" : "doorbell#1", "count" : 1, "num_rows" : 1, "rows" : [Signal caught, exiting!
{"len" : 2, "data" : "0"}]}
{"time" : "2018-02-13 15:31:45", "model" : "doorbell#1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 1, "data" : "0"}]}

-- x00 lines removed --
{"time" : "2018-02-13 15:31:50", "model" : "doorbell#1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 1, "data" : "0"}]}

-- x00 lines removed --
{"time" : "2018-02-13 15:31:55", "model" : "doorbell#3", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 1, "data" : "0"}]}

User cancel, exiting...
Reattached kernel driver
`

zuckschwerdt, 2018-02-13:
Well, `match={20}fb39d8`is really specific and should never match randomly. (Btw `fb39d` is already 20 bits, so either `match={20}fb39d` or `match=fb39d` will also do.)

I'm really interested if increasing the timeout helps, and what amount is needed. I remember another user with this kind of problem, but he never followed up.

jonathanmbradshaw, 2018-02-13:
I'll get the timeout change made and let you know how it works out.  Also I'll try running with just one filter and see if that helps.  BTW Test files for the transmitter I am using are available at "Byron-BY34 Doorbell" in the Test repo.

jonathanmbradshaw, 2018-02-14:
Do you have any guidance on the new OOK_PWM parameter?
'Bad flex spec, missing gap limit!'

zuckschwerdt, 2018-02-14:
`gap_limit` is the maximum spacing (gap) between pulses.
If doorbell#1 is `tests/Byron-BY34/01/gfile008.cu8` please note that all timings are given in µs! With `-A` output:
```
Pulse width distribution:
 [ 0] count:  650,  width:   128 [127;135]	( 512 us) <-- short
 [ 1] count:  400,  width:   253 [251;257]	(1012 us) <-- long
Gap width distribution:
 [ 0] count:  600,  width:   244 [242;247]	( 976 us)  <-- gap if much bigger
 [ 1] count:  400,  width:   120 [117;122]	( 480 us)
 [ 2] count:   49,  width:   746 [744;748]	(2984 us) <-- reset if much bigger
```
i.e. gap = 1200, reset = 3300, so
```
-X "doorbell#1:OOK_PWM:500:1000:3300:1200,match={20}c1a7d,countonly"
```
gives
```
{"time" : "@1.572864s", "model" : "doorbell#1", "count" : 25}
```

jonathanmbradshaw, 2018-02-14:
Thanks for that.  Tested and working, with and without the timer update.  Great job, let me know if you need any more testing.  All the Best Jonathan

closed 2018-02-14 by jonathanmbradshaw

zuckschwerdt, 2018-02-15:
The Byron-BY34 seems to be the same as Byron-BY101 from https://github.com/merbanan/rtl_433_tests/pull/210

```
rtl_433 -q -F json -A -r tests/Byron-BY101/01/g001_433.92M_250k.cu8
```

gives the timing:

```
Pulse width distribution:
 [ 0] count:  700,  width:   121 [117;125]	( 484 us) <-- short symbol
 [ 1] count:  350,  width:   246 [242;249]	( 984 us) <-- long symbol
Gap width distribution:
 [ 0] count:  650,  width:   251 [249;257]	(1004 us) <-- long gap, gap limit somewhat bigger
 [ 1] count:  350,  width:   127 [125;132]	( 508 us)
 [ 2] count:   49,  width:   753 [751;757]	(3012 us) <-- packet gap, reset limit somewhat bigger
```

e.g.
```
rtl_433 -q -F json -R 0 -X "Byron-BY101:OOK_PWM:500:1000:3100:1200" -r tests/Byron-BY101/01/g001_433.92M_250k.cu8
```

gives
```
{"time" : "@1.572864s", "model" : "Byron-BY101", "count" : 25, "num_rows" : 25, "rows" : [{"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}, {"len" : 21, "data" : "e768f8"}]}
```