merbanan/rtl_433 #685: Some flex improvement suggestions

mzealey, 2018-02-16:
Hi there, as pr for doorbell was rejected in #684 I wanted to suggest a few ways that flex currently falls short of custom driver (parsing byron test data with `-X 'Byron-BY#1:OOK_PWM:500:1000:3300:1200,match={2}0x3,repeats>=4,bits=21'`):

1. Doesn't output anything in CSV mode. Domoticz uses this mode to read in data from rtl433 so certainly for my use-case I need it but doesn't seem like it would be impossible to implement which leads to
2. Ways of extracting certain parts of data into certain fields. This would enable you to specify eg 'id field is bits 3-8' or 'status flag is bits 10-13'. This could be specified like `field=id:3-8,field=status:10-13` (example, not correct for byron)
3. Probably the reason for 1 not being implemented is because of lack of suppression of duplicate lines. This one repeats > 25 times, all are captured into 1 packet. Perhaps there should be some flag to say `print_repeats=1` to only print 1 of the repeats in the packet (why you would want more than 1 I'm not sure but default of 0 could do all, and 2, 3 etc limit to that number of repeats printed)

As a further suggestion for usability, rather than a manual lookup table of flex functions, a new option that lets you refer to a flex function by name and loads it from the lookup table. Eg then anyone with byron doorbell could do `-X Byron-BY101` and it would use the predefined flex function. This would then have all the functionality of a device but without any device-specific code.

Thanks for this great program & feature!

Mark

zuckschwerdt, 2018-02-16:
Great ideas!
(1) is a limitation on nested output in CSV, i guess. I.e. `countonly` should work. I'll look for a work-around.
(2) yes, declarative destructuring for simple cases would be very nice.
(3) it's somewhat vague what a duplicate is. There might be an optional trailing bit and such. But I want something like "only the one best match" soon.
(4) this feature just started with the `decoders.cfg`.

mzealey, 2018-02-16:
For (3) perhaps just start with the usecase of first N matching rows within each packet

zuckschwerdt, 2018-02-16:
@mzealey csv output now works.

roger-, 2018-07-27:
+1 on being able to specify bit fields (or masks, perhaps something like `field=status:0x01,field=id:0xF0`). Suppressing duplicates would be very useful too. 

Currently it seems you can use `countonly` and define decoders with different names for every unique bit pattern, but that doesn't seem very efficient.

zuckschwerdt, 2018-12-09:
The getters are now included and I'm working on an easier format spec, similar to [BitBench](http://triq.net/bitbench). Reopen or start a new ticket if anything is still missing or could be improved. Thanks!

closed 2018-12-09 by zuckschwerdt