merbanan/rtl_433 #690: quhwa doorbell not decoding any more

peteakalad, 2018-02-17:
Have a quhwa doorbell which used to decode on old build of rtl_433 but unfortunately on latest build from master (as of 17 Feb 18) it does not decode.

I've tried to change the pulse lengths with no luck.

I've attached a file to this fork (PR raised);
https://github.com/peteakalad/rtl_433_tests/blob/master/tests/quhwa/03/g009_433.92M_250k.cu8

Dump from -a -t for this file;

*** signal_start = 2087317, signal_end = 2499256
signal_len = 411939,  pulses = 912
Iteration 1. t: 174    min: 83 (412)    max: 265 (500)    delta 6586
Iteration 2. t: 174    min: 83 (412)    max: 265 (500)    delta 0
Pulse coding: Short pulse length 83 - Long pulse length 265

Short distance: 168, long distance: 722, packet distance: 1624

p_limit: 174
bitbuffer:: Number of rows: 25
[00] {19} 1b c1 e0 : 00011011 11000001 111
[01] {18} 37 83 c0 : 00110111 10000011 11
[02] {18} 37 83 c0 : 00110111 10000011 11
[03] {19} 1b c1 e0 : 00011011 11000001 111
[04] {18} 37 83 c0 : 00110111 10000011 11
[05] {18} 37 83 c0 : 00110111 10000011 11
[06] {18} 37 83 c0 : 00110111 10000011 11
[07] {18} 37 83 c0 : 00110111 10000011 11
[08] {18} 37 83 c0 : 00110111 10000011 11
[09] {18} 37 83 c0 : 00110111 10000011 11
[10] {19} 2b c1 e0 : 00101011 11000001 111
[11] {18} 37 83 c0 : 00110111 10000011 11
[12] {18} 37 83 c0 : 00110111 10000011 11
[13] {18} 37 83 c0 : 00110111 10000011 11
[14] {18} 37 83 c0 : 00110111 10000011 11
[15] {18} 37 83 c0 : 00110111 10000011 11
[16] {18} 37 83 c0 : 00110111 10000011 11
[17] {18} 37 83 c0 : 00110111 10000011 11
[18] {18} 37 83 c0 : 00110111 10000011 11
[19] {18} 37 83 c0 : 00110111 10000011 11
[20] {18} 37 83 c0 : 00110111 10000011 11
[21] {19} 36 c1 e0 : 00110110 11000001 111
[22] {18} 37 83 c0 : 00110111 10000011 11
[23] {19} 37 81 e0 : 00110111 10000001 111
[24] {18} 3f f3 f8 : 00111111 11110011 11
... Maximum number of rows reached. Message is likely truncated.
signal_bszie = 917504  -      sg_index = 2097152
start_pos    = 935278  -   buffer_size = 3145728
*** Saving signal to file g009_433.92M_250k.cu8
*** Writing data from 935278, len 917504

zuckschwerdt, 2018-02-17:
I'll look into this. The sample files are welcome but please use only the issues here for discussion not the readme files in the tests repo.

peteakalad, 2018-02-17:
Cheers.  Sure - updated my PR to tests repo.

I'm decoding other devices ok btw.

peteakalad, 2018-02-17:
Further testing on latest no issue.  Closing.

closed 2018-02-17 by peteakalad

zuckschwerdt, 2018-02-17:
I updated the Quhwa device anyway. Should be a bit more stable now, see #691

peteakalad, 2018-02-17:
cheers!