merbanan/rtl_433 #692: New Device: Decoding Conrad Pool Thermometer

Philippovitsch, 2018-02-19:
Hi, I've taken some sample files for this new device some days ago.

Got an idea of extracting the temperature, attached a file with my thoughts. I've marked the needed bits red. The rest of the message are the same bits everytime, wheter I re-insert the batteries or pressing the reset button etc., except the last 4 bits.

The message is sent 3 times (1st time is shorter/ with less information, the 2nd and 3rd messages are equal to each other)

[Decoding.pdf](https://github.com/merbanan/rtl_433/files/1737533/Decoding.pdf)

zuckschwerdt, 2018-02-19:
You can use `-A` to get an idea of the timings (or measure in Audacity) and try `-X` to get a simple starting point for a decoder.

Philippovitsch, 2018-02-20:
Ok thank you!
Dug a little bit into it and used `-A` to get some timings:

Detected OOK package	@ @1.048576s
Analyzing pulses...
Total count:   91,  width: 187945		(751.8 ms)
Pulse width distribution:
 [ 0] count:   60,  width:   493 [492;494]	(1972 us)
 [ 1] count:   31,  width:  1500 [1499;1502]	(6000 us)
Gap width distribution:
 [ 0] count:   57,  width:  1440 [1336;1461]	(5760 us)
 [ 1] count:   31,  width:   486 [482;514]	(1944 us)
 [ 2] count:    2,  width:  7319 [7319;7320]	(29276 us)
Pulse period distribution:
 [ 0] count:   88,  width:  1952 [1830;2014]	(7808 us)
 [ 1] count:    2,  width:  7813 [7812;7814]	(31252 us)
Level estimates [high, low]:  16050,   2232
Frequency offsets [F1, F2]:   27529,      0	(+105.0 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
Attempting demodulation... short_limit: 996, long_limit: 1462, reset_limit: 7321, sync_width: 0
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 3 
[00] {19} c4 dd e0 : 11000100 11011101 111
[01] {36} fe d6 62 6e f0 : 11111110 11010110 01100010 01101110 1111
[02] {36} fe d6 62 6e f0 : 11111110 11010110 01100010 01101110 1111

so i read out the following timings:
-short: 1972us
-long: 6000us
-reset: 29276us
-gap: 5760us

tried to get some output with `-X conrad:OOK_PWM:1972:6000:29200:5900` gave me:

Input format: uint8
@1.048576s :	test 	:	1 	:	1 	:	[	:	19 	:	c4dde] 	:	[{19}c4dde]
@1.048576s :	test 	:	1 	:	1 	:	[	:	36 	:	fed6626ef] 	:	[{36}fed6626ef]
@1.048576s :	test 	:	1 	:	1 	:	[	:	36 	:	fed6626ef] 	:	[{36}fed6626ef]

Don't know if that's good or not, I will try to analyze some more files the next few days, if it's correct.

zuckschwerdt, 2018-02-20:
Round the reset limit up, e.g. 30000. I think that `-F json` is easier to read.
The sample is very close to the bandwidth limit (spectrogram shows it at 434.035MHz, see http://triq.net/iqs) so there is some distortion and the first half of the 3 repeats can't be read. But the other 2 repeats look good.

Philippovitsch, 2018-02-20:
Thank you, `-X conrad:OOK_PWM:1972:6000:30000:5900 -F json` gave me:

Input format: uint8
{"time" : "@1.048576s", "model" : "conrad", "count" : 3, "num_rows" : 3, "rows" : [{"len" : 19, "data" : "c4dde"}, {"len" : 36, "data" : "fed6626ef"}, {"len" : 36, "data" : "fed6626ef"}], "codes" : ["{19}c4dde", "{36}fed6626ef", "{36}fed6626ef"]}
Test mode file issued 2 packets

Philippovitsch, 2018-03-01:
Ok, as shown on triq.net the message is repeated 3 times (1st time with 11 short and 1 long pulses, the other 2 times with 7 short and 1 long pulses at the beginning), but I don't know why the first time it has so much distortion.
I've also tested the device with other software/ hardware and the result was the same - the first message is cut-off at the beginning.

zuckschwerdt, 2018-03-02:
the 11 vs 7 shouldn't be a problem, it's likely a preamble meant to warm-up /sync the receiver. I guess thats why the first one is longer. I plan to add logic to the decoder to strip this sync preamble. You should be able to see it as 0's by flipping the bits (`invert` option:
 `-X 'conrad:OOK_PWM:1972:6000:30000:5900,invert'`). Anyway, you can always ignore the first broken packet and use the two left ;)

Philippovitsch, 2018-03-02:
ah, very cool :)
Jep, the `invert` option is is doing it's job well.
This device decoding seems to be similar to the Bresser 3 Ch., with which I'm experimenting a little bit right now (a little bit training for me, as my programming skills are very very basic) :)

merbanan, 2018-12-04:
No sample signals referenced in the issue, nothing more can be done. Closing issue.

closed 2018-12-04 by merbanan