merbanan/rtl_433 #698: Problem decode PIR

michel67260, 2018-03-21:
hello,

i have pir, rtl 433 see it "Kerui PIR Sensor", but another 433 reader see it in EV 1527... now rtl 433 see it "Wireless Smoke and Heat Detector GS 558".... i have tested this line "sudo rtl_433 -R 79 -R 11 -R 12 -R 30 -R 68 -R 1" work one time see pir sensor "Kerui PIR Sensor",  but not work now just detect "Wireless Smoke and Heat Detector GS 558"....  with this line "rtl_433 -R 79 -R 11 -R 12 -R 30 -R 86"

I have accurite out door and Oregon in door, and WH0530, and another Pir sensor in EV 1527...

thk's for reply and help

zuckschwerdt, 2018-03-21:
Kerui and GS558 are also x1527 type devices. Collisions are likely. Upload some samples (-a -t) so we can check.

michel67260, 2018-03-21:
hello,
thk's for reply, only with parameter "-a -t" ?

zuckschwerdt, 2018-03-21:
I always use `-q -F json -C si` -- but that won't be useful here. Use `-f 433.8M` or such if needed. Also `-s 1024k` for a better time resolution and wider band if needed. Drop your sample files in http://triq.net/iqs to check if they are good. (Compare to files from https://github.com/merbanan/rtl_433_tests/tree/master/tests/Kerui-D026/01 and https://github.com/merbanan/rtl_433_tests/tree/master/tests/smoke_gs558/01 maybe. E.g. https://github.com/merbanan/rtl_433_tests/blob/master/tests/waveman/01/gfile016.cu8 is really bad, at least 6 overlapping signals...)

merbanan, 2018-12-06:
These devices should be handled with the flex decoder. And no samples was referenced, closing issue.

closed 2018-12-06 by merbanan

kbahey, 2019-10-05:
Thanks for the reply. 
I did run `rtl_433 -A`
It is nothing I can understand, nor there is a -X line suggested.
Not sure how to proceed on this. 
Do you have a link to a tutorial somewhere that I can follow?

zuckschwerdt, 2019-10-06:
To get an idea what's in your signal: https://github.com/merbanan/rtl_433/wiki/View-a-saved-cu8-file
and https://github.com/merbanan/rtl_433/wiki/digitech_xc0324.README#discover-the-radio-encoding-ie-modulation-protocol