merbanan/rtl_433 #701: Keelog is is supported?

bilbolodz, 2018-04-01:
Keelog is is supported?

zuckschwerdt, 2018-04-01:
Needs more info. What is Keelog? Do have sample data?

bilbolodz, 2018-04-01:
https://en.wikipedia.org/wiki/KeeLoq

bilbolodz, 2018-04-01:
I can provide sample data if you need.

merbanan, 2018-04-01:
Sample data would be nice.

merbanan, 2018-04-01:
https://web.archive.org/web/20070423090909/http://www.keeloq.boom.ru/decryption.pdf

merbanan, 2018-04-01:
https://web.archive.org/web/20120722105816/http://cryptolib.com/ciphers/keeloq/KeeLoq.c

merbanan, 2018-04-01:
This is the relevant info with regards to Keeloq. So we have source code and specifications, we are missing sample signals and keys.

alberto-fc, 2018-07-06:
Hi,

I've a very rought (ugly as burning trees) implementation with keeloq parsing/decoding (in a very very old rtl_433 version). But I can provide you with sample signals and keys (I already have this working, but it's just a ninja hack).

I'm not familiar with rtl_433 (I patched the code on-the-fly just to adapt it to my needs) and the steps to provide the relevant info but I'd like to see this done so I only need some guide or steps to make the capture process and I can digest the raw and make sense to anyone from there.

Unfortunately -for me- I'm very well informed about Keeloq.

I've access to anything from HCS301 to HCS362, PIC implementations and for 433MHz and 868 (the fields and the data are the same, just a change in the duty cycle, actually you can change the duty cycle on 433MHz but it's very rare to find in the field).

Though there is a wide difference on timings for keeloq implementations, but the preamble phase solve should be good enough to identify keeloq and timings. In other words, I've a decoder mixed with rtl_433 code so it can be done :)

Thank you!

merbanan, 2018-07-06:
For samples signals follow this guide:
https://github.com/merbanan/rtl_433#supporting-additional-devices-and-test-data

Regarding the code, just send whatever you have in the form you can and we'll help out.

alberto-fc, 2018-07-06:
Well, I'm blind:

https://github.com/merbanan/rtl_433#supporting-additional-devices-and-test-data

It seems the way to go.

zuckschwerdt, 2018-07-06:
Not having looked into KeeLoq I'd say we could resonably add a demodulator to output the raw bitstream. But the processing and application of keys is something I rather see in a flexible script (perhaps python). It sounds like an involved project, like HAM packet radio. And there are great tools for that, wouldn't want to reinvent them. Though I might be completly wrong.
Note that we don't have infrastructure to pass configuration info to decoders and the demodulators assume a fixed and predetermined timing.

merbanan, 2018-12-07:
As always sample signals are needed to support a protocol. Closing issue.

closed 2018-12-07 by merbanan