merbanan/rtl_433 #702: Maverick ET-73 (not 733)

sumptersmartt, 2018-04-08:
I've had this wireless thermometer for a while now, and have used rtl_433 in the past. I'd like to submit the following data for the maverick ET-73 dual probe wireless thermometer. If I run "rtl_433 -G", the software seems to think it's a different thermometer, but the values are all over the place and obviously incorrect:
2018-04-08 14:33:17 :   Thermopro TP12 Thermometer
        Id:              159
        Temperature 1 (Food):    186.4 C
        Temperature 2 (Barbecue):        213.7 C

I am happy to provide any additional data. I have attached .cu8 files in a zip and a .txt file with the associated readings (in F). The receiving unit has the ability to change between C and F but the transmitter displays only in F. 1 is Food and 2 is Smoker, the same as the TP12 mentioned above.

This piece from the instructions may be relevant. I'm not sure if I caught any sort of sync signal but if I did it's probably in the first file:
Turn Transmitter on by sliding switch on the rear of the transmitter from OFF to ON. This needs to be
done within 60 seconds of turning on the receiver in order for the receiver to register the signal from the
transmitter. If it doesn’t beep within 5-10 seconds slide switch OFF wait a few seconds and then switch back
ON. Pressing and holding the TX button for 2 seconds will do the same thing as turning transmitter OFF
and then back ON

http://www.maverickhousewares.com/et-73-redichek-remote-smoker-thermometer/
https://amazingribs.com/thermometers/maverick-et-73-remote-smoker-thermometer

[maverick-et73-readme.txt](https://github.com/merbanan/rtl_433/files/1887537/maverick-et73-readme.txt)
[ET-73-samples.zip](https://github.com/merbanan/rtl_433/files/1887533/ET-73-samples.zip)

merbanan, 2018-12-04:
Support added in commit https://github.com/merbanan/rtl_433/commit/897239280b35154155537a7bc568f9c6eea955c0

closed 2018-12-04 by merbanan

zuckschwerdt, 2018-12-05:
If possible please collect and post about 100-1000 lines of unique bitrows, there is a chance that I can recover the checksum/CRC/LFSR.
Stick `bitbuffer_print(bitbuffer);` in the top of `maverick_et73_sensor_callback()` and collect unique lines like `{48} 73 10 91 0a c0 b8`.

merbanan, 2018-12-05:
I only implemented support based on the sample data files provided in this ticket.

zuckschwerdt, 2018-12-05:
:) @sumptersmartt if you can post a good number of samples I'll try to figure out the checksum.

sumptersmartt, 2019-02-25:
@zuckschwerdt this fell off my radar but I'll try to collect some this week. Good excuse to cook some ribs!