merbanan/rtl_433 #706: Toyota TPMS sensor

yuret, 2018-04-16:
Hi!
I have some problems decoding data from Toyota TPMS (Factory installed on Toyota Auris).
Received data:

16.04.2018 18:31 | Toyota | TPMS | 5f66cb42 | 8b160043 | CRC
16.04.2018 18:32 | Toyota | TPMS | 5f64fec0 | 8b160043 | CRC
16.04.2018 18:32 | Toyota | TPMS | 5f64fee7 | 74e8ffbc | CRC
16.04.2018 18:32 | Toyota | TPMS | 5f64fed4 | 8b170043 | CRC

But my sensors ID is:
0D68164
0D681BE
0D5AEE3
0D681A0

Can we do something to get correct decoded data?
Thank you.

yuret, 2018-04-17:
Sampled data will be very similar to what we have on screenshots.
If you need any additional data please let me know.

[image]
[image]

[433.92M_250k.zip](https://github.com/merbanan/rtl_433/files/1918747/433.92M_250k.zip)

zuckschwerdt, 2018-04-17:
Very nice! Is this some official Toyota "Tire Assistant"-App? It connects directly to the car's tire controller by BT?

yuret, 2018-04-17:
No, its unofficial. App made by very talented gays from France. They also have very useful app Hybrid Assistant.
 it's connected to car ECU via BT-OBD adapter.

zuckschwerdt, 2018-04-17:
In that sense it's "official" data published by the on-board controller. Pefect as reference to decode the signal.

yuret, 2018-04-17:
In that sense - yes. We have something to compare with.

yuret, 2018-04-17:
May be this also can be useful:
https://github.com/xnk/pacific-tpms

zuckschwerdt, 2018-04-17:
I know about that project, but the PMV-107J is different from this PMV-C210 (one byte longer and different CRC poly).
Actually the four decoded bistreams you captured are
```
0101111101100100111111101100000010011110001111110000000001101001
0101111101100100111111101101010010011100001111101111111110010010
0101111101100100111111101110011101100000111010101111111110010100
0101111101100110110010110100001010011100110000110000000001101100
```
playing with the bits (invert, reverse, shift, ...) and comparing to the labels didn't yield a match :/

yuret, 2018-04-17:
Here somebody have luck to decode it:
http://forum.rav4driversclub.com/thread/74/rav4-receiving-decoding-tpms-valve

Main lines:
Pacific Industries PMV-C210 (Toyota part number 42607-02031)

434MHz FSK transmission. 10KHz bit rate, 30 kHz deviation. **Biphase Mark Coding.**

8ms approx transmission time, 72 data bits. Repeat approx 90s, while stationary. Moving as yet unknown. 30 rapid transmissions on deflation.

4 Sensor ID bytes (4 top bits presently unused? – set to F).

1 single bit always “1” (Or a status bit?)

Pressure Data Byte = 4*Pressure in PSI + 29 (decimal) offset (To be confirmed)

Temperature Data Bye = Temperature in C + 40

7 bits always “0” (or status bits TBD). Bit 2 (i.e. value 4) seems to get set on low pressure/deflation.

Pressure Data Byte, with bits inverted

Checksum byte (CRC8)

zuckschwerdt, 2018-04-17:
Oh well, now it clicks into place. I did wonder why parts of the code randomly looked inverted. It's [DMC](https://en.wikipedia.org/wiki/Differential_Manchester_encoding) not [MC](https://en.wikipedia.org/wiki/Manchester_code).

yuret, 2018-04-18:
Sorry for being unpatient, but can you tell if there are any progress?

zuckschwerdt, 2018-04-18:
with the new info it basically works. I need to integrate DMC code for FSK but other than that it's no big task. Expect a working output maybe next week. I'll perhaps open a testing branch for you today.

yuret, 2018-04-18:
Thank you very much.

zuckschwerdt, 2018-04-18:
You can test the feat-pmvc210 branch: https://github.com/zuckschwerdt/rtl_433/tree/feat-pmvc210

yuret, 2018-04-18:
Thank you again!

yuret, 2018-04-18:
All looks like true. Great job. Thank you

[image]

zuckschwerdt, 2018-04-18:
If you can, try to test and compare the pressure reading with values from the app. 4*PSI+29 sounds wrong, it would allow a range of -7.25 to 56.5 -- is that sensible? The test code assumes kPa with -68 offset (not great either). A log of extreme values would allow a better guess.

yuret, 2018-04-20:
Tomorrow i will deflate one tire completely and pump it to 3.0 bars with logging. Temperature luck quite accurate.

yuret, 2018-04-21:
Formula look like must be **kPA/1.72+28** . Please take a look and correct me if I'm wrong.
[Logs_and_Calc.xlsx](https://github.com/merbanan/rtl_433/files/1934714/pressure.xlsx)
[rtl433-raw.log](https://github.com/merbanan/rtl_433/files/1934666/rtl433-raw.log)

zuckschwerdt, 2018-04-21:
That 1.72 is very close to exactly a quarter psi (in kPa): 6.894757 /4 = 1,72368925. The sensor might report in 1/4th psi, offset by -7 though.

merbanan, 2018-04-22:
Maybe the device doesn't have a floating point unit. Then it should be possible to derive the formula by that. But this as is looks really good too.

zuckschwerdt, 2018-04-23:
This will go in after #711 and use 1/4 -7 PSI.

merbanan, 2018-05-01:
Can we close this soon?

zuckschwerdt, 2018-05-01:
I'll push merge the changes in now.