merbanan/rtl_433 #710: Battery Status and booleans for data_make

rct, 2018-04-22:
Summary - inconsistencies with data output for a simple field like `battery_low` which may be compounded by the lack of a boolean output data type.

(I thought this was discussed already but I didn't find it.  Close if this is a duplicate issue.)

There is a bit of inconsistency in the structured data field for battery OK vs. LOW.  Many if not most decoders are outputting the strings `OK` or `LOW` using a name of either `battery` or `battery_low`.  Some decoders use an `int` for the battery status. 

It appears there is no boolean type for `data_mate()`.  In a number of cases it might make things clearer than using an int or a string.   Is the lack of a boolean type intentional for some reason?

For some devices like security sensors, there are a lot of strings used where a bunch of booleans would be more succinct.

Note: I'm not advocating a rationalization of all of the existing battery fields right now,  This something to possibly tackle as a 1.0 or later release when we just bite the bullet and make a bunch of wholesale changes..  However, it might be good to have some guidance for what new decoders should use.

For reference here is a survey I recently did of some temperature/humidity sensor output:

Acurite devices mostly agree on using `battery` and `OK` or `LOW`. The tower sensor uses `battery` and an int, which is ambiguous.  (This was my fault.)
```
{"time" : "@0.000000s", "model" : "Acurite 5n1 sensor", "sensor_id" : 839, "channel" : "A", "sequence_num" : 0, "battery" : "OK", "message_type" : 49, "wind_speed_mph" : 2.679, "wind_dir_deg" : 337.500, "wind_dir" : "NNW", "rainfall_accumulation_inch" : 0.000, "raincounter_raw" : 66}
{"time" : "@0.000000s", "model" : "Acurite tower sensor", "id" : 12053, "channel" : "C", "temperature_C" : 26.700, "humidity" : 74, "battery" : 0, "status" : 68}
{"time" : "@0.000000s", "model" : "Acurite 609TXC Sensor", "id" : 202, "battery" : "OK", "status" : 2, "temperature_C" : 26.200, "humidity" : 76}
{"time" : "@0.000000s", "model" : "Acurite Rain Gauge", "id" : 162, "rain" : 513.500}
{"time" : "@0.524288s", "model" : "Acurite 606TX Sensor", "id" : -93, "battery" : "OK", "temperature_C" : -12.800}
{"time" : "@0.000000s", "model" : "00275rm", "probe" : 0, "id" : 3253, "battery" : "OK", "temperature_C" : 16.700, "humidity" : 23, "mic" : "CRC"}
```

A sampling of other devices (that have a humidity field to limit it to just weather/environment):

```
{"time" : "@0.000000s", "model" : "Ambient Weather F007TH Thermo-Hygrometer", "device" : 37, "channel" : 5, "battery" : "Low", "temperature_F" : 67.800, "humidity" : 35}
{"time" : "@0.000000s", "model" : "LaCrosse TX141TH-Bv2 sensor", "id" : 67, "temperature_C" : 9.300, "humidity" : 73, "battery" : "OK", "test" : "No"}
{"time" : "@0.000000s", "brand" : "OS", "model" : "THGR122N", "id" : 106, "channel" : 2, "battery" : "OK", "temperature_C" : 28.300, "humidity" : 26}
{"time" : "@0.524288s", "model" : "AlectoV1 Temperature Sensor", "id" : 44, "channel" : 1, "battery" : "OK", "temperature_C" : 28.800, "humidity" : 36, "mic" : "CHECKSUM"}
{"time" : "@1.048576s", "model" : "Bresser 3CH sensor", "id" : 230, "channel" : 1, "battery" : "OK", "temperature_F" : 68.600, "humidity" : 65, "mic" : "CHECKSUM"}
{"time" : "@0.524288s", "model" : "Esperanza EWS", "id" : 8, "channel" : 1, "temperature_F" : 39.400, "humidity" : 1}
{"time" : "@0.000000s", "model" : "HIDEKI TS04 sensor", "rc" : 4, "channel" : 1, "battery" : "OK", "temperature_C" : 24.100, "humidity" : 65}
{"time" : "@0.000000s", "model" : "Fine Offset Electronics, WH2 Temperature/Humidity sensor", "id" : 209, "temperature_C" : 24.600, "humidity" : 33, "mic" : "CRC"}
{"time" : "@0.000000s", "model" : "Telldus/Proove thermometer", "id" : 167, "temperature_C" : 24.200, "humidity" : 53, "mic" : "CRC"}
{"time" : "@0.000000s", "model" : "Fine Offset Electronics WH1080/WH3080 Weather Station", "msg_type" : 0, "id" : 133, "temperature_C" : 5.600, "humidity" : 97, "direction_str" : "NE", "direction_deg" : "45", "speed" : 0.000, "gust" : 0.000, "rain" : 335.100, "battery" : "OK"}
{"time" : "@1.048576s", "model" : "inFactory sensor", "id" : 198, "temperature_F" : 66.500, "humidity" : 77}
{"time" : "@0.000000s", "model" : "Fine Offset WH5 sensor", "id" : 3, "temperature_C" : 25.700, "humidity" : 55, "mic" : "CRC"}
{"time" : "@0.524288s", "model" : "TFA-Twin-Plus-30.3049", "id" : 4, "channel" : 0, "battery" : "OK", "temperature_C" : 14.100, "humidity" : 6}
{"time" : "@0.524288s", "model" : "Oregon Scientific SL109H", "id" : 188, "channel" : 3, "temperature_C" : 20.800, "humidity" : 64, "status" : 4, "mic" : "CHECKSUM"}
{"time" : "@0.524288s", "model" : "Prologue sensor", "id" : 9, "rid" : 78, "channel" : 1, "battery" : "OK", "button" : 0, "temperature_C" : 1.000, "humidity" : 204}
{"time" : "@1.048576s", "model" : "Mebus/433", "id" : 24, "battery" : "LOW", "channel" : 4, "unknown1" : 1, "unknown2" : 11, "temperature_C" : -150.300, "humidity" : -79}
{"time" : "@0.000000s", "model" : "WT450 sensor", "id" : 13, "channel" : 1, "battery" : "OK", "temperature_C" : 25.640, "humidity" : 14}
```

Security decoders:
```
{"time" : "@2.097152s", "model" : "Interlogix", "id" : "200054", "device_type" : "unknown", "raw_message" : "105048", "battery" : "LOW", "switch1" : "CLOSED", "switch2" : "CLOSED", "switch3" : "OPEN", "switch4" : "OPEN", "switch5" : "CLOSED"}
{"time" : "@1.572864s", "model" : "X10 Security", "id" : "f58e", "code" : "84", "event" : "DS10A DOOR/WINDOW CLOSED"}
{"time" : "@0.000000s", "model" : "Honeywell Door/Window Sensor", "id" : 645640, "channel" : 8, "event" : 64, "state" : "closed", "heartbeat" : "no"}

merbanan, 2018-04-22:
Most devices has a battery low indicator. That is then lit when a bit is set. So my thought is that it should be named battery_low and when set to 0, the battery isn't low and to 1 when it is.

zuckschwerdt, 2018-04-23:
There is some discussion in PR 655 https://github.com/merbanan/rtl_433/pull/655

I really want to hear that popular consumers (TSDBs) can handle booleans well beforehand. Otherwise an int 0/1 is more flexible and something that can easily be stored and graphed. Also that leaves room for precentages (0.00-1.00) if needed.

Ideally a device would report a normalized percentage of battery time remaining. In that sense `battery: 1` (100%) to `battery: 0` (0%) would be natural (maybe with steps between). And since it is more likely an "alarm" flag the opposite is also natural (what @merbanan said).
And that this issue is far from an obvious solution shows in the divergence in devices ;)

Maybe there could be a transformation, similar to °F/°C. This could be introduced without touching every device.

rct, 2018-04-23:
@zuckschwerdt, Thanks.  I don't know why my search didn't turn up PR #655.  (Maybe I was only searching issues.) 

The point about booleans and time-series storage and analysis is an excellent one.  I suspect many time series systems wouldn't naturally handle booleans.  

While 0 and 1 are good approximations for a boolean, how does a consumer know that this is really a boolean type instead of an `int` or `float`?.  The values of 'true' and 'false' would be unambiguous.

merbanan, 2018-04-23:
{"time" : "@0.524288s", "model" : "Acurite 606TX Sensor", "id" : -93,

negative id ?

{"time" : "@0.000000s", "model" : "Acurite 5n1 sensor", "sensor_id" : 839,

sensor_id -> id ?

rct, 2018-04-24:
First, @zuckschwerdt, Thoughts, comments?   Does this seem reasonable to you?

I think PR #655 adds DATA_BOOL.  It currently has some conflicts since things have moved forward since thien.  Is this PR reasonable (other than we'd like to ad a conversion option to coerce `bool` to `int` for consumers that can handle `bool`?

@merbanan -  regarding the inconsistencies you pointed out above.
> {"time" : "@0.524288s", "model" : "Acurite 606TX Sensor", "id" : -93,
> negative id ?

The 606 is one of the Acurite decoders I didn't implement.  That output is from running -G -F json on all the files in rtl_433_tests.  tests/acurite/05 generated those files. Clearly something is broken there.   Looking at the code `sensor_id` is defined as `int8`.  Changing that to `uint8` might fix that.   Acurite uses parity bits on data bytes for a number of their devices, though on those they aren't using parity for the ID bytes to have 8 bits available.   I have no details on that device.  So I have no idea if that is parity or an ID > 127.

> {"time" : "@0.000000s", "model" : "Acurite 5n1 sensor", "sensor_id" : 839,
> sensor_id -> id ?

Agree for consistency that should change.  But given the age of that decoder, I'm loathe to make people have to change their parsers just for the sake of one consistency improvement without any other benefit to them.   At a minimum, It would be good to make this change at the same time as other consistency changes, such as the battery flag as part of a 1.0 release.

One idea is let's create an issue (or set of issues) with the list of stuff that should change at the time of release?  Possibly use github labels to categorize the issues "1.0 release", "future".  Also it probably wouldn't hurt to use labels to triage the open issues as 'decoder issues', 'core', 'bug', 'enhancement'. ...

Thoughts?

zuckschwerdt, 2018-04-24:
I'd say that the data_make() instances in many devices would benefit from more options if there are practical rules and examples how to use. It's guesswork to find out how to output a flag, switch or even a channel number (0- /1-based, letters?).
I think booleans are not the whole solution. Your ad-hoc list gives a good glimpse of the challenge -- there are binary states that don't map well to true/false without a bias or context. Perhaps we need some kind of enums (with an appropriate, user-selectable mapping to 0/1, t/f)?

On the matter of actual output I see `0` and `1` as universal, where `false`, `False`, `FALSE`, `F`, `No`, `NO`, `N`, `Off`, ... (you get the point) I always need to lookup. What will the e.g. CSV output be?

Another wild idea: the flexible decoder will some day run from a config file (so people can easily spec their doorbell, window alarms, smoke detectors). Also MQTT output needs some kind of templating to be useful (the mqtt topic isn't that simple to map). If those are in place, maybe that is also a solution to map battery vs. battery_low, open/close vs 0/1, etc...
That way we don't need to offer the (arguably) best output, but only simple consistent one.

TL;DR if 0/1 can't be read as f/t, n/y right now than the challenge is something else, we need to think bigger ;)

gdt, 2018-05-23:
I don't see 0/1 vs false/true (following the C/C++ standards?) as a big deal, but adopting a coding convention that in a 0/1 field that 0 is false  and 1 is true seems useful.  The real consistency issue is about "battery_low" vs "battery_ok" and "battery_fraction", and I don't think 1/true really bears on that much.

Overall,  I think we have to say that most devices of interest  have a "battery_low" indicator which is asserted when low (since it's an alarm condition), and therefore should go with a generic battery_low field that each driver transforms to.

zuckschwerdt, 2018-05-23:
Yes, and that fits nicely with the idea of triggering events downstream to alert the user of a need for a battery replacement. But still there are devices that actually transmit a battery gauge, perhaps in mV.
On the other hand we could generalize to a `battery_level` float with values from 1.0(good) to 0.0(battery low) and also use that for genuine battery levels. But it is additional processing of raw device data and we usually don't want that obfuscation.

Maybe just a "use battery_low: 1 where applicable" rule of thumb for now.

rct, 2018-05-23:
@zuckschwerdt wrote:

> On the matter of actual output I see 0 and 1 as universal, where false, False, FALSE, F, No, NO, N, Off, ... (you get the point) I always need to lookup. What will the e.g. CSV output be?

Correct me if I'm wrong, but JSON (one of the 3 output types) has boolean as a fundamental type.   

The JSON output can output true/false which should be understood by the majority of JSON parsers.  

CSV, (and KV?) can coerce BOOLs to INTs  if that makes it easier.

zuckschwerdt, 2019-09-05:
With the `newmodel` option all devices now output `battery_ok` with a value `0 – 1`. Sufficiently easy to understand and machine readable (add the level of all your sensors and get a gauge how much juice is left in your sensor network). We might add conversions once the `newmodel` change is final.

closed 2019-09-05 by zuckschwerdt