merbanan/rtl_433 #712: feat-soapysdr branch: Hackrf not decoding/detecting any devices

SoliForte, 2018-04-25:
Recently built the feat-soapysdr branch, and using the rtl-sdr or the hackrf, neither radio seems to detect or decode any of the nearby devices, however, switching to master with just the rtl I can get detections. I saw that in January there was mention of merging feat-soapysdr and was wondering if there was any movement on that...

or, perhaps some ideas as to why I'm not getting any demod of devices?

Thanks in advance.

zuckschwerdt, 2018-04-25:
I guess you would need to set the gain values. Also the value range is not optimal and (depending on the hardware) you will need a very good signal.

zuckschwerdt, 2018-05-20:
Did you solve this with setting gain values?

rigid, 2018-12-04:
I successfully detected devices with the HackRF one. The relevant changes i made to the example .rtl_433.conf:

`device          "driver=hackrf"                   # default device "" did not work.`
`gain              "LNA=40,AMP=14,VGA=30"`
`sample_rate    1000k`

where:
* AMP = RF (0-14 dB)
* LNA = IF (0-40 dB, 8 dB steps)
* VGA = BB (0-62 dB, 2 dB steps)`

zuckschwerdt, 2018-12-04:
Great! Thanks for the feedback!

closed 2018-12-04 by zuckschwerdt

y0d4a, 2020-10-28:
> I successfully detected devices with the HackRF one. The relevant changes i made to the example .rtl_433.conf:
> 
> `device "driver=hackrf" # default device "" did not work.`
> `gain "LNA=40,AMP=14,VGA=30"`
> `sample_rate 1000k`
> 
> where:
> 
>     * AMP = RF (0-14 dB)
> 
>     * LNA = IF (0-40 dB, 8 dB steps)
> 
>     * VGA = BB (0-62 dB, 2 dB steps)`
> 
> 
> Then I launched: `rtl_433 -G`

can you be more precise ? :)
i make rtl_433.conf with data which you said and i am unable to run it , where i am wrong? :)

tnx.

zuckschwerdt, 2020-10-29:
@y0d4a if SoapySDR is working (check `SoapySDRUtil --make`) then setting the `device` / `-d` is all that's needed. Please try a few things and www.catb.org/~esr/faqs/smart-questions.html#beprecise

y0d4a, 2020-10-29:
yeap, with -d "" i successful start it...
thank you!

flaotte, 2021-09-02:
> "LNA=40,AMP=14,VGA=30"

I try to set it this way, but only VGA is set. How can I turn on AMP?

zuckschwerdt, 2021-09-05:
The gain (`-g`) parameter is passed off to SoapySDR per element when using a SoapySDR device using `SoapySDRDevice_setGainElement()` https://github.com/merbanan/rtl_433/blob/master/src/sdr.c#L695
You should see verbose information on that ("Setting gain element..." / "WARNING: setGainElement ... failed").