merbanan/rtl_433 #719: Bresser 5-in-1 Weather Central

palstek, 2018-05-13:
I would like to add support for this sensor device and captured several files with 'rtl_433 -a -t -f 868300000 -s 500000'. Some of them could be pulse analysed with rtl_433 and give some reasonable output. I uploaded one of those cu8 file to https://dk-ware.de/nextcloud/index.php/s/5jBoGnTqKJdYo8Y 

```
$ rtl_433 -A -r g055_868.3M_500k.cu8 
Test mode active. Reading samples from file: g055_868.3M_500k.cu8
Input format: uint8
Detected FSK package	@ @0.000000s
Analyzing pulses...
Total count:   61,  width:  7224		(28.9 ms)
Pulse width distribution:
 [ 0] count:    1,  width:     0 [ 0; 0]	(   0 us)
 [ 1] count:   41,  width:    29 [29;31]	( 116 us)
 [ 2] count:    9,  width:    58 [58;58]	( 232 us)
 [ 3] count:    5,  width:    87 [87;87]	( 348 us)
 [ 4] count:    4,  width:   260 [232;289]	(1040 us)
 [ 5] count:    1,  width:   635 [635;635]	(2540 us)
Gap width distribution:
 [ 0] count:    5,  width:   255 [230;288]	(1020 us)
 [ 1] count:   41,  width:    28 [28;29]	( 112 us)
 [ 2] count:    5,  width:    86 [86;86]	( 344 us)
 [ 3] count:    9,  width:    57 [57;58]	( 228 us)
Pulse period distribution:
 [ 0] count:    9,  width:   286 [241;318]	(1144 us)
 [ 1] count:   27,  width:    57 [57;59]	( 228 us)
 [ 2] count:   12,  width:   115 [115;116]	( 460 us)
 [ 3] count:   11,  width:    86 [86;87]	( 344 us)
 [ 4] count:    1,  width:   721 [721;721]	(2884 us)
Level estimates [high, low]:  15867,      5
Frequency offsets [F1, F2]:   11559, -18677	(+44.1 kHz, -71.2 kHz)
Guessing modulation: Pulse Code Modulation (Not Return to Zero)
Attempting demodulation... short_limit: 29, long_limit: 29, reset_limit: 29696, sync_width: 0
pulse_demod_pcm(): Analyzer Device 
bitbuffer:: Number of rows: 1 
[00] {286} 00 aa aa aa aa aa 2d d4 ed 93 7f eb bf ee ef fb fc d6 bf ff ff 12 6c 80 14 40 11 10 04 03 29 40 00 00 00 00 

Test mode file issued 1 packets
```

Now I thought, I could use the flex decoder to apply similar settings to other capturings, but I failed even on this one:

```
$ rtl_433 -X bresser:FSK_PCM:20:20:2969 -f 868290000 -s 1024000 -R 0 -r g055_868.3M_500k.cu8 
Registering protocol [1] "General purpose decoder 'bresser'"
Disabling all device decoders.
Registered 1 out of 101 device decoding protocols
Test mode active. Reading samples from file: g055_868.3M_500k.cu8
Input format: uint8
@0.000000s :	bresser 	:	1 	:	1 	:	[	:	640 	:	000000000000fc0fc0fc1f81f81f83f03f03f03f07e07e07e0fc0fc0fc1f81f81f81f80003f03ffc1ffff03f07e003fffe07ff83ffc007e003ffc1ffffffffffffff81f83fffe07ffffffffffff81fff] 	:	[{640}000000000000fc0fc0fc1f81f81f83f03f03f03f07e07e07e0fc0fc0fc1f81f81f81f80003f03ffc1ffff03f07e003fffe07ff83ffc007e003ffc1ffffffffffffff81f83fffe07ffffffffffff81fff]
Test mode file issued 1 packets
```

The decoded hex string is not the same as with the pulse analyzer. How do I need to configure the flex decoder to get the same result as the pulse analyzer?

My plan was to use the flex decoder and then to play with the weather station to reverse engineer meaning of the hexstring.

Any help is appreciated.

cu,
  Daniel

arpu, 2018-05-14:
https://www.bresser.de/en/Weather-Time/7002520000000.html
do you have this device?

arpu, 2018-05-14:
for this device i get 
`` arpu@localhost  ~/Work/githubsources/rtl_433/build   master  sudo /usr/local/bin/rtl_433   -f 868300000 -s 500k -F json -R 0  -X 'bresser:FSK_PCM:58:58:600'                                                                                                5 ↵  10312  18:53:53 
Disabling all device decoders.
Registering protocol [1] "General purpose decoder 'bresser'"
Registered 1 out of 101 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
Invalid sample rate: 500000 Hz
WARNING: Failed to set sample rate.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 868300000 Hz.
{"time" : "2018-05-14 18:54:06", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 186, "data" : "00aaaaaaaaaa2dd4effebfffcbfffdf8ffb6f93ffff1000"}], "codes" : ["{186}00aaaaaaaaaa2dd4effebfffcbfffdf8ffb6f93ffff1000"]}
{"time" : "2018-05-14 18:54:06", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 13, "data" : "a000"}], "codes" : ["{13}a000"]}
{"time" : "2018-05-14 18:54:06", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 14, "data" : "d000"}], "codes" : ["{14}d000"]}
{"time" : "2018-05-14 18:54:06", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 47, "data" : "81c00920d800"}], "codes" : ["{47}81c00920d800"]}
{"time" : "2018-05-14 18:54:18", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 199, "data" : "00555555555516ea76febfffcbfffdf91fb6f93ffff1202800"}], "codes" : ["{199}00555555555516ea76febfffcbfffdf91fb6f93ffff1202800"]}
{"time" : "2018-05-14 18:54:18", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 14, "data" : "d000"}], "codes" : ["{14}d000"]}
{"time" : "2018-05-14 18:54:18", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 46, "data" : "81b81241b000"}], "codes" : ["{46}81b81241b000"]}
{"time" : "2018-05-14 18:54:30", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 187, "data" : "00555555555516ea77ff5fffe5fffefc7fdb7c9ffff8800"}], "codes" : ["{187}00555555555516ea77ff5fffe5fffefc7fdb7c9ffff8800"]}
{"time" : "2018-05-14 18:54:30", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 13, "data" : "a000"}], "codes" : ["{13}a000"]}
{"time" : "2018-05-14 18:54:30", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 14, "data" : "d000"}], "codes" : ["{14}d000"]}
{"time" : "2018-05-14 18:54:30", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 47, "data" : "81c00920d800"}], "codes" : ["{47}81c00920d800"]}
{"time" : "2018-05-14 18:54:42", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 199, "data" : "00555555555516ea76febfffcbfffdf91fb6f93ffff1202800"}], "codes" : ["{199}00555555555516ea76febfffcbfffdf91fb6f93ffff1202800"]}
{"time" : "2018-05-14 18:54:42", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 14, "data" : "d000"}], "codes" : ["{14}d000"]}
{"time" : "2018-05-14 18:54:42", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 46, "data" : "81b81241b000"}], "codes" : ["{46}81b81241b000"]}
{"time" : "2018-05-14 18:54:54", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 187, "data" : "00555555555516ea77ff5fffe5fffefc7fdb7c9ffff8800"}], "codes" : ["{187}00555555555516ea77ff5fffe5fffefc7fdb7c9ffff8800"]}
{"time" : "2018-05-14 18:54:54", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 13, "data" : "a000"}], "codes" : ["{13}a000"]}
{"time" : "2018-05-14 18:54:54", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 14, "data" : "d000"}], "codes" : ["{14}d000"]}
{"time" : "2018-05-14 18:54:54", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 47, "data" : "81c00920d800"}], "codes" : ["{47}81c00920d800"]}
{"time" : "2018-05-14 18:55:06", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 188, "data" : "00555555555516ea77ff5fffe5fffefc7fcdbe4ffffc400"}], "codes" : ["{188}00555555555516ea77ff5fffe5fffefc7fcdbe4ffffc400"]}
{"time" : "2018-05-14 18:55:06", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 13, "data" : "a000"}], "codes" : ["{13}a000"]}
{"time" : "2018-05-14 18:55:06", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 14, "data" : "d000"}], "codes" : ["{14}d000"]}
{"time" : "2018-05-14 18:55:06", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 47, "data" : "81c00920d800"}], "codes" : ["{47}81c00920d800"]}
{"time" : "2018-05-14 18:55:18", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 187, "data" : "00555555555516ea77ff5fffe5fffefc7fdb7c9ffff8800"}], "codes" : ["{187}00555555555516ea77ff5fffe5fffefc7fdb7c9ffff8800"]}
{"time" : "2018-05-14 18:55:18", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 13, "data" : "a000"}], "codes" : ["{13}a000"]}
{"time" : "2018-05-14 18:55:18", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 14, "data" : "d000"}], "codes" : ["{14}d000"]}
{"time" : "2018-05-14 18:55:18", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 47, "data" : "81c00920d800"}], "codes" : ["{47}81c00920d800"]}
[... cut ...] 77ff5fffe5fffefc7fdb7c9ffff8800"]}
{"time" : "2018-05-14 18:56:18", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 13, "data" : "a000"}], "codes" : ["{13}a000"]}
{"time" : "2018-05-14 18:56:18", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 14, "data" : "d000"}], "codes" : ["{14}d000"]}
{"time" : "2018-05-14 18:56:18", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 47, "data" : "81c00920d800"}], "codes" : ["{47}81c00920d800"]}
``

palstek, 2018-05-14:
I have this device https://www.bresser.de/en/Weather-Time/Weather-Stations/BRESSER-Weather-Center-5-in-1.html . But I think the outdoor sensor from arpu looks the same.

palstek, 2018-05-14:
@zuckschwerdt: Your comment helped me a lot. Now I got further. Yesterday, I captured a lot of samples. With your settings of flex decoder, rtl_433 can decode all of those samples. Even those, where the pulse analyzer had no clue.

Now I see another issue: The preamble is sometimes 55 instead of aa. See https://dk-ware.de/nextcloud/index.php/s/P3ZyY5NFRnSwTR8 for example. It seems that the data bits are shifted by one bit. Do I need to solve this in the packet decoder or is this a "bug" in pulse_demod_pcm()?

zuckschwerdt, 2018-05-14:
You should not use `-s 500k`, let alone 48k. Use the default (of 250k) or 1024k for more resolution/bandwidth if needed.
The misaligned preamble is not a problem and shifts of 1-3 bits are expected. I might add an option to align someday though.
When you figure out the proper decoding the alignment is usually done in the decoder. Flex is just for testing and first results here.

arpu, 2018-05-15:
ok i using the default  and get this values looks good ? 
 https://gist.github.com/arpu/60d9497cafd4bd46e14b3ce796e41565

zuckschwerdt, 2018-05-15:
yes, but no matter the sample rate, the flex parameter are always in µs. You should use
```
rtl_433 -q -F json -R 0 -X 'bresser:FSK_PCM:58:58:600'
```
(this assumes that the original was indeed at 500k, which translates periods of 29 samples to 58 µs. If you look in the captured samples at default 250k the pulses should be about 14 samples long now.)

palstek, 2018-05-15:
A have it running: https://github.com/palstek/rtl_433/tree/feature/bresser_5in1

I have best decoding results with 500k sample rate. If I use 250k, I get the following debug output from my decoder. It seems pulse_demod_pcm() or the baseband processor didn't get/decode the entire packet as a single packet.
```
bresser_5in1_callback bit_per_row 145 out of range
bresser_5in1_callback bit_per_row 1 out of range
```

Nevertheless, rtl_433 is a really great project. I'm astonished how fast it is possible to implement a custom decoder without deep knowledge in radio signal demodulation.

arpu, 2018-05-15:
@palstek i testing your branch with 
sudo /usr/local/bin/rtl_433 -f 868300000  -s 500k -R 102                                                                                                                         5 ↵  10048  23:10:18 
Registering protocol [1] "Bresser Weather Center 5-in-1"
Registered 1 out of 102 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
Invalid sample rate: 500000 Hz
WARNING: Failed to set sample rate.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 868300000 Hz.

but i do not get any decodings :/ any idea whats wrong?

zuckschwerdt, 2018-05-16:
@palstek I'm curious what dongle you are using. For me 500k isn't a valid sample rate, it's 250k, 1000k, 1024k,...

zuckschwerdt, 2018-05-16:
Fascinating. I have the current 20180501 of rtl-sdr here and some different dongles, R820T and R820T2. Mostly NooElec NESDR Mini look-alikes (in blue and black), also genuine and NooElec NESDR SMArt as well. Never seen 500k work.

palstek, 2018-05-16:
@zuckschwerdt: You're right. 500k is really an invalid sample rate. Maybe my old rtl-sdr version just doesn't refuse this invalid sample rate, but makes crazy things. If I use the default 250k sample rate, I need to adjust the short and long limits to 124 µs.

zuckschwerdt, 2018-05-20:
There is now also a `preamble` option for the flex decoder to help with future endeavors. E.g.
```
rtl_433 -q -F json -R 0 -X 'bresser:FSK_PCM:124:124:1500,preamble=5554' \
 -y "{187}00555555555516ea77ff5fffe5fffefc7fdb7c9ffff8800 {186}00aaaaaaaaaa2dd4effebfffcbfffdf8ffb6f93ffff1000"
```
gives
[…]
```
"codes" : [
 "{137}5ba9dffd7fff97fffbf1ff6df27fffe2000",
 "{137}5ba9dffd7fff97fffbf1ff6df27fffe2000"]}
```

merbanan, 2018-12-09:
@palstek can you submit a good set of samples? I want to merge your decoder.

merbanan, 2018-12-09:
@brainexe or maybe you could submit some sample signals for the test repository?

zuckschwerdt, 2018-12-12:
You use `-f 868.272M`, are you sure your device is within 868.15M to 868.39M?

iMarkus, 2018-12-12:
I have this device
https://www.bresser.de/en/Weather-Time/Weather-Stations/BRESSER-Weather-Center-5-in-1.html

Regarding andrea fabrizi it should be in this range (https://github.com/andreafabrizi/BresserWeatherCenter)

iMarkus, 2018-12-12:
This is my Realtek device:

https://www.amazon.de/gp/product/B013Q94CT6/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1

merbanan, 2018-12-12:
Try with -s1000k and -A. You might have issues with locating the signal in the spectrum.

merbanan, 2018-12-12:
Maybe play with the gain also (-g).

iMarkus, 2018-12-12:
`rtl_433 -f 868272000 -F json -s100k -A` seems like to work

Detected OOK package    @ 2018-12-12 17:55:06
Analyzing pulses...
Total count:   23,  width:  2094                ( 2.1 ms)
Pulse width distribution:
 [ 0] count:    6,  width:    48 [42;57]        (  48 us)
 [ 1] count:    6,  width:    17 [16;18]        (  17 us)
 [ 2] count:    2,  width:    29 [29;30]        (  29 us)
 [ 3] count:    4,  width:   116 [109;123]      ( 116 us)
 [ 4] count:    4,  width:    74 [64;84]        (  74 us)
 [ 5] count:    1,  width:   227 [227;227]      ( 227 us)
Gap width distribution:
 [ 0] count:    1,  width:    12 [12;12]        (  12 us)
 [ 1] count:    8,  width:    35 [34;36]        (  35 us)
 [ 2] count:   11,  width:    22 [22;24]        (  22 us)
 [ 3] count:    1,  width:    47 [47;47]        (  47 us)
 [ 4] count:    1,  width:    61 [61;61]        (  61 us)
Pulse period distribution:
 [ 0] count:    5,  width:    56 [51;66]        (  56 us)
 [ 1] count:    4,  width:    82 [78;92]        (  82 us)
 [ 2] count:    5,  width:    39 [39;40]        (  39 us)
 [ 3] count:    5,  width:   144 [131;159]      ( 144 us)
 [ 4] count:    2,  width:   104 [104;105]      ( 104 us)
 [ 5] count:    1,  width:   288 [288;288]      ( 288 us)
Level estimates [high, low]:   1006,    172
Frequency offsets [F1, F2]:  -17254,      0     (-263.3 kHz, +0.0 kHz)
Guessing modulation: No clue...

Detected OOK package    @ 2018-12-12 17:55:06
Analyzing pulses...
Total count:    1,  width:   754                ( 0.8 ms)
Pulse width distribution:
 [ 0] count:    1,  width:   754 [754;754]      ( 754 us)
Gap width distribution:
Pulse period distribution:
Level estimates [high, low]:   1814,     14
Frequency offsets [F1, F2]:  -23036,      0     (-351.5 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...

merbanan, 2018-12-12:
This looks like noise, try with -g 35-49 also and see if that helps.

iMarkus, 2018-12-12:
Now it prints the following:

Detected OOK package    @ 2018-12-12 17:59:21
Analyzing pulses...
Total count:    2,  width: 66702                (66.7 ms)
Pulse width distribution:
 [ 0] count:    1,  width: 37210 [37210;37210]  (37210 us)
 [ 1] count:    1,  width: 27236 [27236;27236]  (27236 us)
Gap width distribution:
 [ 0] count:    1,  width:  2256 [2256;2256]    (2256 us)
Pulse period distribution:
 [ 0] count:    1,  width: 39466 [39466;39466]  (39466 us)
Level estimates [high, low]:  15435,     32
Frequency offsets [F1, F2]:    7876,      0     (+120.2 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with fixed gap
Attempting demodulation... short_limit: 32223, long_limit: 2257, reset_limit: 2257, sync_width: 0
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {2} 40 : 01

merbanan, 2018-12-12:
I meant -g [35-45] what you get is still not good enough signal data.

iMarkus, 2018-12-12:
Thanks for your help!!!

Hope it's ok to print the messages here:

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 868272000 Hz.

Detected OOK package    @ 2018-12-12 18:02:50
Analyzing pulses...
Total count:   23,  width:  2090                ( 2.1 ms)
Pulse width distribution:
 [ 0] count:    4,  width:    45 [43;51]        (  45 us)
 [ 1] count:    6,  width:    19 [17;23]        (  19 us)
 [ 2] count:    2,  width:    33 [33;33]        (  33 us)
 [ 3] count:    4,  width:   118 [110;125]      ( 118 us)
 [ 4] count:    3,  width:    80 [71;85]        (  80 us)
 [ 5] count:    3,  width:    60 [59;61]        (  60 us)
 [ 6] count:    1,  width:   230 [230;230]      ( 230 us)
Gap width distribution:
 [ 0] count:    1,  width:    12 [12;12]        (  12 us)
 [ 1] count:    8,  width:    33 [32;37]        (  33 us)
 [ 2] count:   11,  width:    19 [17;21]        (  19 us)
 [ 3] count:    1,  width:    47 [47;47]        (  47 us)
 [ 4] count:    1,  width:    59 [59;59]        (  59 us)
Pulse period distribution:
 [ 0] count:    5,  width:    73 [63;80]        (  73 us)
 [ 1] count:    5,  width:    38 [37;40]        (  38 us)
 [ 2] count:    3,  width:    53 [52;54]        (  53 us)
 [ 3] count:    5,  width:   144 [131;157]      ( 144 us)
 [ 4] count:    3,  width:   100 [92;105]       ( 100 us)
 [ 5] count:    1,  width:   289 [289;289]      ( 289 us)
Level estimates [high, low]:   1029,    177
Frequency offsets [F1, F2]:  -14084,      0     (-214.9 kHz, +0.0 kHz)
Guessing modulation: No clue...

Detected OOK package    @ 2018-12-12 18:02:50
Analyzing pulses...
Total count:    1,  width:   755                ( 0.8 ms)
Pulse width distribution:
 [ 0] count:    1,  width:   755 [755;755]      ( 755 us)
Gap width distribution:
Pulse period distribution:
Level estimates [high, low]:   1590,     14
Frequency offsets [F1, F2]:  -23434,      0     (-357.6 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...

Detected OOK package    @ 2018-12-12 18:02:52
Analyzing pulses...
Total count:    2,  width:   757                ( 0.8 ms)
Pulse width distribution:
 [ 0] count:    1,  width:   466 [466;466]      ( 466 us)
 [ 1] count:    1,  width:   274 [274;274]      ( 274 us)
Gap width distribution:
 [ 0] count:    1,  width:    17 [17;17]        (  17 us)
Pulse period distribution:
 [ 0] count:    1,  width:   483 [483;483]      ( 483 us)
Level estimates [high, low]:   1048,     19
Frequency offsets [F1, F2]:  -24855,      0     (-379.3 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with fixed gap
Attempting demodulation... short_limit: 370, long_limit: 18, reset_limit: 18, sync_width: 0
pulse_demod_pwm(): Analyzer Device
[... 214 lines cut ...]

Detected OOK package    @ 2018-12-12 18:04:20
Analyzing pulses...
Total count:   10,  width: 190602               (190.6 ms)
Pulse width distribution:
 [ 0] count:    8,  width:  7442 [6499;8923]    (7442 us)
 [ 1] count:    1,  width: 53861 [53861;53861]  (53861 us)
 [ 2] count:    1,  width: 38821 [38821;38821]  (38821 us)
Gap width distribution:
 [ 0] count:    2,  width:  6850 [6759;6942]    (6850 us)
 [ 1] count:    6,  width:  3804 [3281;4259]    (3804 us)
 [ 2] count:    1,  width:  1857 [1857;1857]    (1857 us)
Pulse period distribution:
 [ 0] count:    2,  width: 15171 [14478;15865]  (15171 us)
 [ 1] count:    5,  width: 11070 [10422;11883]  (11070 us)
 [ 2] count:    1,  width:  8356 [8356;8356]    (8356 us)
 [ 3] count:    1,  width: 57732 [57732;57732]  (57732 us)
Level estimates [high, low]:  15868,     13
Frequency offsets [F1, F2]:    7016,      0     (+107.1 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_limit: 7442, long_limit: 53861, reset_limit: 6943, sync_width: 38821
pulse_demod_pwm_precise(): Analyzer Device
bitbuffer:: Number of rows: 2
[00] {9} f7 80 : 11110111 1
[01] {0} :

merbanan, 2018-12-12:
Everything but this looks bogus.

[00] {366} 0d 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 50 78 9d 91 08 1f e3 f7 f7 47 f7 ff 7f cc 77 d7 ff ff f8 07 c7 ff f7 bc 2f f8

This is a real signal. No idea what though.

merbanan, 2018-12-12:
I still think you have the wrong frequency. Or the device uses an unknown protocol.

iMarkus, 2018-12-12:
I managed to get this once:

Detected FSK package    @ 2018-12-12 18:17:49
{"time" : "2018-12-12 18:17:49", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 640, "data" : "1ce1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c38736d8db736398db86e63"}], "codes" : ["{640}1ce1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c3870e1c38736d8db736398db86e63"]}

iMarkus, 2018-12-12:
I tested some of the above settings and it looks better now.

rtl_433 -f 868300000 -s 500k -F json -R 0 -X 'bresser:FSK_PCM:58:58:600'
Disabling all device decoders.
Registering protocol [1] "General purpose decoder 'bresser'"
Registered 1 out of 101 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
Invalid sample rate: 500000 Hz
WARNING: Failed to set sample rate.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 868300000 Hz.

{"time" : "2018-12-12 18:37:15", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 75, "data" : "1f3672988727069b8ce"}], "codes" : ["{75}1f3672988727069b8ce"]}
{"time" : "2018-12-12 18:37:57", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 76, "data" : "47367573264e07918e3"}], "codes" : ["{76}47367573264e07918e3"]}
{"time" : "2018-12-12 18:37:57", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 27, "data" : "6880a20"}], "codes" : ["{27}6880a20"]}
{"time" : "2018-12-12 18:38:39", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 79, "data" : "07b472b4499b81ac31c6"}], "codes" : ["{79}07b472b4499b81ac31c6"]}
{"time" : "2018-12-12 18:38:39", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 27, "data" : "6109a20"}], "codes" : ["{27}6109a20"]}

merbanan, 2018-12-12:
I forgot the decoder is disabled by default. Try to enable it.

iMarkus, 2018-12-12:
I managed to get it work!!! Thank you so much.

What I did now is I cloned the repository from @palstek with the bresser branch and build it.

This is the only command which works so far for me:

`rtl_433 -f 868300000 -s 1000k -F json`

{"time" : "2018-12-12 20:46:12", "model" : "Bresser Weather Center 5-in-1", "id" : 206, "temperature_C" : -0.700, "humidity" : 68, "gust" : 0.000, "speed": 0.000, "direction_deg" : 225.000, "direction_str" : "SW", "rain" : 20.800, "data" : "ec317fff5fffeff8ff97f7ddf713ce8000a00010070068082208"}
{"time" : "2018-12-12 20:46:24", "model" : "Bresser Weather Center 5-in-1", "id" : 206, "temperature_C" : -0.800, "humidity" : 68, "gust" : 0.000, "speed": 0.000, "direction_deg" : 225.000, "direction_str" : "SW", "rain" : 20.800, "data" : "ee317fff5fffeff7ff97f7ddf711ce8000a00010080068082208"}

merbanan, 2018-12-12:
So upstream rtl_433 -R 119 did not work ? Anyway can you run the branch you got working and collect some samples? If you run the palstek tree run it with rtl_433 -s 1000k -a -t.

iMarkus, 2018-12-12:
Sure please find them here and tell me if you need something else:

https://github.com/iMarkus/CU8

I will delete the repository after you downloaded the files.

iMarkus, 2018-12-12:
Oh sorry. Please try again :)

merbanan, 2018-12-12:
The samples you recorded didn't contain the Bresser data. You can test the files yourself with rtl_433 -r "recorded file" if any file decodes into bresser data that is the files we want.

iMarkus, 2018-12-12:
I used the same command but with -a and -t

`rtl_433 -f 868300000 -s 1000k -F json`

`rtl_433 -f 868300000 -s 1000k -a -t`

merbanan, 2018-12-12:
That is ok, but you never know what you are recording. You need to check that afterwards.

iMarkus, 2018-12-12:
It is recording very slow. Got only 4 files right now

iMarkus, 2018-12-12:
Are you sure that this is working with the current feature/bresser_5in1 branch from @palstek ?

merbanan, 2018-12-12:
Yes this is an old feature.

iMarkus, 2018-12-12:
I am very sorry, it works like a charm now but when I use -a -t it doesn't.

Any other suggestion?

merbanan, 2018-12-12:
Maybe add -l 300 ?

merbanan, 2018-12-12:
Or try mainline rtl_433 with -S all -f 868300000 -s 1000k -F json.

iMarkus, 2018-12-12:
-l 300 produces a lot of files, how can I test them quickly?

merbanan, 2018-12-12:
rtl_433 -r *.gfile should do it.

iMarkus, 2018-12-12:
no the command does not work, so far i have no valid sample file :( i will try some other things tomorrow

merbanan, 2018-12-13:
It should decode the signal to the Bresser output.

iMarkus, 2018-12-13:
I tried now several settings and received some signals but none from bresser :(

Do you need them?

@0.000000s :    Efergy e2 CT
        Transmitter ID:  128
        Current:         nan A
        Interval:        6s
        Battery:         LOW
        Learning:        NO

@1.048576s :    TFA-Twin-Plus-30.3049   :       18      :       2
        Battery:         OK
        Temperature:     -38.0 C
        Humidity:        9 %

iMarkus, 2018-12-13:
Also mainline did not work

Test mode active. Reading samples from file: all
Input format: uint8
Test mode file issued 1637 packets

merbanan, 2018-12-13:
This is really strange. Can you try the mainline code with -R 199 -f 868300000 -s 1000k -F json and see if you get any signals?

iMarkus, 2018-12-13:
One additional info: When I clone your repo and build it I only get these results

`rtl_433 -f 868300000 -s 1000k`

2018-12-13 11:41:07 :   Ford    :       TPMS    :       00000000        :       000000  :       CHECKSUM
2018-12-13 11:41:07 :   Renault         :       TPMS    :       000000  :       00      :       0000 0000       :       CRC

`rtl_433 -f 868300000 -s 1000k -F json -R 119`

no output

merbanan, 2018-12-13:
Only other thing I can think about is to play with gain (-g) and see if the signal grabber works better. Try with 35 to 49 or so.

zuckschwerdt, 2018-12-13:
There seem to be too many problems at once here.

First use CubicSDR or Gqrx to reliably spot your transmissions. Note the center frequency, how often the signal appears, if it is strong (bright in the waterfall) and if there is much noise or other signals.

> `rtl_433 -S all -R 199 -f 868300000 -s 1000k -F json` but no signals from bresser found (Filesize was more than 100MB)

That's a really bad sign. It means your receiver was way overloaded. Each single grab file (.cu8) should only be some 128k. If the grab never stops there is a continous signal.

Try to get smaller grabs. Also drop the grabs onto http://triq.net/iqs and compare to what you get in CubicSDR/Gqrx waterfall, or other samples from the tests repo.

iMarkus, 2018-12-13:
To be honest I am totally satisfied with @palstek branch. I just wanted to help and generate some sample files. Unfortunately my current setup is only a console without GUI. Maybe @palstek or @brainexe is able to provide the samples

merbanan, 2018-12-13:
Hmm, maybe we need to add better statistics to the signal grabber, right now is it quite silent. Can you send that 100mbfile though? Might be good enough. I'll really don't want to tell people to use some random fork but without any samples it is hard to debug why upstream rtl_433 isn't working.

zuckschwerdt, 2018-12-13:
@iMarkus you are helping, it's just that we are not certain if there is a good sample that fails or if something in the setup or current rtl_433 is broken ;)
And looking over it, I guess I found a fault in out Bresser 5in1 code (parity check fails on the test sample). I'll push a fix shortly.

iMarkus, 2018-12-13:
alright give me a shot when you pushed the fix, I will then rebuild the stuff and test again

zuckschwerdt, 2018-12-13:
Fixed. Big thanks! I broke it 4 days ago. m(

zuckschwerdt, 2018-12-13:
I'm always the one to talk about integer-promotion rules and then I write stupid stuff like `~some_u8 == some_u8_inverted` (note the `some_u8` gets promoted to int and negating that gives you 32(or such) inverted bits, not 8). Embarrassing.

zuckschwerdt, 2018-12-13:
I guess plain `rtl_433 -s 1024k -f 868.3M -R 119` should work. Maybe even without `-s 1024k`.

[43 more replies]