merbanan/rtl_433 #727: Gather Statistics

zuckschwerdt, 2018-05-23:
Picking up the idea from @rct, capturing statistics for the decoders:

rtl_433 should keep statistics on the decoder operations.
Rough outline how to do that:

For an easy integration the return code from the decoders can be extended with the following semantics:

- each invocation will be accounted
- the number of successful decodes will be counted (return value != 0)
- the number of processed messages will be counted (sum return value if > 0)
- the number of failed decodes will be counted (return value <= 0)
- decode failed with bad message length (return FAIL_LENGTH_ = -1)
- decode failed on early checks, maybe a prefix (return FAIL_EARLY = -2)
- decode failed because of the message integrity check (return FAIL_MIC = -3)
- decode failed because of sanity check  (return FAIL_SANITY = -4)

The statistics could be shown at exit, periodically, at exit or e.g. with SIGUSR1.

This is just a quick idea to get this going. All comments and suggestions welcome. I'll turn this into some experimental code soon.

gdt, 2018-05-23:
For stats, I see this as being like SNMP and counting errors in network packets.   Your list seems like a good starting point and I think we should expect it to change as a few decoders are enhanced.  Two suggestions:
 - there are sometimes fixups to data to make things decode when something is off, or at least comments indicate an extra zero that shouldn't be there, and this is notable
 - the situation of cleanly recognizing a packet as not for us is different from thinking it is and finding a problem.  Specifically I am thinking of length/checksum and then parity/value-range in Acurite, just because that's the code I have read.   Your list captures this, but I wanted to emphasize it.

gdt, 2018-05-23:
Also, there probably should be stats for the demodulators (vs. decoders), and these will be quite different, even if they are integrated into the same stats output method.

zuckschwerdt, 2018-05-23:
Good points. A difference between "decoder not applicable" and "packet broken /decoder failed" should be made clear.
If in some decoders "fixups" are needed that is likely a kludge and best handled with debug printf's -- the root cause needs to be found and solved.

rct, 2018-05-23:
So what I have in mind as the potential use cases:
1. For running (static) tests on recorded data after changes, we should be able to identify and evaluate which decoders are called and the results.  I think this will add another, dimension besides just comparing data output to help evaluate whether the matching is working correctly.
1. For long term runs (i.e. normal usage) - be able to collect data that would let us evaluate whether there are any improvements to be made.  Are tolerances too tight or too loose?  Are decoders not getting called for some buffers where they should?  
1. Are decoders getting to the point where valid packets seem to be received but some sanity check is failing because there is some piece of the protocol that isn't understood yet?

Ideally the statistics would be a first step for the case of problem where a user reports that they have Device X which appears to have a decoder, but they aren't getting any output.

merbanan, 2020-01-06:
@zuckschwerdt can this be closed or is there something missing? Can you elaborate what if so.

zuckschwerdt, 2020-01-06:
Improving statistics is low priority and mostly works already ;)

closed 2020-01-06 by zuckschwerdt