merbanan/rtl_433 #729: Add option to save non-decoded packets during normal operation

gdt, 2018-05-23:
Currently, one can operate normally and get decoded packets, or save packets to files.   It would be useful, in looking for unknown signals, to be able to operate normally, outputting decoded packets, while saving data that was not decoded.

zuckschwerdt, 2018-05-23:
Maybe this can be chained to #727 i.e. if there is a) no decoder, b) decoder fails for reason X then save? I'd find that immensely helpful working on new protocols.

gdt, 2018-05-23:
I think this is separate from stats, which is more or less necessarily about counting events while executing decoders.

What I basically meant is:

if energy is detected (whatever -t does), and no decoder claims to decode, save (like -t does), but otherwise don't save.

(The hard part would seem to be the sliding window decoding at only one offset, but it seems that this is already handled.)

rct, 2018-05-23:
Agreed.  I'd like to be able to do `-A -I 1 -t`  or  just `-I 1 -t` to get "analysis and retention of "unmatched" packets" or "just retention".

gdt, 2018-05-23:
I don't follow what -I does from the help output.  (I realize this is asking for a lot of effort, but it seems this needs a full man page or equivalent, in addition to the help output.)

If "-I 1 -t" means:
  decode packets that are matched by any of the enabled drivers, with normal output
  save any packets that pass energy detect but are not decoded

then that sounds good to me.  Presumably "-I 2 -t" would only save samples that are decoded, which perhaps someone would want.   But the big point is that this should be something that can be left running for weeks, while feeding json to a logger/weewx-injector/etc., in production, and also drop samples of oddiities.   That way more people can be on the lookout for interesting things not yet understood.

rct, 2018-05-25:
Depending on the implementation it might address #697 a request to save all *and* decode.

zuckschwerdt, 2018-12-08:
No need for `-a`, `-A`, `-t`, or `-I` any more. `-S none|all|unknown|known` covers all these cases.

closed 2018-12-08 by zuckschwerdt