merbanan/rtl_433 #732: Unit testing decoders: test scaffolding that can create bitbuffers

rct, 2018-05-24:
A nice to have:  The ability to unit test decoders without having a recorded signal that covers every possible case.  In some cases it isn't feasible to generate and record a signal that exercises all aspects of a decoder.

This could be satisfied with test scaffolding that has the ability to create a bitfuffer and call a decoder. 

Ideally some text description of a bitbuffer would be used that makes it easy to create them as needed from some sort of test/driver script.

[On a separate note: perhaps it is time to add github labels to this repo to help categorize/triage issues for bug, enhancemant, core code, device-specific, etc. ?]

zuckschwerdt, 2018-05-24:
The `[-y <code>] Verify decoding of demodulated test data (e.g. "{25}fb2dd58") with enabled devices` option should be able to provide basic support for this.
It's not obvious how to use this and what paramters to pass. More documentation, examples and maybe a (regression test) script would be appreciated.

rct, 2018-05-29:
Ok - cool - didn't know that was there.  I spent a few minutes, but was unable to make it work.  I'll need to spend some figuring out what I'm doing wrong or why it didn't work for me.

closed 2018-12-05 by zuckschwerdt