merbanan/rtl_433 #740: Failed to open rtlsdr device error

sergray, 2018-05-30:
I got this error, which is clearly a permission issue:
```
usb_open error -3
Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules
Failed to open rtlsdr device #0.
```

Found a solution following http://www.instructables.com/id/rtl-sdr-on-Ubuntu/, but it would be nice to   describe how to create rtl-sdr.rules within the project and provide example file.

merbanan, 2018-05-30:
Can you create a PR ? If not we'll leave this issue open for tracking.

zuckschwerdt, 2018-05-30:
The first two lines of that error message (the instructive part…) are from [librtlsdr](https://github.com/osmocom/rtl-sdr/blob/master/src/librtlsdr.c#L1493). An enhancement of that message would best be placed there.

sergray, 2018-06-04:
@zuckschwerdt thanks for the hint! There's available even an example of the rules https://github.com/osmocom/rtl-sdr/blob/master/rtl-sdr.rules

I'm going to try to see how to resolve it in the best possible way.

zuckschwerdt, 2018-06-04:
Great! This issue and your link will likely also help out if people search for the message here :)

closed 2018-06-04 by zuckschwerdt

lnee94, 2022-06-10:
Note if you run in to this problem, check if the user you are using is in the `plugdev` group

typpos, 2023-04-27:
FWIW this error also occurs when another application uses the device, making he error message quite misleading.

* RTL2832
* Windows 10 Pro
* Airspy SDR# v1.0.0.1910
* rtl_433 v22.11-23-g4c495fd3