merbanan/rtl_433 #743: Feature: output protocol numbers at startup

zuckschwerdt, 2018-06-03:
The `Registering protocol [x]` output at startup should list the protocol numbers as with `-h`, not just an index. See https://groups.google.com/forum/#!topic/rtl_433/OXOC_5qoBKc

merbanan, 2018-06-03:
The index is kinda useless in this case. Better replace it with the real id.

zuckschwerdt, 2018-06-03:
Yes, I'm pretty sure I once made the mistake of using that index with -R myself :) I'll look into it next week.

gdt, 2018-06-03:
Related: having -h output the driver list makes help output way too long.   So I suggest having a command to list the drivers, and to remove the driver list from -h.    I find the list of registering all drivers to be very voluminous on regular startup, so I also suggest omitting the driver list unless specific drivers were requested.  In the default case, a "registered 80 drivers" would suffice.

merbanan, 2018-12-03:
Is this issue completed?

closed 2018-12-03 by zuckschwerdt