merbanan/rtl_433 #745: Add metadata to decodes

gdt, 2018-06-04:
Currently, the decoded output (json in particular) does not have metadata about the decoded packet.  This information is interesting, and seems good to add.  Specifically, I'd like to see:
 - frequency
 - signal to noise ratio (somehow)
 - probably any other decoder parameters that were estimated and used

zuckschwerdt, 2018-06-04:
Understandably. I'm working on some of these, at low priority though.

merbanan, 2018-06-04:
modulation
repeats
bits in message/total
ms length of ones and zeros

I think that if it would be possible to synthesize a signal based on the message and the metadata then that should cover almost all possible use of this

zuckschwerdt, 2018-08-05:
Some meta data stats will be implemented with  #733

frief, 2018-12-04:
having RSSI as metadata would be great. Could e.g. be used to assess antenna polarization of a sender.

closed 2018-12-08 by zuckschwerdt