merbanan/rtl_433 #748: Using rtl_433 with stdin

jlmcgraw, 2018-06-10:
I've got a setup where I'm using a script to pipe rtl_tcp output to stdin of rtl_433
( see [rtl_client.sh](https://github.com/jlmcgraw/rtl_sdr_reader/blob/master/rtl_client.sh) and [both_readers.sh](https://github.com/jlmcgraw/rtl_sdr_reader/blob/master/both_readers.sh) for more detail on how I'm doing this ).  

This allows me to start rtl_tcp once and use it with multiple programs in rotation, each of which runs for a limited amount of time.

In order to support this I had to made some minor modifications to util.c and rtl_433.c
Without the patches, timestamps are in seconds ( because reading from a file is typically for testing ) and durations don't work properly.

These patches work for my purposes but I'm not sure they're the best general approach, perhaps there is a better way to do this?

Either way, here are my patches in the hope that they may be useful to others

> Patch rtl_433.c to honor duration timer settings when reading from a file
  
    @@ -1165,6 +1165,9 @@ int main(int argc, char **argv) {
                     duration = atoi_time(optarg, "-T: ");
                     if (duration < 1) {
                         fprintf(stderr, "Duration '%s' not a positive number; will continue indefinitely\n", optarg);
    +                } else {
    +                    time(&stop_time);
    +                    stop_time += duration;
                     }
                     break;
                 case 'y':

    @@ -1389,7 +1392,7 @@ int main(int argc, char **argv) {
                 rtlsdr_callback(test_mode_buf, n_read, demod);
                 i++;
                 sample_file_pos = (float)i * n_read / samp_rate / 2;
    -        } while (n_read != 0);
    +        } while (n_read != 0 && !do_exit);
     
             // Call a last time with cleared samples to ensure EOP detection
             memset(test_mode_buf, 128, DEFAULT_BUF_LENGTH);  // 128 is 0 in unsigned data

> Patch util.c to make it report sample times properly when reading from
> a file and also make it report local times formatted per ISO 8601

    @@ -139,10 +139,12 @@ char* local_time_str(time_t time_secs, char *buf)
     
         if (time_secs == 0) {
             extern float sample_file_pos;
    -        if (sample_file_pos != -1.0) {
    -            snprintf(buf, LOCAL_TIME_BUFLEN, "@%fs", sample_file_pos);
    -            return buf;
    -        }
             time(&etime);
         } else {
             etime = time_secs;

    @@ -150,7 +152,8 @@ char* local_time_str(time_t time_secs, char *buf)
     
         tm_info = localtime(&etime); // note: win32 doesn't have localtime_r()
     
    -    strftime(buf, LOCAL_TIME_BUFLEN, "%Y-%m-%d %H:%M:%S", tm_info);
    +    // Output local times as ISO 8601
    +    strftime(buf, LOCAL_TIME_BUFLEN, "%FT%T%z", tm_info);
         return buf;
     }

jlmcgraw, 2018-06-11:
One minor clarification, not that it makes much of a difference: What I'm actually doing is using process redirection to treat stdout from the rtl_client.sh script as an input file

Geoff99, 2018-10-24:
Thanks @jlmcgraw  and @zuckschwerdt for this discussion.

I'm just exploring rtl_433 for the first time, and this discussion helped me understand what `local_time_str` is meant to do.

FWIW, just in case anyone else ends up here, in rather than edit `util.c` as suggested, I changed the invocation in my (in development) device from
```
    local_time_str(0, time_str);
```
to
```
    time_t current;
    local_time_str(time(&current), time_str);
``` 
and that had the effect I desired. (In my case I'm piping debug output from runs on multiple saved test files via a `grep` filter to a txt (actually csv) file.  Ideally I want the file name, but use `time_str` as a failsafe when that's not available ).

zuckschwerdt, 2018-10-25:
If you are batch processing, maybe something like this will work?

    for f in *.cu8 ; do echo "Reading from $f"; rtl_433 -q -F json -r "$f" |grep foobar ; done

Or you might want to just grep the filenameoutput too:

    for f in *.cu8 ; do rtl_433 -q -F json -r "$f" |egrep 'foobar|Reading samples from file' ; done

Also `rtl_433` will probably support multiple files on `-r` shortly.

zuckschwerdt, 2018-10-25:
You might even want to use [stat](https://linux.die.net/man/2/stat) and use the `st_mtime` for the timestamp. But I'd say those uses are rather special purpose.

merbanan, 2018-12-03:
Not sure if we want to support stdin input. It might be useful in some kind of proprietary plugin scenario.

merbanan, 2018-12-04:
std input is already working and with the rtl_tcp support shaping up should be unneeded.

zuckschwerdt, 2018-12-05:
Both ideas are interesting but seem unexpected. The question seems to be: does stdin fit with read-file semantics or with the SDR input semantics?
I'm undecided. I try to never use pipes. Let's keep this issue around for later discussion perhaps?

zuckschwerdt, 2018-12-05:
Great project! We will continue working towards an API which I hope can then be used from e.g. Python to fully control rtl_433 :)

zuckschwerdt, 2018-12-06:
I guess adding the time and event limit options to stdin (and read-file) inputs can't hurt much.

Also I'll add the time output selection to `-M`. You'll be able to select absolut time output with `-M time` (preset on live inputs), relativ time with `-M reltime` (preset on files and stdin), change to UTC with `-M utc` and extend the output to microseconds with `-M hires`. All options can be combined or individually deselected with e.g. `-M notime`…

zuckschwerdt, 2018-12-08:
The time reporting option is in #905

zuckschwerdt, 2018-12-09:
The time and event limit options for files (and stdin) went in with de5a7c5

closed 2018-12-09 by zuckschwerdt