merbanan/rtl_433 #749: pearl/freetec thermometer with 2 way communication?

iz3man, 2018-06-14:
that's not really an issue, but maybe still a way to get an answer:
i have a cheap pearl pool thermometer (freetec pt-250) which gets identified as prologue temp sensor.

`2018-06-14 10:55:05 :Prologue sensor :9 :161
Channel: 1
Battery: LOW
Button: 0
Temperature: 25.00 C
Humidity: 0 %
`
the issue i habe: the probe has an LCD where a little "wireless signal symbol" lights up every time the sensor sends updated values. when i'm inside the house and next to my receiver the red light of the sensor lights up as well, and rtl_433 shows the new values.
when i'm out of range the red light does NOT light up, and i don't get updated values.
how can this be? there is no 2-way communication happening that i know of? the dvb-t stick is only able to receive, correct?
could there be some other base station communicating with my sensor?
is this is common thing for 433mHz sensor to use 2-way communication for such non critical data??
i hope someone can help me solve this problem, as for the time being a pool thermometer not working outside in the pool is useless for me :)

merbanan, 2018-06-14:
Either 2-way communication or you interpret the sensor light incorrectly.

zuckschwerdt, 2018-06-14:
"red light on sensor" means base station/wall unit, right? And it lights up when it receives a transmission?
Or is this red light on the temperature sensor probe? It should light up with every sending transmission.
I have never seen a 2-way communication other than in (>20€) networked fire/heat sensors.

iz3man, 2018-06-14:
the red led is on the SENSOR. that's the weird thing. i don't even have a base station. this thermometer has an LCD showing the temp. but it also CAN send the temp to a base station (which i didn't buy as i use rtl_433 to send temperature to homeassistant). 
so in sends an update every 50s (indicated on the lcd with a little wireless symbol). if it's in range (of what so ever device) the red led on the sensor lights up as well. if i move into the garden there is no red light on the sensor coming up, and there is no signal received by rtl_433.
strange but true. i can make a video to make it clear if you wish.

zuckschwerdt, 2018-06-14:
A thorough explanation is good enough ;) Maybe the red led indicates some other stat related to measurements. I don't think any cheap devices have a receiver in the sensor. It's a battery problem (a sender can be activated every minute or 5..., but a receiver needs to be constantly on).
To narrow it down: the rtl-sdr doesn't have any send capability whatsoever.

iz3man, 2018-06-14:
i know. that's what puzzles me. i have several other 433 capable devices around. maybe the sensor connects to one of the other devices. but this would be very unlikely again, as they all use different protocols. a really strange thing i must say.
btw. the receiver (if present) would only be activated during sending to know if the transmission was ok. on the other hand: what for? the manual says nothing about the red led. an information that noone can read is useless?!

iz3man, 2018-06-14:
another strange thing: i got a house made of bricks. the 433 antenna is inside next to one corner of the more or less square house. i can get good reception anywhere in the house (9x9m). EVEN in the 3rd floor!! through TWO concrete floors. but i can NOT get reception outside of the house 3m away from the antenna (only one brick wall and/or a window inbetween).
isn't that REALLY STRANGE??? sorry for becoming OT now, i hope you don't mind. if so please close the thread.

merbanan, 2018-06-14:
To close and you overload the receiver.

zuckschwerdt, 2018-06-14:
As @merbanan said, overload and perhaps excessive aliasing due to a bad antenna. Are you using the antenna that came with the rtl dongle? It's tuned to DVB-T ranges and a true 433 MHz (ISM band) antenna will help.
You might want to watch the signal in CubicSDR or Gqrx. Much easier to make deductions then.

iz3man, 2018-06-14:
@merbanan aeeehhhmm. it DOES work directly next to the receiver. and inside the house. 2 stories up and down. but when i move just outside of the building reception is lost. if i open a window and have direct line of sight it works again.
but i guess i'll have to learn to use cubicsdr to see what really goes on.

iz3man, 2018-06-14:
@zuckschwerdt yes. "original" dvb-t antenna. i already thought it can't be perfect as it's for a different frequency range. i will see if i can find a good 433 antenna for it.

iz3man, 2018-06-15:
i just bought this antenna to see how things improve. 3 weeks delivery and i can post my results ;)

[image]

zuckschwerdt, 2018-06-15:
Looks ok. I also had good results with antenna similar to [Delock 88877 ISM](http://www.bilder.tragant.de/produkte/orig/5a5633e43b04e9.05101718.jpg). The plug won't fit but you can just use the metal rod and replace the one that came with the rtl dongle.

iz3man, 2018-06-15:
ah ok. i have that one too, but was not sure if cutting/soldering the connector may have a bad result on the reception. that one had and SMA connector or something similar with a screw type nut. so you say cutting that off and the connecting it to the connector of the dvbt antenna will work? will try that while waiting for the other antenna

iz3man, 2018-07-21:
zuckschwerdt: i feel quite stupid: after reading your post about the other antenna (which i already had bought some time ago), i noticed the the metal part is screwed on, and both antennas share the same thread *facepalm*. i'll do some test with this antenna now ...

iz3man, 2018-07-21:
at least i got some good results. i can now get some data from one of the two pool sensors.

zuckschwerdt, 2018-07-22:
Oh, I should said to unscrew the actual antenna metal rod, sorry. If you unsolder the cable there should be no harm though. In HF applications it's better not to solder things, there might be unwanted signal reflections.
In e.g. CubicSDR, select the rtl-sdr as input, set the sample rate to 1024k or 2048k and tune to 434 MHz. Watch the waterfall, it should be mostly blue-green. The transmissions will stand out as red streaks. Try a few possible receiver locations and see where you get a good transmission without interference -- maybe some other noise signal is shadowing the reception.

iz3man, 2018-07-22:
thanks christian. unfortunately i can't get any output from cubicsdr. i open the app, choose this

[image]
then, press start. then set sample rate to 1024, and set frequency to 434Mhz. but NOTHING.
but sorry, this should not have turned into a cubicsdr faq thread ;)

iz3man, 2018-07-22:
Thanks. Do i choose FM or AM or anything else? Do i set Frequency or Center Frequency?

zuckschwerdt, 2018-07-22:
FM, AM, … are only if you want to listen to the signal as audio. Use AM, then click where the signal should be to get better feedback. The frequency is just an offset for the display, the center frequency is the actual tuning.

merbanan, 2018-12-09:
This issue is related to signal strength and transmitter details. This is out of scope for rtl_433 thus closing issue.

closed 2018-12-09 by merbanan