merbanan/rtl_433 #757: Leaks with an input-file

gvanem, 2018-07-02:
Using a command like:
```
rtl_433 -F json -r .\rtl_433_tests\tests\TFA-Pool-thermometer-30.3160\04\25.9_ch1_newdev.cu8
```

shows leaks for *all* protocols added using `register_protocol()`. This:
```c
    for (i = 0; i < demod->r_dev_num; i++)
        free(demod->r_devs[i]);
```

isn't called since with a `in_filename`, `exit(0)` was called.
Can we please tidy the `main()` code to free all allocated memory in some common place.?

*PS1.* The leaks were discovered using MSVC's handy `-MDd` mode and the `_CrtMemCheckpoint()` function.

*PS2.*  In installed the `rtl_433_tests` repo under my `rtl_433` root. But `rtl_433_tests/bin/run_tests.py` is *not* able to run under the `rtl_433` root. It must be run under `rtl_433_tests` root.  I'd like to make a PR for that too.

merbanan, 2018-07-02:
Patches are very welcome for this.

gvanem, 2018-07-02:
> Patches are very welcome for this.

That patch would include many things as I've patched `rtl_433.c` heavily;
1) The *Quiet Mode* is not really quiet:
```
rtl_433.-qF csv -r rtl_433_tests\tests\akhan_remote\01\gfile001.cu8
rtl_433: warning: 10 "Acurite 609TXC Temperature and Humidity Sensor" does not support CSV output
rtl_433: warning: 20 "Calibeur RF-104 Sensor" does not support CSV output
rtl_433: warning: 28 "Chuango Security Technology" does not support CSV output
...
```
2) I've patched `run_test.py` too to use `-q` and parse `stdout` only. But I don't understand all these:
```
 values_changed
  * root[0]['time']
  ```
  What does it mean?  With a `rtl_433.exe -qF json -r rtl_433_tests/tests/wg_pb12v1/01/210C.cu8`:
  ```
  Test mode active. Reading samples from file: rtl_433_tests/tests/wg_pb12v1/01/210C.cu8`
  {"time" : "@0.000000s", "model" : "WG-PB12V1", "id" : 8, "temperature_C" : 21.000, "mic" : "CRC"}
  {"time" : "@0.000000s", "model" : "WG-PB12V1", "id" : 8, "temperature_C" : 21.000, "mic" : "CRC"}
  ```
  Looks okay to me.

zuckschwerdt, 2018-07-03:
regarding the quiet mode, I'd say those warning are needed (you are requesting csv ouput where it can not be deliverd) and the best fix would be to enbale csv in those devices.

gvanem, 2018-07-03:
There seems to be a memory-issue with a live SDR capture too. From *Process Explorer's* Private Bytes graph after 5 minutes:
[image]

Memory piles-up, but not leaked after exit. I used this command:<br>
 `rtl_433.exe -q -f432M -f432.2M -f432.4M -f432.6M -f432.8M -f433.0M -f433.2M -f433.2M -f433.4M -H 20`.

Listening on a single frequency, the *Private Bytes* is at a constant 24 MBytes.

zuckschwerdt, 2018-07-03:
That leak sounds rather bad. We aren't doing anything exciting there, basically:
```
            frequency_current = (frequency_current + 1) % frequencies;
            center_frequency = frequency[frequency_current];
            r = rtlsdr_set_center_freq(dev, center_frequency);
```
This might be a leak in rtl-sdr?

gvanem, 2018-07-03:
I see that on Windows, there is no `SIGALRM` handler. Could that be an issue if `rtlsdr_cancel_async(dev)` is not called before calling `rtlsdr_set_center_freq()` again. Looking at `librtlsdr.c`, I see no possibility for leaks.

zuckschwerdt, 2018-07-03:
The `rtlsdr_read_async()` call is blocking, so if the loop continues we already cancelled. The alarm is just a watchdog if the async events stop arriving timely.

gvanem, 2018-07-03:
I changed back to `_RELEASE` / `-MD` mode and see no *staircase graph* as in `_DEBUG`-mode. So those *Private Bytes* must reflect be how `_DEBUG`-mode works. I've really no idea, but there seems to be no leaks in `rtl_433.exe`.

zuckschwerdt, 2018-07-04:
Thanks for taking a look. Zero runtime leaks is important for a long running program. The leaks on exit sure are not the best style but not all that bad. Let's keep this issue open until we get around to refactor the main program flow for a clean exit handling.

gvanem, 2018-07-04:
> Patches are very welcome for this.

Currently, the flow in `main()` isn't so clear. AFAICS, there are these 3 *modus operandi*:
* Offline with an IQ-file.
* Live sync-mode.
* Live asynch-mode.

So how about splitting `main()` up into something like this:
```c
{
  struct dm_state *demod = NULL;
  int    rc = parse_cmd_line();

  if (rc == 0)
  {
    rc = init_demodulator (in_filename, out_filename, &demod);
    if (rc == 0)
    {
      if (demod->in_file)
         rc = run_infile_loop (demod);
      else if (sync_mode)
         rc = run_live_sync_loop (demod);
      else
         rc = run_live_async_loop (demod);
    }
  }
  main_cleanup (demod); // free 'demod' and all output-handlers
  return rc;
}
```
This would need more stuff in `struct dm_state` and more global variables. Any thoughts?

zuckschwerdt, 2018-07-04:
There are pending changes in the soapysdr (and mqtt) branch. Preferably we could address this refactoring afterwards.
With SoapySDR the sync mode will perhaps go away. I'm not too clear on why it would be needed anyway.