merbanan/rtl_433 #764: Add Support for WH24 - Fine Offset HP1000 and its derivatives

Zodiac-69, 2018-07-08:
Please add support for the Fine Offset HP1000 and it's derivatives.
The outdoor unit measures:
1. Wind - Direction, Speed and Gust
2. Rain
3. Temperature and Humidity
4. Light intensity in LUX
5. UV

Attached is the protocol.
[WH24 HF.xlsx](https://github.com/merbanan/rtl_433/files/2173657/WH24.HF.xlsx)

merbanan, 2018-07-08:
Do you have sample signals for this device ?

Zodiac-69, 2018-07-08:
Here is a few i captures a while ago with the SDR unit.
You will see that the 1st bit needs to be dumped, my RFM69 does that
Below them is the hex data from today.
The hex data is the payload excluding the Pr-amble
[WH24 Capture.xlsx](https://github.com/merbanan/rtl_433/files/2173761/WH24.Capture.xlsx)
 and sync word

merbanan, 2018-07-08:
Sorry I was unclear, can you follow this guide:

https://github.com/merbanan/rtl_433#supporting-additional-devices-and-test-data

without test signals it is hard to verify that code is working as intended.

Zodiac-69, 2018-07-09:
Please find attached captured files attached.
There should be about 10, one of them i am not sure of as the timing did not correspond with my RFM69 status. 
[RTL_433_WH24.ZIP](https://github.com/merbanan/rtl_433/files/2177710/RTL_433_WH24.ZIP)

Zodiac-69, 2018-07-12:
Do you require more info?
I can do more sample signals and also give you the decoded real time values if you require them.

zuckschwerdt, 2018-07-12:
The documentation is good. I might implement this someday. It will be some weeks until I can find time for this, though.

Zodiac-69, 2018-07-27:
One more item on this.
Lux to W/m2 = Lux/126
Supplied info from Fine Offset

zuckschwerdt, 2018-09-07:
With the superb documentation it's easy to read the data using the flex decoder.
The 17.241 kbps equals 58 µs bit-width, e.g.:

    rtl_433 -q -F json -R 0 -X 'wh24:FSK_PCM:58:58:1500,preamble=AAAA' -r gfile001.data

shows that we can read data

    d55555555516ea125f8571032704010025000080000047839

Requiring some 0xAA preamble

    rtl_433 -q -F json -R 0 -X 'wh24:FSK_PCM:58:58:1500,preamble=AAAA' -r gfile001.data

shows the expected sync-word

    aaaaaa 2dd4 24bf0ae2064e0802004a00010000008f072

requiring the sync-word as prefix

    rtl_433 -q -F json -R 0 -X 'wh24:FSK_PCM:58:58:1500,preamble=aaaa2dd4' -r gfile001.data

shows the expected plain data

    24bf0ae2064e0802004a00010000008f072

I'll turn this into a proper decoder shortly.

closed 2018-09-08 by zuckschwerdt

zuckschwerdt, 2025-01-02:
@Zodiac-69 I take it those UV Index mapping values are experimental? Could there be some variation or error in the ranges for some numbers? What does `4651-5029  12` and `>=5230  13` mean? It should have been `5030` right?

A linear regression shows `UVI = (value - 57.4) / 384.5` as best fit, would that match your observations close enough? Or maybe even `UVI = value / 384.5`, which then would be about 0.1 to 0.2 index points higher.

Zodiac-69, 2025-01-03:
@zuckschwerdt 
You are correct, it seems like a typo, UV value of >=5030 = 13.