merbanan/rtl_433 #766: Adding 433MHz wireless door/window sensor Sunlit Tech model SL-PA67R ?

vg8020, 2018-07-11:
433MHz Wireless Door/Window Sensor model SL-PA67R
Brand Name: SUNLIT TECH
Model Number: SL-PA67R
Frequency: 433MHz
Code: EV1527

This sensor seems to beam only on door open (no door close).
On door open rtl_433 -G will output (door opened three times):
```
2018-07-11 10:30:32 :   Smoke detector GS 558   :       4889    :       5       :       0
        Raw Code:        626325
2018-07-11 10:30:58 :   Smoke detector GS 558   :       4889    :       5       :       0
        Raw Code:        626325
2018-07-11 10:31:28 :   Smoke detector GS 558   :       4889    :       5       :       0
        Raw Code:        626325
```
Should it be treated as an unsuported sensor and first managed in rtl_433_tests ? As it is no smoke detector at all. Please, advise.

zuckschwerdt, 2018-07-11:
Doorbell-style sensors are supported by the flex decoder. Use the `-A` option and post some output here. I'll explain how to convert to a flex setting then.

vg8020, 2018-07-11:
First two OOK packages...
Detected OOK package    @ 2018-07-11 23:21:31
Detected OOK package    @ 2018-07-11 23:21:43
obtained by pressing a button on the door sensor. The same Smoke detector text is output by rtl_433 -A (and pressing button on the door sensor) and rtl_433 -G (by pressing button on the door sensor or opening door).
```
pi@raspberrypi:~ $ rtl_433 -A
[...]
Detected OOK package    @ 2018-07-11 23:21:31
2018-07-11 23:21:31 :   Smoke detector GS 558   :       4889    :       5       :       0
        Raw Code:        626325
Analyzing pulses...
Total count:  125,  width: 56828                (227.3 ms)
Pulse width distribution:
 [ 0] count:   50,  width:   288 [286;290]      (1152 us)
 [ 1] count:   70,  width:   101 [100;103]      ( 404 us)
 [ 2] count:    5,  width:   135 [134;137]      ( 540 us)
Gap width distribution:
 [ 0] count:   50,  width:    85 [83;87]        ( 340 us)
 [ 1] count:   70,  width:   272 [270;273]      (1088 us)
 [ 2] count:    4,  width:  2842 [2842;2843]    (11368 us)
Pulse period distribution:
 [ 0] count:  120,  width:   373 [372;375]      (1492 us)
 [ 1] count:    4,  width:  2978 [2977;2979]    (11912 us)
Level estimates [high, low]:  15840,    143
Frequency offsets [F1, F2]:    1087,      0     (+4.1 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_limit: 101, long_limit: 288, reset_limit: 2844, sync_width: 135
pulse_demod_pwm_precise(): Analyzer Device
bitbuffer:: Number of rows: 6
[00] {24} 5b 39 b9 : 01011011 00111001 10111001
[01] {24} 5b 39 b9 : 01011011 00111001 10111001
[02] {24} 5b 39 b9 : 01011011 00111001 10111001
[03] {24} 5b 39 b9 : 01011011 00111001 10111001
[04] {24} 5b 39 b9 : 01011011 00111001 10111001
[05] {0} :

Detected OOK package    @ 2018-07-11 23:21:43
Analyzing pulses...
Total count:   55,  width: 26258                (105.0 ms)
Pulse width distribution:
 [ 0] count:   29,  width:   130 [129;133]      ( 520 us)
 [ 1] count:   26,  width:   375 [373;378]      (1500 us)
Gap width distribution:
 [ 0] count:   54,  width:   235 [232;237]      ( 940 us)
Pulse period distribution:
 [ 0] count:   29,  width:   366 [365;369]      (1464 us)
 [ 1] count:   25,  width:   610 [609;612]      (2440 us)
Level estimates [high, low]:  15936,    195
Frequency offsets [F1, F2]:     510,      0     (+1.9 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with fixed gap
Attempting demodulation... short_limit: 252, long_limit: 238, reset_limit: 238, sync_width: 0
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {55} fe 94 c2 5c 75 33 58
```

Following OOK packages obtained by opening door:
```
Detected OOK package    @ 2018-07-11 23:23:41
2018-07-11 23:23:41 :   Smoke detector GS 558   :       4889    :       5       :       0
        Raw Code:        626325
Analyzing pulses...
Total count:  625,  width: 295173               (1180.7 ms)
Pulse width distribution:
 [ 0] count:  250,  width:   287 [284;293]      (1148 us)
 [ 1] count:  350,  width:   100 [98;104]       ( 400 us)
 [ 2] count:   25,  width:   135 [133;139]      ( 540 us)
Gap width distribution:
 [ 0] count:  250,  width:    85 [81;89]        ( 340 us)
 [ 1] count:  350,  width:   272 [269;275]      (1088 us)
[... 167 lines cut ...]

Detected OOK package    @ 2018-07-11 23:24:55
Analyzing pulses...
Total count:   55,  width: 26257                (105.0 ms)
Pulse width distribution:
 [ 0] count:   29,  width:   130 [128;134]      ( 520 us)
 [ 1] count:   26,  width:   375 [373;378]      (1500 us)
Gap width distribution:
 [ 0] count:   54,  width:   235 [232;237]      ( 940 us)
Pulse period distribution:
 [ 0] count:   29,  width:   366 [365;370]      (1464 us)
 [ 1] count:   25,  width:   610 [609;613]      (2440 us)
Level estimates [high, low]:  15934,    177
Frequency offsets [F1, F2]:    -539,      0     (-2.1 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with fixed gap
Attempting demodulation... short_limit: 252, long_limit: 238, reset_limit: 238, sync_width: 0
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {55} fe 94 c2 5c 75 33 58

^CSignal caught, exiting!
Reattached kernel driver
```

Kindly advise.

zuckschwerdt, 2018-07-12:
Lets pick the first set of values. (btw. 24 bits of data seems right. E.g. those very cheap x1527 encoder will produce that.)
Pulse width modulation means you only care about the width of the pulses. You need a short and a long value.
Pick the most prevalent, 1152 µs and 404 µs above. (The 540 µs will be some kind of preamble/sync which we can disregard.)
Now estimate the maximum gap (silence) between data pulses and between repeated transmissions.
I.e. round the 1088 µs up to e.g. 1200 µs and the 11368 µs to 13000 µs.

Now see

    rtl_433 -X help

Pick `OOK_PWM` with 

    short = 404
    long = 1152
    reset = 13000
    gap = 1200

That turns into this line

    rtl_433 -q -F json -R 0 -X 'SL-PA67R:OOK_PWM:404:1152:13000:1200'

Instead of `-R 0` you can of course enable all the other wanted decoders. But don't just enable all of them, there will be false positives then.

Also you can refine the detection by matching the expected output, e.g.

    -X 'SL-PA67R:OOK_PWM:404:1152:13000:1200,invert,match={24}0x5b39b9,repeats>=3'

That should read the remote control reliably.

vg8020, 2018-07-12:
Thank you very much for the excellent explanation. I have given a look at your values and the values in the samples to gain a better understanding. gap and reset, just as you mentioned, seem to work better with a bit higher values.

```
404	1152	1088(1200)		11368(13000)

400	1148	1088			11304
404	1156	1076			11300
404	1152	1084			11300
404	1148	1088			11300
```

The refined option seems to produce more accurate results indeed. Actually, it stops bursts of false-positives that are output when there is no refining. So, success! 👍 
```
pi@raspberrypi:~ $ rtl_433 -q -F json -R 0 -X 'SL-PA67R:OOK_PWM:404:1152:13000:1200,match={24}0x5b39b9,repeats>=3'
{"time" : "2018-07-12 10:43:42", "model" : "SL-PA67R", "count" : 25, "num_rows" : 25, "rows" : [{"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}], "codes" : ["{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98"]}
{"time" : "2018-07-12 10:44:16", "model" : "SL-PA67R", "count" : 25, "num_rows" : 25, "rows" : [{"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}], "codes" : ["{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98"]}
{"time" : "2018-07-12 10:44:39", "model" : "SL-PA67R", "count" : 25, "num_rows" : 25, "rows" : [{"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}, {"len" : 25, "data" : "5b39b98"}], "codes" : ["{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98", "{25}5b39b98"]}
^CSignal caught, exiting!
Reattached kernel driver
```

closed 2018-07-12 by vg8020

zuckschwerdt, 2018-07-12:
Add `,countonly` if your `match` is exact, like in this example, to shorten the output.

vg8020, 2018-07-12:
Much better 👍 
```
pi@raspberrypi:~ $ rtl_433 -q -F json -R 0 -X 'SL-PA67R:OOK_PWM:404:1152:13000:1200,match={24}0x5b39b9,repeats>=3,countonly'
[....]
{"time" : "2018-07-12 12:42:48", "model" : "SL-PA67R", "count" : 25}
^CSignal caught, exiting!
Reattached kernel driver
```