merbanan/rtl_433 #767: see what protocol is being used

dwc56, 2018-07-12:
is there a way to see what protocol is being used when data is being displayed or is the only way to enable  one at a time?

zuckschwerdt, 2018-07-12:
Interesting point. You'd have to compare the "model" key in the output with the listing of `rtl_433 -h`. But the listing will be more verbose than the model-key and there might be multiple possible model's per decoder.
We should perhaps include the protocol number with the data output.

merbanan, 2018-12-09:
@zuckschwerdt maybe this should be added to the -M option ?

merbanan, 2018-12-09:
@zuckschwerdt I sort of need this for my next steps. Is this a quick fix for you are should I do this?

zuckschwerdt, 2018-12-10:
This needs #905 to go in first, but the win32 implementation is untested. I guess I'll merge anyway and win32 needs to be fixed later ;)

zuckschwerdt, 2018-12-10:
I snuck this in with 0e99cb2 now.

closed 2018-12-10 by zuckschwerdt