merbanan/rtl_433 #771: Allocating 15 zero-copy buffers

juergenschubert, 2018-07-17:
I run on a RaspberryPI and do get the follwoing

pi@RTL-433:~ $ rtl_433 -R 32
Registering protocol [1] "Fine Offset Electronics WH1080/WH3080 Weather Station"
Registered 1 out of 102 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
Allocating 15 zero-copy buffers

Just wonder what this does and how to troubleshoot this?

zuckschwerdt, 2018-07-17:
`zero-copy` just means *fast*. Those buffers are for USB data transfer.

closed 2018-07-17 by zuckschwerdt

zuckschwerdt, 2018-07-17:
You are not mistaken, the wording there is somewhat scary. But it's from the rtl-sdr lib, nothing we can do about it ;)

rradar, 2018-08-07:
hmmm I also get this message and afterwards nothing happens?

```
[...]
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tunre gain set to Auto.
Reading samples in async mode...
Tunes to 433920000 Hz.
Allocating 15 zero-copy buffers
```

Any ideas what's up or how to debug? @zuckschwerdt @merbanan maybe?

rradar, 2018-08-07:
Hi @zuckschwerdt 
thank's for your reply. I get exactly the same out put with `-a` and `-A`

```
[...]
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tunre gain set to Auto.
Reading samples in async mode...
Tunes to 433920000 Hz.
Allocating 15 zero-copy buffers
```

zuckschwerdt, 2018-08-07:
Then maybe run `rtl_sdr -f 433920000 -s 250000 -n 15000000 data.cu8` and inspect the data.cu8 with e.g. http://triq.net/iqs
The goal is to see if there is any data reception. Compare the visual output with samples from here https://github.com/merbanan/rtl_433_tests/tree/master/tests to get a feel for it.

rradar, 2018-08-07:
Hey @zuckschwerdt 

I did it! And it looks there is something?

[image]

rradar, 2018-08-07:
Wierd. The "15 zero-copy" message is gone.

Now it just stops like this:

```
[...]
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
```

zuckschwerdt, 2018-08-07:
That looks ok. But the band is completly empty -- no signals. Try another frequency, maybe a known radio station. Say there is an "98.3 FM" then use `-f 98300000` and again look at the signal.
Try for different frequencies. Maybe also record 10 minutes worth of data or raise the sample rate to 2048000.

rradar, 2018-08-19:
I could getting results with adding the external antenna to the stick and let it run for a longer time.

Thank's for your help!

gersmit, 2018-12-02:
Hello you all, hope someone can help me:  
i read text above but is not helping me out. I have the problem with "allocating 15 zero-copy buffers". When i start rtl_test after installing RTL_SDR on the RPI 3B+ i get the row "allocating 15 zero-copy buffers". When i run, in my case AIS decoder, then i get decoding for few seconds and stops. What can i do to solve this problem, my other RPI 3B give no problems.
Many tanks for answer and greetings Ger

zuckschwerdt, 2018-12-02:
You might want to poke around the rtlsdr issues about this or ask there.

gersmit, 2018-12-02:
Hi Christian, 
Thanks for replay, i did already poke around but nobody reacting. I read the issues above and thought maybe i can find answer here. But not so i think?. Probably it is possible to overrule the zero-copy? i dont no.
If you no something what i can try, please let me no.

Thanks, Ger

gersmit, 2018-12-02:
Sorry Christian and Benjamin, did not see commend from Benjamin. Thanks for that and going to try.

gersmit, 2018-12-02:
HI Benjamin, can you tell me what is the best way to change the librtlsdr? I am not a specialist.

Thanks Ger

merbanan, 2018-12-02:
@gersmit no sorry, it depends on the distribution.

mojso, 2019-03-14:
sudo apt-get install librtlsdr0

In rtl_433 no longer shows 
Allocating 15 zero-copy buffers

but with librtlsdr0, I can not use rtl_tcp

merbanan, 2019-03-14:
I guess you need to compile librtl-sdr on your own.

mojso, 2019-03-14:
can you tell me how to do it

zuckschwerdt, 2019-03-15:
Describe in more detail what you want and what you already tried. [Ask smart](http://www.catb.org/~esr/faqs/smart-questions.html). 

> sudo apt-get install librtlsdr0
> In rtl_433 no longer shows: Allocating 15 zero-copy buffers

What did you have installed before? Try `find / -name librtlsdr.\*`

> but with librtlsdr0, I can not use rtl_tcp

The program `rtl_tcp` is in `rtl-sdr` package, the `librtlsdr0` is only a library.