merbanan/rtl_433 #772: error

dwc56, 2018-07-18:
when i have this runing on a raspberry pi 3 to send mqqt after about 3 hours then it gives a error and stops if i restart it it works for about 3 hours how can i fix it

cb transfer status: 1,  canceling...
WARNING: async read failed (-5),

Library error -5, exiting...

zuckschwerdt, 2018-07-18:
Using a supervisor and using network instead of pipes might help. See http://triq.net/articles/rtl-433-networking

closed 2018-07-18 by zuckschwerdt

rct, 2018-07-18:
I think you might also want to check your RPi power supply.  I seem to recall others have had similar problems with long running processes on the RPi that were resolved by ensuring there was adequate clean power.  I think the R820 dongles draw close to .5 Amp.   So make sure you've got a quality supply that can deliver on the order of 2.5 Amps.   Also make sure you are using a USB cable that isn't causing voltage drop because the conductors are too small.  Some regulated supplies are designed to provide 5.1 or 5.25 Vdc to make up for the drop on the cable.

https://www.raspberrypi.org/documentation/hardware/raspberrypi/power/